evcc-io/evcc #1134: Berücksichtigung der Site-Power im PV only Setup

MehrCurry, 2021-06-10:
**Describe the bug**
Wenn man ohne Grid-Meter, nur mit einem PV Meter arbeitet, wird der Ladestrom schrittweise bis zum Maximum erhöht, da sich die Site-Power nicht verringert.

Siehe https://github.com/andig/evcc/discussions/1130

**To Reproduce**
Setup nur mit PV meter

```
site:
  title: Zuhause # display name for UI
  meters:
    pv: solarlog
```

**Expected behavior**
Korrekte Berechnung des Ladestroms

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.56 (cad5996)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here
[main  ] INFO 2021/06/10 17:28:05 evcc 0.56 (cad5996)
[main  ] INFO 2021/06/10 17:28:05 using config file evcc.yaml
[mqtt  ] INFO 2021/06/10 17:28:05 connecting evcc-1635883614 at 192.168.187.36:1883
[mqtt  ] DEBUG 2021/06/10 17:28:05 192.168.187.36:1883 connected
[lp-1  ] WARN 2021/06/10 17:28:05 poll mode '{always 5m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2021/06/10 17:28:05 poll interval '5m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

pv: solarlog
------------
Power: 635W

loadpoint 1
===========

charger: wallbox
----------------
Charge status: A
Enabled:       false
Duration:      1h26m38s
Diagnostic dump:

vehicle 0
---------
SoC:           79%
Charge status: A
Capacity:      64kWh
Range:         334km
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug

[... 120 lines cut ...]
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 next soc poll remaining time: 4m49s
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 vehicle soc: 56%
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 vehicle range: 295km
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 max charge current: 15.0A = 13.2A + 1.8A (-1269W @ 3p)
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 pv timer reset
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 pv max charge current: 15A
Jun 10 07:26:23 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:23 max charge current: 15
Jun 10 07:26:33 evcc evcc[9730]: [site  ] DEBUG 2021/06/10 07:26:33 ----
Jun 10 07:26:33 evcc evcc[9730]: [site  ] DEBUG 2021/06/10 07:26:33 pv power: 1269W
Jun 10 07:26:33 evcc evcc[9730]: [site  ] DEBUG 2021/06/10 07:26:33 site power: -1269W
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 charge power: 10382W
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 charger status: C
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 vehicle status: C (Weißfuß)
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 next soc poll remaining time: 4m49s
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 vehicle soc: 56%
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 vehicle range: 295km
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 max charge current: 16.0A = 15.0A + 1.8A (-1269W @ 3p)
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 pv timer reset
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 pv max charge current: 16A
Jun 10 07:26:33 evcc evcc[9730]: [lp-1  ] DEBUG 2021/06/10 07:26:33 max charge current: 16
</pre></details>

Kommentar von @andig in der Diskussion:

_Verdammt! Ich glaube das war genau der Grund warum ich das ursprünglich nciht erneut eingebaut hatte. Verstehe nicht, warum das beim Issue nicht aufgefallen ist._

andig, 2021-06-10:
Follow-up to https://github.com/andig/evcc/issues/953

@junibart: jetzt bin ich verwirrt, warum das bei Dir funktioniert?

andig, 2021-06-11:
/cc @newmcdonald ping @junibart

mark-sch, 2021-06-16:
@andig Würde mich wundern wenn es vorher funktioniert hat, aktuell wird die PV Leistung 1:1 in die Grid Leistung als Einspeisung kopiert. Die WB nimmt als Verbraucher keinen Einfluss aktuell auf die Einspeiseleistung, somit ist es kein Regelkreislauf und die WB marschiert immer zum Maxcurrent.

andig, 2021-06-16:
Vermute ich auch. Deshalb hab mich mich so gewundert… Ich überleg mir was.

premultiply, 2021-06-18:
Das darf ja dann nicht die Einspeisung sein sondern muss die direkte Leistungsvorgabe für die Ladung sein - abzüglich des Wallbox-Verbrauchs. Letzteres ist nur bei mehreren Wallboxen relevant.

closed 2021-06-21 by andig