evcc-io/evcc #1139: OCPP server

drc38, 2021-06-10:
**Is your feature request related to a problem? Please describe.**
A number of modern chargers are providing an ocpp1.6j compliant client, and have a requirement that the user is able to determine the ocpp server they connect to for demand response/aggregation, fleet charging management etc. Evcc could provide a ocpp server and then use the smart charging profile and other features to provide a generic interface for vehicle chargers that support the ocpp protocol. To add the charger the user would then need to change the ocpp host on the charger to the server evcc is being run on and configured websocket tcp port.

**Describe the solution you'd like**
It would be great if you could consider adding an ocpp server and generic ocpp charger profile to the evcc project.

**Describe alternatives you've considered**

**Additional context**

andig, 2021-06-11:
Bisher ist die Idee immer daran gescheitert, dass kein entsprechender Charger zu Testzwecken zur Verfügung stand. Um das überhaupt zu evaluieren bräucht zu Zugang zu einem Charger, vorzugsweise zum Administrationsinterface.

andig, 2021-06-12:
ping @drc38

drc38, 2021-06-13:
I have a ocpp 1.6j compatible charger and a car with ovms, no solar just yet though. This python library and home assistant front end could be useful https://github.com/drc38/ocpp/tree/dev starting points.

andig, 2021-06-13:
> Bisher ist die Idee immer daran gescheitert, dass kein entsprechender Charger zu Testzwecken zur Verfügung stand. Um das überhaupt zu evaluieren bräucht zu Zugang zu einem Charger, vorzugsweise zum Administrationsinterface.

Libraries habe ich, Geräte nicht...

andig, 2021-06-16:
Mangels Zugriff auf ein Testsystem mache ich hier mal zu.

andig, 2021-06-27:
/cc @hackex

andig, 2021-07-04:
Scheint so als wäre OCPP SC eine tolle Idee die aber mangels Geräten niemand braucht ;)

xpcone, 2021-07-07:
OCPP wird spannend mit Juice Techniology Dingen, aber aktuell warte ich noch auf die Antwort für den Juice Charger Easy 1 , ob der wie seine Brüder wenigstens OCPP kann, aktuell scheint die Box ja nur ne zu große Anschlussdose mit RasPi 3B und RFID zu sein.

andig, 2021-07-07:
> OCPP wird spannend mit Juice Techniology Dingen, aber aktuell warte ich noch auf die Antwort für den Juice Charger Easy 1 , ob der wie seine Brüder wenigstens OCPP kann, aktuell scheint die Box ja nur ne zu große Anschlussdose mit RasPi 3B und RFID zu sein.

Verstehe ich nicht. Was willst Du damit sagen bzw. was hat das mit dem Thema zu tun?

premultiply, 2021-07-07:
Vor allem was soll bei Juice spannend sein?
Immer vorn dabei mit Features die man garantiert nicht braucht exklusive die marktüblichen Grundfunktionen.
Wichtig: Überfahrbarkeit und gebürstetes Alu. Was braucht der Elektrofahrer in der Praxis mehr? :D

xpcone, 2021-07-07:
Man kann die "Stationen" (den RasPi zum JuiceBooster) steuern, z.B. zum geführten Laden oder erkennen verschiedener Fahrzeuge.
Ein Juice Charger ist ja auch nur ein JuiceBooster mit Zubehör, für mcih war z.B. interessant, weil man mit der "Easy Station" halt einfach mal 900€ zurück bekommen hat.

andig, 2021-07-08:
Lass uns doch bitte hier bei Ocpp Smart Charging bleiben.

andig, 2021-07-08:
> I have an ocpp 1.6j compatible charger 

@drc38 any interest in joining Slack for further discussion?

frzifus, 2021-07-25:
which slack channel? is there a matrix bridge?

andig, 2021-07-25:
Slack access is documented in the Wiki: https://github.com/andig/evcc/wiki#requirements

andig, 2021-08-18:
Sieht aus als gäbe es tatsächlich einfach keinen Bedarf für OCPP 😅. Falls doch machen wir hier wieder auf.

closed 2021-08-18 by andig

sfrevel, 2021-11-25:
Hallo zusammen, ich kram das nochmal raus... denn ich stehe aktuell vor der Herausvorderung eine ABL emH2 gerne  einzubinden. Gerade hab ich mit einem Service Mitarbeiter von ABL gesprochen  (sehr nett übrigens) der mit folgenden Link auf die Doku geschickt hat https://www.ablmobility.de/global/downloads/software/Integration_Manual_1.7.pdf Die https://www.ablmobility.de/global/downloads/datenblaetter/emh2/eMH2_DE/ABL_2W2241_DE.pdf müsste  man so eigentlich anbinden können, richtig? Ich hab aktuell eine ABL_2W2231 - also einen Slave im Standalone Betrieb und noch nicht montiert (!) - von den örtlichen Stadtwerden geliefert bekommen. Die würden den 2231 aber gegen einen 2241 Master austauschen, da sie selbst auch an dem Thema PV-Überschussladen Interesse hätten aber leider nicht so wirklich den technischen Hintergrund was Software usw. angeht. evcc hab ich mit SMA WR und EMETER schon laufen. Das funktioniert echt prima und super simple !  👍

premultiply, 2021-11-25:
Alle ABL-Boxen mit dem EVCC2-Controller, also eMH1, eMH2 und eMH3 solltest du via RS485 bereits anbinden können.
Siehe hier: https://docs.evcc.io/docs/devices/chargers#abl-emh
Slave sollte dazu völlig ausreichend sein. Wenn man mehreren Boxen unterschiedliche IDs zuweist kann man die auch über einen Bus wie vorgesehen ansteuern.

andig, 2021-11-25:
> Ich hab aktuell eine ABL_2W2231 - also einen Slave im Standalone Betrieb und noch nicht montiert (!) - von den örtlichen Stadtwerden geliefert bekommen. Die würden den 2231 aber gegen einen 2241 [...]

@sfrevel frag doch Dein Stadtwerk mal, ob sie sich nicht unter info@evcc.io mal melden wollen und wir schauen ob es nicht ein spannendes Pilotprojekt werden könnte?

sfrevel, 2021-11-25:
Das Problem ist das die Slave box keiner LAN anschluß hat die Master aber schon. Bei der Slave müsste man dann einen RS484 zu Ethernet converter einbauen, oder? Hat das schon mal jemand mit einer emH2 gemacht? 

@andig Da ich gerade im Austausch mit den Stadtwerken bin kann ich gern  fragen.

premultiply, 2021-11-25:
Mach gerne dazu ein (weiteres) ABL-Thema unter Discussions auf. ;)
Sollte aber technisch grundsätzlich kein Problem. Ob RS485 oder LAN ist wurscht. Das eine braucht eine Kupferdoppelader, das andere zwei. Wenn du ohnehin Daten- und oder Telefonkabel liegen hast oder legst geht da ebenfalls beides drüber.

sfrevel, 2021-11-25:
hmmm, wo würde man den TCP-RS485 converter den anschließen ?
[image]
[image]
Die zwei RJ45 Anschlüsse sind für Master/Slave Connect wenn ich es richtig vertsanden habe...

andig, 2021-11-25:
Bitte neues Thema. Hier geht es um OCPP!

andig, 2022-01-05:
https://github.com/evcc-io/evcc/pull/1963 nimmt langsam Gestalt an. Wer mag und sich traut kann schonmal einen Blick drauf werfen.

demaggus83, 2022-03-06:
Ich hab eine EVBox Elvi 22 (OCPP 1.6), eine 5,9 kWp Photovoltaik, ab Mai eine E3DC S10x Compact, ab Juni nen Fiat 500 und Plan von Go 😊 Wie kann ich Supporten?

premultiply, 2022-03-06:
/cc @andig

andig, 2022-03-07:
> Ich hab eine EVBox Elvi 22 (OCPP 1.6), eine 5,9 kWp Photovoltaik, ab Mai eine E3DC S10x Compact, ab Juni nen Fiat 500 und Plan von Go (und Vue) 😊 Wie kann ich Supporten?

@demaggus83 das ist ja mal ein starker Antritt, danke! Du kannst Dir gerne https://github.com/evcc-io/evcc/pull/1963 anschauen was in Grundzügen auch schon funktioniert. Noch besser wäre Du meldest Dich mal bei Slack, dann könnten wir zusammen mit @xantalor der auch viel an dem PR gemacht hat besprechen wo es gerade fehlt.

demaggus83, 2022-03-07:
@andig 
Sehr schön. Ich melde mich vermutlich heute Abend in Slack an und melde mich bei euch.

hackex, 2022-03-14:
> Ich hab eine EVBox Elvi 22 (OCPP 1.6), eine 5,9 kWp Photovoltaik, ab Mai eine E3DC S10x Compact, ab Juni nen Fiat 500 und Plan von Go (und Vue) blush Wie kann ich Supporten?
> 
> Aktuell noch ein SMA Wechselrichter mit einer SMA Webbox. Könnte also, falls noch nicht vorhanden, die Webbox anbinden.

Wenn du nicht warten willst, empfehle ich die EVBox Elvi mit folgendem OCPP-Backend von MobilityHouse auf einem HomeAssistant OS (auf einem Raspberry Pi) laufen zu lassen. Funktioniert bei einem Kunden von mir inklusive Regelung des Ladestrom zwischen 8-16A. Ladedauer-Anzeige ist etwas unzuverlässig, aber es hilft für PV-Überschussladen.

Hier die Links:
HomeAssistant OS: https://www.home-assistant.io/
OCPP-Backend für HomeAssistant basierend auf der MobilityHouse Implementierung: https://github.com/lbbrhzn/ocpp

200Puls, 2022-04-25:
Hi,

ich habe die Pulsar Plus von Wallbox. Diese unterstützt auch OCCP und ich suche derzeit anch einer Möglichkeit die mit meiner Fronius PV Anlage zu steuern. 

Ich warte allerdings noch auf das Auto.

andig, 2022-04-25:
Pulsar siehe https://github.com/evcc-io/evcc/pull/3210. Ocpp ist weiter in der Entwicklung.

200Puls, 2022-04-25:
@andig Ach perfekt. Das hatte ich noch gar nicht gesehen.

andig, 2022-07-10:
OCPP has finally been added in https://github.com/evcc-io/evcc/pull/1963

ThomDietrich, 2022-08-25:
Um das hier kurz zu Protokoll zu geben: Seit Firmware Version 054.0 unterstützt der go-eCharger (rev3) nun auch OCPP.

andig, 2022-08-25:
@ThomDietrich wäre mal interessant zu testen, wie der mit https://github.com/evcc-io/evcc/issues/4008 funktioniert- Ergebnisse gerne dort.

zaphod1966, 2022-09-12:
Hallo zusammen, ich bin über Google-Suche auf den Beitrag gestossen.. Ich versuche meine EVBox Elvi mit meiner 18Kwp Anlage und einem Kostal WR mit RS485 zusammen zu bringen. Soweit ich die Beiträge richtig gedeuted habe, ist die OCCP Integration geplant. Ist das schon realisiert? Hat jemand Die EVBox Elvi schon integriert? 
Danke...

andig, 2022-09-12:
Siehe https://github.com/evcc-io/evcc/discussions/3806