evcc-io/evcc #1142: Charging with KEBA is not stopped, if mode is set to "stop" in gui

KanteTaete, 2021-06-11:
**Describe the bug**
If the charging mode is switched from "Nur PV" (only PV) to "Stop" in the GUI, charging continues and is not stopped

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the GUI, charging is ongoing in 'Nur PV' (only PV) mode
2. Click on 'Stop'
3. Charging should stopp, but in the 'Fahrzeug' (vehicle) information, still 'lädt' (charging) is shown
4. See error

**Expected behavior**
Charging should stopp if it's selected in the GUI.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.56 (cad5996)

```

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
`
uri: xxx.xxx.xxx.xxx:7070 # uri for ui
interval: 10s # control cycle interval

log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: fatal

meters:
- name: grid
  type: default
  power:
    type: calc
    add:
    - type: modbus
      value: Power
      model: sunspec
      uri: xxx.xxx.xxx.xxx:502 # Wechselrichter
      id: 126
      scale: -1
    - type: http
      uri: http://xxx.xxx.xxx:8081/xxx
      jq: .data[0].tuples[0][1] #parste response json
      scale: 0.01
# Berechnung: grid = xxx - Wechselrichter

- name: pv
  type: default
  power:
    type: modbus
    value: Power
    model: sunspec
    uri: xxx.xxx.xxx.xxx:502 # Wechselrichter
    id: 126

chargers:
- name: keba
  type: keba
  uri: xxx.xxx.xxx.xxx

site:
  title: xxx
  meters:
    grid: grid
    pv: pv # pv meter

loadpoints:
- title: Garage # display name for UI
  charger: keba # charger
  mode: pv
  soc:
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 100 # charge to 100%
  phases: 2 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: -2800 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 0 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
`
</pre></details>

<details><summary>Show evcc log output with <code>--log error</code>:</summary><br><pre>
[lp-1  ] INFO 2021/06/11 14:18:02 set charge mode: off
[lp-1  ] DEBUG 2021/06/11 14:18:03 charge power: 3801W
[lp-1  ] DEBUG 2021/06/11 14:18:03 charge currents: [8.28 8.19 0]A
[lp-1  ] DEBUG 2021/06/11 14:18:03 detected phases: 2p [8.28 8.19 0]A
[lp-1  ] DEBUG 2021/06/11 14:18:03 charger status: C
[lp-1  ] DEBUG 2021/06/11 14:18:03 charger vehicle id: 0000000000000000
[lp-1  ] DEBUG 2021/06/11 14:18:03 max charge current: 0
[lp-1  ] DEBUG 2021/06/11 14:18:09 charge power: 3824W
[lp-1  ] DEBUG 2021/06/11 14:18:09 charge currents: [8.27 8.2 0]A
[lp-1  ] DEBUG 2021/06/11 14:18:09 detected phases: 2p [8.27 8.2 0]A
[lp-1  ] DEBUG 2021/06/11 14:18:09 charger status: C
[lp-1  ] DEBUG 2021/06/11 14:18:09 charger vehicle id: 0000000000000000
</pre></details>

andig, 2021-06-11:
/cc @mark-sch

KanteTaete, 2021-06-11:
Please find below some details of my analysis and some information from the KEBA manual which indicates, that the method of stopping charging can't work as implemented.

> [keba  ] TRACE 2021/06/11 14:54:54 send to xxx.xxx.xxx.xxx:7090 curr 0

It looks as if evcc sets the current 'correctly' to 0A.

>[keba  ] TRACE 2021/06/11 14:55:00 recv from xxx.xxx.xxx.xxx:7090 {"Max curr": 6000}

But the charger (KEBA) doesn't stop charging then but reduces to the minimum charging current of 6A.

Furthermore, the manual from KEBA says:
>In general, it is not recommended to use the curr command, since the current can be easily controlled via the non-permanent currtime.

> Command structure:
> Command: curr [c]
> Parameters: [c]: Current setting
> Additional Description: Current value in mA, Possible values: 6000 - 63000

--> so the minimum current which can be set is 6A.

In contrast, with currtime as recommended, the charging can be stopped as well:
>Command: currtime [c] [t]
> Parameters: [c]: Current setting Current value in mA.
> Additional Description: Possible values: 0; 6000 - 63000
> Parameters: [t]: Time delay 
> Additional Description: Timeout in seconds before the current will be applied. Possible values: 0; 1 - 860400. Values higher than 860400 will be discarded.

Would a change to that command be needed?

andig, 2021-06-11:
`curr` hat bisher funktioniert. Dein Log zeigt nirgendwo, welche Daten egtl. an die KEBA geschickt werden, ohne lässt sich auch nicht sagen warum das nicht funktionieren sollte.

andig, 2021-06-11:
Nutzt du rfid?

andig, 2021-06-13:
ping @KanteTaete

KanteTaete, 2021-06-13:
Sorry, wurde von github nicht über die neuen Kommentare informiert... also weiter auf deutsch...

> Dein Log zeigt nirgendwo, welche Daten egtl. an die KEBA geschickt werden

doch, hier:
`[keba  ] TRACE 2021/06/11 14:54:54 send to xxx.xxx.xxx.xxx:7090 curr 0`

Nein, rfid nuze ich nicht.
Das Manual sagt aber ganz klar, dass es nicht funktionieren kann.
Gibt's denn jemanden, der mit einer aktuellen Keba WB prüfen kann, ob es dort funktioniert?

andig, 2021-06-13:
> Das Manual sagt aber ganz klar, dass es nicht funktionieren kann.

Genau deshalb machen wir das ja auch nciht so: https://github.com/andig/evcc/blob/master/charger/keba.go#L231. Du hast in Deinem Log irgendeinen irrelevanten Teil rausgegriffen und machst daran einen Fehler fest den es so nicht gibt :(

andig, 2021-06-14:
@mark-sch kannst Du hier Licht ins Dunkel bringen?

andig, 2021-06-16:
Da uns hier ein aussagekräftiges Logfile fehlt mache ich mal zu. Gerne mit Log einschl. der `ENA` Statements wieder öffnen.

mark-sch, 2021-06-16:
@andig Kann das Problem auch nicht nachvollziehen, habe ja einige Kebas im Einsatz.

KanteTaete, 2021-06-16:
Schade, dass Ihr die Issues so schnell schließt. Da kommt man ja zeitlich nicht mal dazu, die gewünschten Daten zur Verfügung zu stellen... :-(

Jedenfalls ist in den mir derzeit zur Verfügung stehenden Logs kein ena-Kommando enthalten, der betreffende Log ist aber auch nicht vollständig.
Da mein Raspi derzeit spinnt, kann ich den vermutlich auch nicht innerhalb von ein paar Stunden zur Verfügung stellen.
Sobald ich einen vollständigen Trace-Log habe, werde ich den dahingehend nochmals untersuchen und zur Verfügung stellen.

Was mir allerdings aufgefallen ist in der https://github.com/andig/evcc/blob/master/charger/keba.go#L224:
`var d int
 if enable {
   d = 1
}`

Die Variable "d" wird definiert und wenn "enable" true ist, wird diese auf 1 gesetzt.
Im Falle von "enable = false" wird diese aber nie auf 0 gesetzt.

Frage: Wird die Variable bei der Definition automatisch mit "0" initialisiert? 

...mit Go kenne ich mich nicht aus und bei anderen Programmiersprachen ist dies nicht der Fall.

andig, 2021-06-16:
Ja! Es geht ja bei Anderen auch…

andig, 2021-06-19:
> Schade, dass Ihr die Issues so schnell schließt. Da kommt man ja zeitlich nicht mal dazu, die gewünschten Daten zur Verfügung zu stellen... :-(

meld Dich einfach wenn Du eins hast. Bist dahin gibts nix zu tun.

KanteTaete, 2021-06-20:
Also, wie gewünscht, anbei ein neuer Trace-Log.

Aber vorab kann ich bereits sagen, dass im kompletten Trace an keiner Stelle das "ena" Kommando gesendet wird!
Es werden folgende Befehle gesendet: 
- send to 192.168.yyy.xxx:7090 report ... also Report 2, 3, 100)
- send to 192.168.yyy.xxx:7090 curr ...

Hier nun der Log.
Da ich nicht die 6000 Zeilen posten wollte, habe ich die [http] und [modbus] Einträge gekürzt.
Bevor der Verdacht aufkommt: Ich habe natürlich kein "send to 192.168..." gelöscht.

[main  ] INFO 2021/06/20 14:44:01 evcc 0.56 (cad5996)
[main  ] INFO 2021/06/20 14:44:01 using config file /etc/evcc.yaml
[main  ] INFO 2021/06/20 14:44:01 listening at 192.168.yyy.yyy:7070
[modbus] ...
[lp-1  ] WARN 2021/06/20 14:44:12 soc.levels are deprecated and will be removed in an upcoming release
[cache ] TRACE 2021/06/20 14:44:12 title: Garage
[cache ] TRACE 2021/06/20 14:44:12 minCurrent: 0
[cache ] TRACE 2021/06/20 14:44:12 maxCurrent: 16
[cache ] TRACE 2021/06/20 14:44:12 phases: 2
[cache ] TRACE 2021/06/20 14:44:12 activePhases: 2
[cache ] TRACE 2021/06/20 14:44:12 hasVehicle: false
[keba  ] TRACE 2021/06/20 14:44:12 send to 192.168.yyy.xxx report 2
[cache ] TRACE 2021/06/20 14:44:12 mode: pv
[cache ] TRACE 2021/06/20 14:44:12 targetSoC: 100
[cache ] TRACE 2021/06/20 14:44:12 minSoC: 0
[keba  ] TRACE 2021/06/20 14:44:12 recv from 192.168.yyy.xxx {
"ID": "2",
"State": 1,
"Error1": 0,
"Error2": 0,
"Plug": 3,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 0,
"Max curr %": 1000,
"Curr HW": 16000,
"Curr user": 14448,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 7595,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"Serial": "aaaaaaaa",
"Sec": 2150333
}
[site  ] INFO 2021/06/20 14:44:12 site config:
[site  ] INFO 2021/06/20 14:44:12   meters:    grid ✓ pv ✓ battery ✗
[cache ] TRACE 2021/06/20 14:44:12 siteTitle: daheim
[site  ] INFO 2021/06/20 14:44:12     grid:    power ✓ energy ✗ currents ✗
[cache ] TRACE 2021/06/20 14:44:12 gridConfigured: true
[site  ] INFO 2021/06/20 14:44:12     pv:      power ✓ energy ✗ currents ✗
[cache ] TRACE 2021/06/20 14:44:12 pvConfigured: true
[cache ] TRACE 2021/06/20 14:44:12 batteryConfigured: false
[lp-1  ] INFO 2021/06/20 14:44:12 loadpoint 1:
[lp-1  ] INFO 2021/06/20 14:44:12   mode:      pv
[lp-1  ] INFO 2021/06/20 14:44:12   charger:   power ✓ energy ✓ currents ✓ timer ✗
[lp-1  ] INFO 2021/06/20 14:44:12   meters:    charge ✓
[lp-1  ] INFO 2021/06/20 14:44:12     charge:  power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2021/06/20 14:44:12   vehicles:  ✗
[cache ] TRACE 2021/06/20 14:44:12 chargeConfigured: true
[site  ] DEBUG 2021/06/20 14:44:12 ----
[modbus] ...
[site  ] DEBUG 2021/06/20 14:44:12 pv power: 5820W
[cache ] TRACE 2021/06/20 14:44:12 pvPower: 5820
[modbus] ...
[... 265 lines cut ...]
[modbus] ...
[http  ] ...
[site  ] DEBUG 2021/06/20 14:49:03 grid power: -5538W
[site  ] DEBUG 2021/06/20 14:49:03 site power: -5538W
[keba  ] TRACE 2021/06/20 14:49:03 send to 192.168.yyy.xxx report 3
[cache ] TRACE 2021/06/20 14:49:03 gridPower: -5538.139999999999
[cache ] TRACE 2021/06/20 14:49:03 mode: off
[keba  ] TRACE 2021/06/20 14:49:03 recv from 192.168.yyy.xxx {
...
}
[lp-1  ] DEBUG 2021/06/20 14:49:03 charge power: 1041W
[keba  ] TRACE 2021/06/20 14:49:03 send to 192.168.yyy.xxx report 3
[cache ] TRACE 2021/06/20 14:49:03 chargePower: 1040.51
[keba  ] TRACE 2021/06/20 14:49:03 recv from 192.168.yyy.xxx {
...
}
[lp-1  ] DEBUG 2021/06/20 14:49:03 charge currents: [2.28 2.29 0]A
[lp-1  ] DEBUG 2021/06/20 14:49:03 detected phases: 2p [2.28 2.29 0]A
[cache ] TRACE 2021/06/20 14:49:03 chargeCurrents: [2.277 2.294 0]
[cache ] TRACE 2021/06/20 14:49:03 activePhases: 2
[cache ] TRACE 2021/06/20 14:49:03 chargeCurrent: 0
[keba  ] TRACE 2021/06/20 14:49:03 send to 192.168.yyy.xxx report 3
[keba  ] TRACE 2021/06/20 14:49:03 recv from 192.168.yyy.xxx {
...
}

andig, 2021-06-20:
Wo hast du denn etwas ein/ausgeschaltet?

KanteTaete, 2021-06-20:
Den Kommentar bzgl. Formatierung hatte ich schon erwartet, allerdings hatte ich auf die Schnelle nicht gefunden, wie man formatieren kann und musste weg.
Nun ist das Ganze inkl. Bedienungs-Reihenfolge, Beobachtung und erwartetem Verhalten oben angepasst. Ich hoffe, das genügt den Anforderungen.

andig, 2021-06-20:
Ich sehe viele Leerzeilen, aber sonst erscheint mir das ein Suchspiel zu sein. Magst Du mal die Fragen beantworten?

andig, 2021-06-21:
OK, das ist merkwürdig. 

Kannst Du selbst compilieren? Dann würde es helfen, in der `loadpoint.go` ab Zeile 478 mal ein paar `lp.log.INFO` statements einzubauen um zu verstehen was da passiert. Gerne in jedes `if` statement eins. Bitte auch eins in Zeile 1016 um zu schauen ob er da überhaupt rein läuft oder schon vorher aussteigt.

andig, 2021-06-21:
**UPDATE** kannst Du Dir sparen. Du hast `minCurrent: 0` konfiguriert was quatsch ist. Bitte nimm den Defaultwert.

closed 2021-06-21 by andig

KanteTaete, 2021-06-21:
Ich hab's eben nochmals mit dem Default-Wert getestet.
Nun werden die `ena` Kommandos gesendet. :-)

Als Du (@andig) geschrieben hast, `Es geht ja bei Anderen auch`, hätte ich selbst darauf kommen können, dass da etwas in der Konfiguration anders (falsch) sein könnte...
Außerdem bin ich mir nicht sicher, ob ich das Abschalten bei den eingestellten langen Hysterese-Zeiten für `disable`|`delay: 5m` und `guardduration: 10m` überhaupt beobachten hätte können, wenn ich nur wenige Minuten Log verfügbar habe. Für den Test habe ich `guardduration: 3m` gesetzt. Das muss ich mir mal genauer anschauen, wenn ich wirklich im PV-Modus laden kann (bei langzeit-vollem Akku wegen HomeOffice und Netzbezug statt Einspeisung ist das schwierig :-) ).

**Vielen Dank für die Hilfe!**

Noch eine Frage zum Abschluss:
Hätte ich den "Minimum-Wert" für `mincurrent` irgendwo selbst sehen können oder wäre es ggf. sinnvoll, den Wertebereich der Parameter in der Konfiguration zu hinterlegen, um solche Fälle zukünftig zu vermeiden?