evcc-io/evcc #1150: Feature Request: Speedwire-Plugin

premultiply, 2021-06-13:
Dank #1114 haben wir nun vollständigen Speedwire-Support.

Allerdings kann es im Gegensatz zum schon bisher bereits unterstützten SHM nun auch mehrere Wechselrichter (PV oder Batterie) installationsseitig geben, die mittels calc-Plugin addiert werden müssen.

Dazu fehlt allerdings bislang das passende Plugin analog zu Modbus.

andig, 2021-06-14:
/cc @DerAndereAndi @bboehmke seht ihr den Bedarf? M.E. wäre das mit wenig zusätzlichem Code machbar, nur die Schnittstelle zum Plugin wäre eine Andere.

DerAndereAndi, 2021-06-14:
Ja, wenn man mehrere PV Wechselrichter hat, ist das hilfreich. Ich hab z.b. 1 SMA und 1 Kostal. Wäre super wenn das `calc` plugin mit allen PV Wechselrichtern funktionieren könnte.

closed 2021-07-04 by andig