evcc-io/evcc #1153: Access to EVCC API via HTTP GET

Alestrix, 2021-06-15:
**Is your feature request related to a problem? Please describe.**
Some home appliances (e.g. Shelly, Tasmota) can send HTTP GET requests on certain triggers, but do not allow HTTP POST or PUT commands. Therefore it is not possible to control EVCC from those devices.

**Describe the solution you'd like**
EVCC should accept control commands also via HTTP GET, e.g. `http://<evcc-ip>/api/mode?off`

**Describe alternatives you've considered**
- Flash my Shelly with Tasmota, but that can't do POST either.
- Send an HTTP request not to EVCC, but to the Wallbox. But the WB also requires HTTP POST. Also, this would cause EVCC and the WB to get out of sync and EVCC would quickly revert the change.
- Set up a web server to receive GET from the appliance and send POST to EVCC, but that would make the functionality rely on an additional service to be always running. Also, I am too stupid to set this up.
- Wait for EVCC to support load management - see below.

**Additional context**
My Shelly 3EM can send an HTTP request when the grid current on one or more phase conductors crosses a set value, e.g. when exceeding 35A. I would like it to disable charging in this case as a safeguard.
Even when EVCC introduces load management, EVCC is possibly not fast enough (dependent on the poll interval), so a trigger from a device that sits "at the source" and can react immediately might still be the better solution.

andig, 2021-06-16:
I don't think we should do this. Making an ugly api to accomodate incapable clients is not the way to go. Workarounds exist, too.

> My Shelly 3EM can send an HTTP request when the grid current on one or more phase conductors crosses a set value, e.g. when exceeding 35A. I would like it to disable charging in this case as a safeguard.

The use case of using error-prone software to safeguard against hardware issues is imho failed. Security should be handled in hardware.

closed 2021-06-16 by andig

Alestrix, 2021-07-18:
> I don't think we should do this. Making an ugly api to accomodate incapable clients is not the way to go. Workarounds exist, too.

Why is allowing GET on top of POST "ugly"? I recently looked into Swagger and they too allow for variables within the path and as a GET query - are the Swagger devs less capable? And calling two of the most active and feature rich smart home devices "incapable" is quite a call...

> The use case of using error-prone software to safeguard against hardware issues is imho failed. Security should be handled in hardware.

Of course this is not the only safeguard - there's still an SLS that switches off electricity if currents are too high. But that doesn't mean it's a bad idea to have an additional line of defence.

Don't get me wrong - I fully understand if this is not a priority, but the reasons stated are IMHO pretextual.

andig, 2021-07-18:
This is not authorative, but for what its worth: https://stackoverflow.com/questions/19637459/rest-api-using-post-instead-of-get#19642955

Alestrix, 2021-07-18:
> This is not authorative, but for what its worth: https://stackoverflow.com/questions/19637459/rest-api-using-post-instead-of-get#19642955

Thanks and nice read! According to those responses (and the referenced Wikipedia article), switching between charging modes should be done via a PATCH request 😈 .

But your point is taken and I have sent a feature request to Shelly. Let's see how this turns out...