evcc-io/evcc #1162: Ladegrenze festlegen

userjp123, 2021-06-19:
**Problem**
Ich bediene evcc fast ausschließlich vom Handy aus.
Dabei passiert es mir öfter, dass ich den Ladebalken berühre:
Ladegrenze wird verstellt, Ladung bricht ab. 
-> Unschön.

**Vorschlag** (im Hinblick auf einfache Realisierung und keine UI-Änderung)
Ladegrenze wird nur verstellt, falls sich die Grenze OBERHALB des aktuellen Ladezustandes befindet.
Andernfalls wird die Grenze einfach ignoriert.
So hat man bei einer versehentlichen Änderung Zeit, diese wieder zu korrigieren ohne dass die Ladung abbricht.

Oder macht es Sinn die Grenze unterhalb des akt. Zustandes zu setzen ? (Mir fällt da nichts ein)

andig, 2021-06-19:
/cc@naltatis. Ignorieren oder mit 10s Delay setzen?

naltatis, 2021-06-19:
Hm, ich glaube schon dass es sinnvoll sein kann die Grenze unter den aktuell SoC zu legen. Wenn ich den Wagen bis 100% voll lade weil ich weiter weg möchte stelle ich das oft direkt danach schon wieder runter auf 80% damit ich’s nicht vergesse.

Zeil sollte eigentlich sein, dass wir die fehlbedienungen vermeiden. Nutzt die Android? Unter iOS Safari funktioniert das eigentlich sehr zuverlässig dass man nicht beim Scrollen den slider verschiebt.

naltatis, 2021-06-19:
> /cc @naltatis. Ignorieren oder mit 10s Delay setzen?

10s delay für den Api call wäre einfach umzusetzen und würde zumindest das teilproblen „Ladung abgebrochen“ lösen wenn man gemerkt hat, dass man es versehentlich verstellt hat.

userjp123, 2021-06-19:
> Hm, ich glaube schon dass es sinnvoll sein kann die Grenze unter den aktuell SoC zu legen. Wenn ich den Wagen bis 100% voll lade weil ich weiter weg möchte stelle ich das oft direkt danach schon wieder runter auf 80% damit ich’s nicht vergesse.

...aber das machst du doch erst, **nachdem** er fertig geladen (100%) hat, oder ? Wenn du **während** der Ladung runterstellst, wird die Ladung direkt abgebrochen.

userjp123, 2021-06-19:
> Nutzt die Android? Unter iOS Safari funktioniert das eigentlich sehr zuverlässig dass man nicht beim Scrollen den slider verschiebt.

Android (LineageOS). Einmal aus Versehen draufgetappt, dann ist es passiert. iPhone habe ich noch nicht getestet. (Haben welche in der Familie)

userjp123, 2021-06-19:
> 10s delay für den Api call wäre einfach umzusetzen und würde zumindest das teilproblen „Ladung abgebrochen“ lösen wenn man gemerkt hat, dass man es versehentlich verstellt hat.

Wenn der Delay **immer** drin ist, ist das evtl. auch nicht so toll. Besser wäre ein Delay nur dann, wenn ein STOP verhindern werden soll. EinstellGrenze < Akt.SOC.

...ich glaube, ich wäre eher für ignorieren. Für mich ist das eine Fehlbedienung. (eine Error-Message ausgeben ?)

andig, 2021-06-19:
Das meinte ich. Delay nur wenns unter aktuell geht. Dann hast du Zeit zu reagieren.

userjp123, 2021-06-19:
...macht ja nur Sinn, wenn der Slider dann auch 10 sec. **sichtbar** ist. Damit ich sehe, dass die Ladegrenze in 10s scharfgeschaltet wird und ich reagieren kann. Meintest du das so ?
Also ein 10-sec-scharfschalten-delay bei SliderPos < SOC.

Evtl. könnte man die 10sec. auch immer (auch bei SliderPos>SOC) ablaufen lassen (Einfacher und kein Nachteil in der Praxis ?)

naltatis, 2021-06-20:
Ich find das mit den „10sec Delay wenn …“ schon eine Komplexität die wir auch vermeiden könnten. Letztendlich ist das ja nur ein Teilworkaround für die versehentliche Sliderbedienung. Würde hier lieber an der Wurzel fixen. Entweder durch eine technische Änderung mit der der chrome/android slider so funktioniert wie der unter iOS (weiß ich aktuell noch ob das einfach geht) oder wir überdenken den SoC einstellprozess generell noch mal. Gab auch schon Kommentare von Leuten die gar nicht rausgefunden haben, dass das ein slider ist. Hier würde vielleicht ein espliziter „SoC ändern“ Modus helfen und beide Probleme lösen.

userjp123, 2021-06-20:
Ich habe nach der geänderten Einstellung (von vorher Buttons auf Klick/Touch auf den Ladebalken) auch nicht sofort gecheckt, wie das nun gemacht wird. Dachte erst die Funktionalität sei entfallen.
In Android sieht man den Slider ja nie. In Firefox(PC) nur, wenn man sich mit dem Mauszeiger über dem Ladebalken befindet.
Von der Einfachheit der Bedienung ist die aktuelle Lösung ja nicht zu überbieten.
Das Schöne an evcc ist ja das aufgeräumte UI und das sollte auch immer ganz oben stehen. 
Bedingungen aus meiner Sicht:
- Immer sichtbare Anzeige der Ladegrenze (da wüsste man dann schon mal, dass man das einstellen kann)
- Intuitive Bedienung zur Einstellung der Ladegrenze (im Prinzip, wie es jetzt ist, aber Fehlbedienung ausschließen.)

andig, 2021-07-10:
/unstale

sebnaf, 2021-08-03:
Ich würde dies UI-seitig mit einem Icon (Schloss) lock/unlock lösen. 
Idealerweise kann ich den initial state per config setzen, um alle User happy zu machen.

rivengh, 2021-08-16:
Mit ist es tatsächlich auch schon ein paar mal passiert, dass ich am Smartphone mit dem Finger kurz auf den Ladebalken gekommen bin und dadurch den Ladevorgang abgrochen habe.

Mein Vorschlag:
* Slider lässt sich nur verschieben, wenn man ihn länger gedrückt hält (250-500 ms reichen wahrscheinlich schon)
* Ein einfacher Klick oder Touch auf eine freie Stelle des Ladebalkens wird ignoriert (alternativ auch mit kurzer Verzögerung, wie beim Slider)

rivengh, 2021-08-16:
@naltatis Nutze ein Android Device mit Ungoogled Chromium. Ein iOS Device habe ich leider nicht zum Testen da :/

naltatis, 2021-09-05:
Inzwischen kann ich das Problem auch hier auf einem Gerät nachstellen. In https://github.com/evcc-io/evcc/pull/1486 gibts einen Fix dafür. Wenn der Slider-Change mit einer Vertikalen Scrollbewegung zusammen ausgeführt wird, wird der SoC nicht mehr geändert. Das sonstige Verhalten (horizontales Wischen, direkter Klick auf die Wunschstelle) bleibt dabei unverändert.

naltatis, 2021-09-06:
Der Fix ist nun im Master. Bitte checkt in den nächsten Tagen mal ob das neue Verhalten für euch besser funktioniert.

userjp123, 2021-09-06:
> Der Fix ist nun im Master. Bitte checkt in den nächsten Tagen mal ob das neue Verhalten für euch besser funktioniert.

Es funktioniert jetzt fast, wie von dir beabsichtigt.
ABER :-)

Testen von Randbedingungen:
Wenn ich folgendes durchführe, wird (fast) immer die Ladegrenze neu gesetzt:
1. LadeBalken mit dem Finger berühren und direkt (nur wenig, 1 cm ?) nach unten scrollen.
2. Direkt (ohne den Finger zu heben) wieder (nur wenig, 1 cm ?) nach oben scrollen.
3. Finger loslassen. Ladegrenze wird neu gesetzt.

Mehrmals testen. AnfangsRichtung ist egal. Der Wechsel der Scrollrichtung scheint die Probleme zu machen.
Ich weiß, dass das nur eine Randbedingung ist, die sehr selten vorkommt. 

Ernüchterung:
Obiges habe ich mit dem Systembrowser von LineageOS(Android) getestet.
Gerade noch mit einem zweiten Browser auf dem Handy getestet. (Fennec/Firefox)
Wenn ich den Ladebalken anfasse und scrollen will, so wird das scrollen verhindert.
Die Grenze wird aber neu gesetzt, wenn ich den Finger gedrückt halte und horizontal verschiebe.

Vermutung:
Für mich (der ich auch programmiere) sieht es so aus, dass sich dieses Problem (mit diesem Ansatz) nicht, oder nicht zufriedenstellend über die Browser hinweg lösen lässt. Oder nur mit unverhältnismäßig großem Aufwand. 
Vll. lohnt es, über einen einfacheren Lösungsansatz nachzudenken. Dieses hier scheint sehr fragil zu sein und das fällt einem dann irgendwann auf die Füße. (zB. nach einem BrowserUpdate)

Trotzdem vielen Dank für die Mühe !

Bin gespannt, ob diese Probleme bei anderen auch auftreten...

xerion3800, 2021-09-07:
Beim mir auf MIUI 12.5.2 (Android 11) und Edge als Browser reagiert es so wie oben beschrieben. Wenn gescrollt wird, dann wird der Slider wieder auf 100% zurückgesetzt.

blenni, 2021-09-09:
Bei mir funktioniert es jetzt sehr gut, keine versehentliche Verstellung mehr - vielen Dank :)
Ich nutze Android 11 (Arrow OS) mit dem Chrome Browser.

rivengh, 2021-09-10:
> Wenn der Slider-Change mit einer Vertikalen Scrollbewegung zusammen ausgeführt wird, wird der SoC nicht mehr geändert. Das sonstige Verhalten (horizontales Wischen, direkter Klick auf die Wunschstelle) bleibt dabei unverändert.

Super, Vielen Dank @naltatis
Kann das beschriebende Verhalten auch bestätigen (Ungoogled Chromium / Android).

_@userjp123 schrieb in https://github.com/evcc-io/evcc/issues/1162#issuecomment-913534177_ :
>     1. LadeBalken mit dem Finger berühren und direkt (nur wenig, 1 cm ?) nach unten scrollen.
>     2. Direkt (ohne den Finger zu heben) wieder (nur wenig, 1 cm ?) nach oben scrollen.
>     3. Finger loslassen. Ladegrenze wird neu gesetzt.

Punkte 1) und 2) kann ich bei mir nicht nachvollziehen, außer man befindet sich noch oder wieder mit dem Finger im aktiven Bereich des Ladebalkens.

Punkt 3) Ist leider nach wie vor noch so, wurde aber auch in #1486 nicht angepasst.

naltatis, 2021-09-15:
Danke für euer Feedback. Schön, dass es nun besser funktioniert.

@userjp123 Zu deinen Randbedingungen: Ja es ist in der Tat so, dass der Mechanismus nicht greift wenn du in einer Bewegung hoch und die gleiche Strecke wieder runterscrollst. Das halte ich in der Praxis aber für unproblematisch. Direktes Tippen auf die SoC Leiste und horizontales Sliden soll ja bewusst so funktionieren wie bisher.

Guter Punkt mit Mobile Firefox. Den hatte ich in der Tat nicht auf dem Schirm. Dem fehlte noch ein ´preventDefault´ damit er sich genau so für der Android Chrome verhält. Habe ich gerade in #1566 eingebaut.

naltatis, 2021-09-16:
Der Issue ist aus meiner Sicht gelöst, daher schließe ich das Ticket. Weitere Optimierungen dann gerne als neuer Issue.

closed 2021-09-16 by naltatis