evcc-io/evcc #1163: Kein Polling der VehicleAPI bei mehr als einem konfigurierten Vehicle

PanNwt, 2021-06-19:
**Beschreibung**
Wenn zwei oder mehr Fahrzeuge konfiguriert sind, so wird deren Connectstatus im Vehicle Cache Interval abgefragt.

**To Reproduce**
Steps to reproduce the behavior:
1. Verwendung von zwei oder mehr Fahrzeugen in der Konfiguration (Knoten: vehicles)
2. Start von evcc mit trace aktiviert
3. die Vehicle Apis werden im Rhythmus der eingestellten Cachezeit abgefragt

**Expected behavior**
Die VehicleAPI wird nur dann aufgerufen, wenn dies notwendig ist.

**EVCC code details:**
Bei jedem Durchlauf von evcc wird die Methode `findActiveVehicle` aufgerufen. Unabhängig der Status vom Loadpoint und des ausgewählten vehicles wird für alle vehicles der Status per API call abgerufen. Hier zieht dann das Cache Interval des jeweiligen vehicles.

PanNwt, 2021-06-19:
**Lösungsvorschlag:** 
- Das aktive Vehicle am Loadpoint bekommt ein Feld für den aktuellen Status.
- Wenn der Status das Loadpoints und des aktiven Vehicles gleich sind, so geschieht nichts.
- Wenn der Status des Loadpoints und des aktiven Vehicles unterschiedlich sind, so wird versucht das passende Fahrzeug zu finden.

thierolm, 2021-06-20:
hm, ich hatte das schon bei mir getestet und evcc macht das m.E. schon so,
wie du das Ziel beschreibst.

Es gibt aber immer Probleme, wenn deine EVs die Status Infos nicht zeitnah
übermitteln... da kann auch evcc nichts machen.
Wo finden wir dein Log?

andig, 2021-06-20:
Ich bin auch unsicher worin das Problem besteht. Prinzipiell sollten wir den Status allerdings nur abfragen, wenn die Autos an/abgesteckt werden- das will ich nochmal verifizieren.

PanNwt, 2021-06-20:
LogFile [_EVChargeControl_logs.txt](https://github.com/andig/evcc/files/6682828/_EVChargeControl_logs.txt)
Contig [evcc.yaml.txt](https://github.com/andig/evcc/files/6682829/evcc.yaml.txt)

thierolm, 2021-06-20:
Hm,
sowohl dein Zoe als auch dein Smart haben in deinem Log 80% zurückgemeldet.
Dein Load-Target hast du auf mindestens 80% in deiner Config definiert (target: 80).
Aus diesem Grund hat evcc auch nicht gestartet...

```
	Line 2941: [lp-1  ] DEBUG 2021/06/20 19:22:29 charge power: 0W
	Line 2966: [lp-1  ] DEBUG 2021/06/20 19:22:29 charge currents: [0 0 0]A
	Line 3017: [lp-1  ] DEBUG 2021/06/20 19:22:30 charger status: B
	Line 3074: [lp-1  ] DEBUG 2021/06/20 19:22:32 vehicle status: A (Zoe)
	Line 3230: [lp-1  ] DEBUG 2021/06/20 19:22:32 vehicle status: B (Smart ED)
	Line 3231: [lp-1  ] INFO 2021/06/20 19:22:32 vehicle updated: Zoe -> Smart ED
	Line 3234: [lp-1  ] DEBUG 2021/06/20 19:22:32 vehicle soc: 80%
	Line 3235: [lp-1  ] DEBUG 2021/06/20 19:22:32 vehicle range: 97km
	Line 3263: [lp-1  ] DEBUG 2021/06/20 19:22:32 targetSoC reached: 80.0 > 80
```

andig, 2021-06-20:
Ich glaub das Problem ist eher, dass zu oft abgefragt wird?

PanNwt, 2021-06-21:
Wenn mehr als ein Fahrzeug konfiguriert ist, so werden folgende Zeilen bei jedem Durchlauf ausgeführt:
```
	if vs, ok := lp.vehicle.(api.ChargeState); ok {
		status, err := vs.Status()

		if err == nil {
			lp.log.DEBUG.Printf("vehicle status: %s (%s)", status, lp.vehicle.Title())

			// vehicle is plugged or charging, so it should be the right one
			if status == api.StatusB || status == api.StatusC {
				return
			}
```

Die Frage ist, wie man die 1. Zeile so gestaltet, dass die VehicleAPI nur dann abfragt wird, wenn nötig.

thierolm, 2021-06-21:
Ich glaube so langsam verstehe ich worauf du hinaus willst:
Du möchtest verhindern, dass nach dem Erreichen des Vehicle Target SoCs, die Vehicle API weiter gepollt wird.
Nur wenn das Vehicle disconnected (erkennen die schlauen Charger, die Pseudo Charger TP-Link etc aber nicht) oder der Target-Soc geändert wird soll das pollen der vehicle API weitergehen.
Richtig?

Definiere bitte "wenn nötig" etwas genauer.

Warum ist das dir so wichtig? Über die Vehicle Caches kannst du verhindern, dass diese zu of angesprochen werden ...
Bspw. Könntest du diese in deiner Config für deinen Zoe und den Smart auf 10m stellen ... dann gibt's maximal nur 6 Polls / Stunde.

andig, 2021-06-21:
Nein, es geht um das Status Api. Das brauchen wir egtl nur 1x wenn angestöpselt wird.

andig, 2021-06-21:
Könnt ihr den PR mal ausprobieren?

PanNwt, 2021-06-21:
Ja gern, mache ich heute Nachmittag.

thierolm, 2021-06-21:
Ich kann das heute nicht sofort testen.
Halte den Ansatz für zu optimistisch.
Ich hatte an meinem go-e auch zwei Fahrzeuge (Zoe + Audi A3) konfiguriert. Beim Umstecken haben beide Vehicle APIs nicht immer sofort den korrekten Status geliefert ... Wenn evcc den Status nur einmal beim connect versucht zu bekommen, dann gibt es keine Möglichkeit das nachträglich zu aktualisieren/korrigiern und die Beladung zu starten.

andig, 2021-06-21:
Die Gefahr besteht. Mittlerweile können wir allerdings bei vielen APIs einen neuen Status anfordern- also einfach mal probieren. Eine bessere Idee hab ich gerade nicht. 

Woran sollten wir denn festmachen, ob der Status jetzt korrekt ist?

PanNwt, 2021-06-21:
Wechsel auf zweites Auto (Smart) hat leider nicht funktioniert.

andig, 2021-06-21:
Warum nicht? Log? Siw unterstützen wir Smart nicht- wieso sollte das funktionieren?

PanNwt, 2021-06-21:
Sorry, vergessen die Ausgabe in ein Log zu kippen. Jetzt ist leider das zweite Auto gerade unterwegs.

Werde schauen, dass ich in den nächsten Tagen es schaffe es genau auszutesten und melde mich dann mit den Informationen.

andig, 2021-06-23:
Ich hab den PR nochmal aktualisiert um MustRetry Fehler abzuhandeln. Jetzt brauchts ein Logfile (TRACE) falls es nciht funktioniert.

> Wechsel auf zweites Auto (Smart) hat leider nicht funktioniert.

Bitte nochmal: warum sollte das überhaupt klappen? Ich habe keine Chance den Smart via API zu erkennen?

PanNwt, 2021-06-23:
Danke für deine Mühen. Es hat sich jetzt so Verhalten wie gewünscht.

Einen Schönheitsfehler habe ich noch gefunden. Wenn während des Starten von evcc schon das zweite konfigurierte Fahrzeug angeschlossen ist, so wird trotzdem das 1. angezeigt. Hier das Log dazu:
[SecondConfiguredCarIsAlreadyConnected.txt](https://github.com/andig/evcc/files/6702698/SecondConfiguredCarIsAlreadyConnected.txt)

Eine Möglichkeit (war erfolgreich) wäre in der Methode `Prepare` aus `loadpoints.go` das Statement
```
	if len(lp.vehicles) > 0 {
		lp.setActiveVehicle(lp.vehicles[0])
	}
```
um einen Aufruf von `lp.findActiveVehicle()` zu erweitern.

Und zu deiner Frage: Smart wird über den OVMS vehicle Typ supported.

andig, 2021-06-23:
>Und zu deiner Frage: Smart wird über den OVMS vehicle Typ supported.

Das beruhigt mich- ich konnte es mir so g ar nicht erklären :O

closed 2021-06-23 by andig