evcc-io/evcc #1167: Awattar API

joanwa, 2021-06-21:
**Describe the solution you'd like**

Awattar offers dynamic electricity pricing and provides a current price api: https://www.awattar.de/services/api 
which allows to start charging once the price falls below the desired threshold. This seems to fit very nicely into this project of smart wallbox controls

**Describe alternatives you've considered**
There are also other provider of flexible energy tariffs, e.g. tibber: https://developer.tibber.com/

andig, 2021-06-21:
Yeah, its on the todo list in this simple form

andig, 2021-06-22:
Solange es keine Prognose gibt wäre die Idee im Prinzip, im PV Modus- der dann eher ein „günstig“ Modus wäre- auf volle Leistung zu gehen, sobald die Preisschwelle unterschritten wird.

joanwa, 2021-06-22:
Beide, Awattar & Tibber erlauben Preisabfragen in der Zukunft, glaube 1 Tag vorraus. Das sind ja Prognosen

andig, 2021-06-23:
PR ist aktualisiert. Damit sollte es jetzt möglich sein, bei "billigen" Preisen sofort mit maximaler Leistung zu laden. Was noch fehlt ist den Konfigurationsmechanismus analog wie sonst zu machen (mit `type` und weiteren Settings), aber fürs Testen sollte es reichen.

Wer von Euch *hat* denn tatsächlich so einen Tarif um das mal auszuprobieren?

andig, 2021-06-24:
> Wer von Euch hat denn tatsächlich so einen Tarif um das mal auszuprobieren?

niemand? Dann verstehe ich den Feature request nicht ;)

joanwa, 2021-06-24:
doch! aber stromanbieterwechsel erst zum August und Tinkerforge Warpcharger Pro noch auf dem lieferweg

ElectricTOBI, 2021-06-27:
Hallo, ich hab aWATTar am laufen, hab schon eine weile nicht reingeschaut!  
Langt es die textzeilen in den EVCC einzufügen oder muss ich für aWATTar was einrichten?

andig, 2021-06-27:
> Langt es die textzeilen in den EVCC einzufügen oder muss ich für aWATTar was einrichten?

Du willst ja erreichen, dass unter einem bestimmten Preis immer geladen wird? Dann muss der Preis in die Config.

andig, 2021-07-23:
Die Implementierung ist mittlerweile fertig. Ich bin darüber gestolpert, dass über das Tibber API auf live Leistungswerte zu beziehen sind. Benutzt das jemand? Bei Bedarf könnten wir das noch als Zähler (=meter) ausprägen, würde ich aber gerne mit echten Zugangsdaten testen wollen.

closed 2021-07-24 by andig

tpulatha, 2022-09-30:
> Die Implementierung ist mittlerweile fertig. Ich bin darüber gestolpert, dass über das Tibber API auf live Leistungswerte zu beziehen sind. Benutzt das jemand? Bei Bedarf könnten wir das noch als Zähler (=meter) ausprägen, würde ich aber gerne mit echten Zugangsdaten testen wollen.

Hi andi, das ganze könnte jetzt wieder interessant werden, da Tibber jetzt das Pulse auf den Markt bringt und damit einen günstigeren Zugang zu dynamischen Tarifen. Außerdem kann man, wenn man das pulse anschließt keinen eigenen IR-Leser mehr betreiben. Daher wird's um so wichtiger das aus der API auslesen zu können. Glaubst du du könntest noch mal das Feature angehen ? :) 

Ich hab auch Zugangsdaten die funktionieren sobald mein Pulse da ist.

weindler, 2022-10-31:
Ich bin ab 1.November bei Tibber, habe aber leider noch nicht den passenden Zähler, hatte ich schon aber der Zähler von Discovergy ist kaputt, und mit denen zusammenarbeiten ist katastrophe. Jetzt bekomme ich einen vom Bayernwerk, aber ab 1. Nov wäre für mich die Api freigeschaltet, also wenn ich euch irgendwie unterstützen kann damit, dann gerne. Habe ja vor daß ich  mit der Sonnenbatterie (sonnenflat gekündigt) und go-e daß dann immer nachts bei günstigen preisen ins auto kloppe.

GrandAmmor, 2022-12-19:
Mein Pulse ist drauf, der Token liegt vor und der Vertrag beginnt am 01.01.23 bei Tibber. Wenn ich etwas dazu beitragen kann, dieses Feature zu unterstützen, sagt einfach Bescheid!

andig, 2022-12-19:
> Wenn ich etwas dazu beitragen kann, dieses Feature zu unterstützen, sagt einfach Bescheid!

Wenn du willst kannst du uns als Sponsor unterstützen. Das Feature ist schon seit Monaten fertig…

GrandAmmor, 2022-12-19:
Danke für dein Feedback! Heute Nacht werde ich es sogar zum ersten Mal testen, wobei es dabei "nur" um den Ladevorgang bei Auslöser durch den "cheap-Wert" im PV-Modus geht.

Oder habe ich das falsch verstanden und eine zeitgesteuerte Ladezielerreichung auf Basis der zukünftigen Strompreise bei Tibber ist schon möglich?

Ungeachtet dessen bin und bleibe ich Sponsor, weil alles bis hier hin schon unfassbar geil ist! Danke an dieser Stelle!

andig, 2022-12-20:
> Oder habe ich das falsch verstanden und eine zeitgesteuerte Ladezielerreichung auf Basis der zukünftigen Strompreise bei Tibber ist schon möglich?

"cheap" geht heute schon, dynamisch die günstigsten Preise zu nutzen kommt mit https://github.com/evcc-io/evcc/pull/5445