evcc-io/evcc #1172: Manual or Off-Mode

KDC666, 2021-06-23:
I’ve seen similar issues/requests so this one might be duplicated, although I went to the whole list and did not find exactly the same thing. 
I’d like to see an additional mode, maybe called ‘manual’ or maybe even ‘off’ which let me set the current by hand without regulation (‘manual’) or let me use the original App/Interface off my charger (‘off’) – in my case go-eCharger – to set the charger current. 

I see a few use cases for this, one is the 70% cut off in Germany (I had a request for such a feature as well) or wanting to charger slow in Summer when enough power is available all day long to avoid high currents and therefore losses. Another use case is when the stove/oven is used in summer. It is switching really fast and often and the evcc if often to slow just for this scenario. E. g. I have 5kW Power available, the stove switches between 0W and 3kW. So, during this period (e. g. 1h) I’d like to be able to set my charger to 2kW fixed instead of off or taking power from the grid. 
It’s also usefull to speed up the Min(+PV) mode a little if necessary without pulling e. g. 7kW in Sofort-Mode. 

One of the modes suggested above would enable me (and other of course) to achieve all of this and would add a high flexibility. Preferred would be the manual mode because the evcc webinterface would be enough and no other app would be required. But the regulation-off mode where no set values go to the charger from evcc and I could use the go-e app for setting the current would be fine as well.

andig, 2021-06-23:
We could probably expose the max/min currents but not another mode.

KDC666, 2021-06-23:
I thought the easiest way might be to just disable evcc per mode/button. Because thats what I do right now by stopping the service. But this is not really handy. 
Despite from that, I don't care too much about the way how it is implemented, but having access over the charging current is my goal. And if I have to set the max current to e. g. 9A and set mode to Sofort I'm fine with it.

andig, 2021-06-23:
> I thought the easiest way might be to just disable evcc per mode/button. Because thats what I do right now by stopping the service. But this is not really handy.

You can already set the mode to off- why isn't that sufficient?

KDC666, 2021-06-23:
[image]

I just tested this (again). It doesn't work because evcc finds an out of sync state and stops the charging.

Maybe it would be a possibility to set the behavior of "Off" in the yaml configuration. Default as it is, and a new mode which stops the charging when button pressed but stop monitoring the charger state until mode is switched again.

naltatis, 2021-06-23:
Just to be clear, disabling the charger with „stop“ works for your, but you want to get rid of the „out of sync“ message that occurred when you manually enable the charger through its own app again, right?

Afaik the message is just a warning that has no negative effect and evcc does not try to disable your charger again, right?

KDC666, 2021-06-23:
No, thats not the case. Like I said: "evcc finds an out of sync state and stops the charging". It's not just the message, ecvv tries to correct the out of sync state.

DerAndereAndi, 2021-06-24:
Correct, when EVCC runs it is meant to be the sole entity that manages the charging process and hence will stop charging if it is set to off. What you are asking for is not „Off“, you want to „DISABLE“ EVCC overall. For that you could setup your own system that quits/starts the EVCC process/docker container/…

Regarding your assumption that losses are bigger with higher currents: Did you measure this and if so how? What EV is this? In all my measurements and to my knowledge this is not commonly correct.

Regarding EVCC being slow in regulating currents: what is your interval in the configuration? Why is this a problem? Does your household grid connection not cover the load? Regulating PV currents will always and with any system not be perfect. The same would be true if a cloud will come over your house for some seconds once in a while and you PV generation will drop. This feels like a desire for an optimization that is not possible.

Regarding limiting the max currents: Right now this is a static configuration as we don‘t offer a web front end for configuration. The idea is to provide this in someday in the future. But this will also have other implications as with other EVs and chargers the EV will provide the possible min and max currents, depending on which EV is connected. So the desire here to limit the max current is something different to what the EV-charger-system is capable off. There will also be a need for other limits, e.g. if the electrical installation requires limiting the max power. This seems to be rather in that direction, but we need to think much more about this and its side effects.

Overall the ticket system doesn‘t match the state this discussion or feature wish. I would recommend to move this to the discussions section under „ideas“. 

Right now I recommend you setup some personal system to disable/enable the EVCC process overall and use the chargers web interface do set whatever you want. EVCC is not designed and meant for such a need. Maybe other systems are a better fit?

KDC666, 2021-06-24:
First of all, don’t get me wrong. I’m totally happy with evcc and I really appreciate the work you guys put into it. 
But since this is an open source project I guess everyone is free to contribute to the project, either by writing code, donating some money, writing documentation(,…) or by giving some feedback/suggesting improvements. 
And that’s all that this is. It’s obviously not only for me because I expect other people to benefit from this too. I added some use cases which apply to my situation, but I expect others for other people.
To me personally it makes total sense to be able to disable the regulation and be able to use the ‘original’ product alone when you want it to. And that’s all what this issue/request/idea is about. The way how this is done doesn’t bother me that much, whatever fits into the system and is feasible to implement if fine for me.
And like I said, right know I stop the linux deamon by hand and restart it later but this has nothing to do with a handy solution. It’s a workaround. Especially when the family comes into play. I could also modify the code myself (e. g. disabling the monitoring in off mode) but this leads to me merging and re-compiling the code with every new version and I’d be the only one able to use this feature/option.

DerAndereAndi, 2021-06-24:
Don't get me wrong either, we appreciate the feedback.

I would recommend to start an Idea discussion in the forum here on GitHub, so others can more easily weight in if they see similar needs.

As you can see, we are not convinced yet that such a feature is really needed. Also because it is not clear why you have those use cases. They currently provide some assumptions (which are not verified) and some just don't make sense to us right now. See my questions. Hence more discussions would be required.

KDC666, 2021-06-24:
I think I have three more than valid use cases concerning my own situation:

1.	I’m aware of the fact that most grid operators don’t allow to pull higher currents than 20A on one phase. Nevertheless, I have the go-eCharger connected 1phase to be able to charger/regulate 6A to 32A. With the installation of my PV system the electrician had to replace the SLS (previous 50A) to only 36A. This means if no sun is around I can only charge at 6A or 32A. 6A might not be enough, 32A brings me close to the limit of the SLS and an additional load (water boiler, dishwasher, etc.) could push it over the edge. In this scenario I’d like to set the charger to 20A, 16 A or maybe 25A. And by the way, the power loss on the line will be different between 6A and 32A because it is squared, but I have to admit the resistance is not so high and the charger won’t work in his sweet spot at 6A and produce higher losses).

2.	The stove is a valid scenario for me. Assuming using 3 hotplates and each being enabled for around 20s and of 20s (simplified) at around 1.5kW. This gives me enough headroom to charger with 2kW. But the hotplates are being switched ‘randomly’ for the evcc. So, the regulation will be off most off the time: pulling current from the grid or not charging when enough power is available. The difference to the cloud scenario is clearly that I know it will happen and it’s absolutely predictable when. Clouds are a different think, but in principal you’re right. It’s the same from the evcc point of view.

3.	70% cutoff in Germany is maybe the most important point in summer. I’m sure you’re aware that you don’t get paid for power which is above 70% of the Ppeak. So, I often charge the car in the middle of the day. So far, no problem, but I’d like to keep the SoC below 80% is possible and not needed. So, when charging at 32A I maybe end up with a fully charged car after 1.5h and giving free power to the grid operator for the next 1.5h instead of charging at 9A for 3h, having the SoC lower and getting paid.

I don’t want to argue about these use cases anymore, but these are the three that apply for my personal situation the most. Maybe it’s just me, but I don’t think so. And I didn’t wanted to start a discussion at all, I just wanted to make a suggestion. So I’ll leave it as it is here and if you decide to implement this I’ll be very happy, if not I’ll find a solution too (probably removing the monitoring from the exiting Off-Mode and creating a local build from time to time because this seems to fit my needs the most)

closed 2021-06-24 by DerAndereAndi

DerAndereAndi, 2021-06-24:
Closing this here as we don't get any further on this.

Just some comments:

1. Overload Protection is one of your issues. In general that is a task for the charger setup as well and usually done by limiting the max power/currents. This is something we have on our todo list to see what we can do about it at some point. For now I suggest to either set the max current to a lower value than 32A (which you say would reduce losses for you anyway) or stop charging when using the oven if this can lead to an electrical issue in your case.
2. The 70% reduction works by either setting a hard power max limit in the PV inverter config to only provide 70% of the theoretical max power or by dynamically adjusting the power using a smart meter device (e.g. SMA Home Manager 2.0) to never feed more than 70% of the kWp value into the grid. This is required by law in Germany and the electrician needs to show proof of this to the grid company.

I don't get why you don't want to discuss this. Just throwing something over the fence isn't how things usually work. And as we are willing to interact and understand things, discussion is the only way. The only other option is to close the ticket. :(