evcc-io/evcc #1180: MQTT Status // Last Will And Testament

goebelmeier, 2021-06-28:
**Is your feature request related to a problem? Please describe.**
As a mqtt subscriber i am unable to detect if evcc is running or stopped / disconnected.

**Describe the solution you'd like**
A new mqtt topic `status `should be created with a retained "online" payload message as soon as evcc connects to mqtt. If evcc disconnects gracefully it should send a retained "offline" payload message. Furthermore the client should instruct the broker to publish this retained "offline" payload message as soon as the broker loses the connection from the evcc mqtt client (Will or Last Will And Testament or simply LWT). See https://pkg.go.dev/github.com/eclipse/paho.mqtt.golang#ClientOptionsReader.WillEnabled for implementation details

andig, 2021-06-28:
Nicht sicher warum ich das noch nicht gemacht habe. Auf jeden Fall gibts https://github.com/andig/evcc/blob/master/server/mqtt.go#L111.

goebelmeier, 2021-06-28:
Ich weiß, aber LWT wäre mehr Best Practice für solches Statushandling. Dann muss man in dem auswertenden Tool nicht rechnen. Soll ich mich mal an nem PR versuchen?

goebelmeier, 2021-06-29:
Wenn du meinst es ist zu kompliziert, dann bau du es so wie es dir am elegantesten passt :-)

closed 2021-07-04 by andig