evcc-io/evcc #1195: Phases to be configured automatically

baycom, 2021-07-04:
I made a fork of SimpleEVSE-Wifi with dynamic phase switching (see https://github.com/baycom/SimpleEVSE-WiFi). This enables a much closer adjustment of the charging current to the PV surplus.
Now I would like to use evcc with dynamic phase switching, but it seems there is no way of configuring anything like that.

A suitable way would be having a function that sets the maximum power for the EVSE by something like that:

```
function evse_set_power_max(power) {
  if(power < 1380) {
    wget(evseuri + "setStatus?active=false");
  } else {
    if(power < 4140) {
      wget(evseuri + "setNumPhases?numPhases=1");
      if(singlePhaseMeter) {
         wget(evseuri + "setMeterFactor?meterFactor=1");
      }
      amps = power/230;
    } else {
      wget(evseuri + "setNumPhases?numPhases=3");
      if(singlePhaseMeter) {
         wget(evseuri + "setMeterFactor?meterFactor=3");
      }
      amps = power/230/3;
    }
    wget(evseuri + "setCurrent?current" + amps);
    sleep(1);
    wget(evseuri + "setStatus?active=true");
  }
```
In this example the metering is switched optionally if the EVSE is using a single phase power meter.
There is also a modbus compliant remote control possible (see URL above).

Would be there any way to add something like that to evcc?

Deti

andig, 2021-07-04:
1p3p is on the todo list. How does the SimpleEvse support this?

baycom, 2021-07-04:
> 1p3p is on the todo list. How does the SimpleEvse support this?

There is an API call 'numPhases' used like 'GET http://192.168.4.1/setNumPhases?numPhases=3' or MODBUS register 40004.

andig, 2021-07-05:
I'll close this issue for now. 1p3p will be available in due time for Chargers supporting it (most likely OpenWB and Easee for starters).

closed 2021-07-05 by andig