evcc-io/evcc #1228: Support HTTPS

fiolux, 2021-07-20:
**Current state:**
EVCC is served using HTTP.
In this day and age, HTTP is considered insecure and specially marked by some browsers (e.g. Google Chrome).

**Solution:**
Add HTTPS support to EVCC.
The certificate, key, and chain certificates should be read from PEM files supplied to EVCC.
The files would need to be named a certain way e.g.
- cert.pem
- privkey.pem
- chain.pem
and would need to be placed in the same location as the evcc.yaml.

andig, 2021-07-20:
It's pretty simple to do and I'd be happy to accept a PR. However, I'm not sure how useful it is. 

Imho https is most useful when used externally. For that, you'd need additional authentication, too. evcc is already dockerized and thus lends itself for use in a modular environment.
I've personally solved this by deploying Traefik reverse proxy (https://github.com/traefik/traefik) in front of evcc. Traefik handles both SSL termination (Letsencrypt) and Authentication (using https://github.com/thomseddon/traefik-forward-auth).

So, that being said, what should _really_ be core competency of evcc here?

DerAndereAndi, 2021-07-20:
I don't think this is something evcc itself should consider right now. It would make setup way harder and result in many more support requests do to setup issues and also browser issues when using locally signed certificates.

Using a reverse proxy already provides a good solution for now.

closed 2021-07-22 by andig