evcc-io/evcc #1241: Könnte man das Auto Versioning für die HA Addons noch mal probieren?

andig, 2021-07-25:
@andig : könnte man das Auto Versioning für die HA Addons noch mal probieren?

Hier steht was noch zum dockerfile hinzu müsste:

https://developers.home-assistant.io/docs/add-ons/configuration/#build-args

Und hier ein Beispiel des Docker files wie es aussehen kann:

https://github.com/hassio-addons/addon-node-red/blob/main/node-red/Dockerfile

Somit müsste mit folgen zwei Zeilen das versioning im HA für evcc als Addon gehen:

```
ARG BUILD_VERSION
LABEL \
    io.hass.version=${BUILD_VERSION} \
```

Vielen Dank für das tolle EVCC!!

_Originally posted by @thecem in https://github.com/andig/evcc/issues/235#issuecomment-886214967_

andig, 2021-07-25:
Siehe PR. Ich bin mir allerdings nicht sicher, ob das den gewünschten Zweck erfüllt. Wäre super, wenn das jemand lokal mit einem eigenen Namen für das Docker Image testen könnte und schauen ob eine Änderung der Versionsnummer überhaupt hilft.

/cc @cathiele

thecem, 2021-07-25:
Habe ich gerade. Im config file mal auf dev gestellt und siege es geht NICHT, jedoch wenn `version`  im config file anstatt `latest` auf zb. `0.59` gestellt wird und danach auf `0.60` funktioniert es.

somit würde eine Änderung der Version auf die jeweils aktuelle im config.yaml vom rep evcc-hasso-addon eine Update Info in HA auslösen bei der gleich ein Update machen kann.

thecem, 2021-07-25:
[image]
[image]
[image]
[image]

thecem, 2021-07-25:
Habe ein PR mit der aktuellen Version erstellt, bei 0.61 sollte es dann bei allen HA Installationen als Update auftauchen…

andig, 2021-07-25:
Refs https://github.com/andig/evcc-hassio-addon/pull/2. Ich finde das Verhalten immer noch merkwürdig- kann den Erfolg jemand bestätigen?

andig, 2021-07-25:
Egtl auch egal, wir probieren das einfach aus. Danke für Deinen Test!

closed 2021-07-25 by andig

thecem, 2021-07-25:
Sobald 0.61 raus ist werde ich berichten, bin mir sicher das es geht.

thecem, 2021-07-25:
Geht sogar gleich! 😉

[image]

Somit werden nun alle HA Nutzer die nicht aktuell sind informiert!

thecem, 2021-07-25:
Danke Dir! @andig 

muss nun nur noch immer mit der neuen Version upgedated werden!

andig, 2021-07-25:
> muss nun nur noch immer mit der neuen Version upgedated werden!

Das evcc docker Image oder das Addon? Ersteres passiert jetzt, habe gerade noch einen Fehler im Build behoben. Letzteres passiert genau nicht, was auch in der Vergangenheit das Problem war...

thecem, 2021-07-25:
Das addon die config.yaml und hier die Version Info.

andig, 2021-07-25:
> Das addon die config.yaml und hier die Version Info.

Dann war die ganze Aktion völlig umsonst- genau das wird nämlich auch weiterhin nicht aktualisiert. Ich wüsste auch nicht wie - das passiert in einem separaten Repository und beim Releasen werden auch keine Dateien geändert.

thecem, 2021-07-25:
Ok. Somit brauchen wir eine Variable die den letzten Tag des evcc docker Image in die config unter version schreibt?

thecem, 2021-07-25:
https://github.com/hassio-addons/repository-updater

thecem, 2021-07-25:
Evtl. Kann man das auch im docker repo mit einem hook erledigen der bei einem neuem Tag den file im addon repo ändert?

andig, 2021-07-26:
Du kannst das gerne mal ausprobieren und dann einen PR erstellen. Mir fehlt die Zeit dafür...

thecem, 2021-07-26:
Ich werde es versuchen….

thecem, 2021-07-26:
Alternativ würde ich jeweils einen PR für das Addon erstellen um die Version zu aktualisieren.