evcc-io/evcc #1244: Wackeliges Ladeverhalten mit Keba P30 und VW ID.4

mweinelt, 2021-07-26:
**Describe the bug**
Der Ladevorgang wird häufig beendet obwohl noch genug Leistung zur Verfügung steht und eine Drosselung des Chargers völlig ausreichend wäre.

[image]

**Expected behavior**
Drosseln der Ladeleistung entsprechend des Überschusses.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.0.1-alpha (HEAD)
```

(Ist 0.60, scheinbar hat er die ldflags nicht gefressen)

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
[main  ] INFO 2021/07/26 12:07:46 evcc 0.0.1-alpha (HEAD)
[main  ] INFO 2021/07/26 12:07:46 using config file /nix/store/jar7rnra78dpxirj344ziidvhwjs574g-evcc.json
[mqtt  ] INFO 2021/07/26 12:07:46 connecting evcc-578295444 at 127.0.0.1:1883
config
======

grid: grid
----------
Power: -6400W

pv: pv
------
Power: 8264W

loadpoint 1
===========

charger: keba-p30-c
-------------------
Power:          0W
Energy:         29.5kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Charged:        3.0kWh
Identifier:     XXXXXXXX00000000
Diagnostic dump:
{ID:100 Serial:22013526 SessionID:0 CurrHW:16000 EStart:264825 EPres:29820 Started:230000 Ended:0 StartedS:230 EndedS:0 Reason:5 TimeQ:0 RFIDTag:XXXXXXXX00000000 RFIDClass:00000000000000000000 Sec:3624}

vehicle 0
---------
SoC:            44%
Charge status:  B
Capacity:       52kWh
Range:          160km
Finish time:    2021-07-26 10:04:00 +0000 UTC
Climate active: false
Outside temp:   20.5°C
Target temp:    20.5°C
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
{
  "chargers": [
    {
      "name": "keba-p30-c",
      "rfid": {
        "tag": "..."
      },
      "type": "keba",
      "uri": "192.168.110.29"
    }
  ],
[... 472 lines cut ...]
Jul 26 12:21:45 server evcc[3077778]: HTTP/2.0 200 OK
Jul 26 12:21:45 server evcc[3077778]: Content-Type: application/json
Jul 26 12:21:45 server evcc[3077778]: Date: Mon, 26 Jul 2021 10:21:45 GMT
Jul 26 12:21:45 server evcc[3077778]: Server: nginx
Jul 26 12:21:45 server evcc[3077778]: 
Jul 26 12:21:45 server evcc[3077778]: {"status":"success","data":{"resultType":"vector","result":[{"metric":{"__name__":"smartmeter_erzeugung_w","device":"1 EMH00 9448210","instance":"192.168.178.104:9001","job":"sml","vendor":"EMH"},"value":[1627294905.32,"8207.999999970198"]}]}}
Jul 26 12:21:45 server evcc[3077778]: [site  ] DEBUG 2021/07/26 12:21:45 pv power: 8208W
Jul 26 12:21:45 server evcc[3077778]: [http  ] TRACE 2021/07/26 12:21:45 GET https://prometheus.example.de/api/v1/query?query=smartmeter_bezug_w-smartmeter_einspeisung_w
Jul 26 12:21:45 server evcc[3077778]: [cache ] TRACE 2021/07/26 12:21:45 pvPower: 8207.999999970198
Jul 26 12:21:45 server evcc[3077778]: [http  ] TRACE 2021/07/26 12:21:45
Jul 26 12:21:45 server evcc[3077778]: GET /api/v1/query?query=smartmeter_bezug_w-smartmeter_einspeisung_w HTTP/1.1
Jul 26 12:21:45 server evcc[3077778]: Host: prometheus.example.de
Jul 26 12:21:45 server evcc[3077778]: User-Agent: Go-http-client/1.1
Jul 26 12:21:45 server evcc[3077778]: Accept-Encoding: gzip
Jul 26 12:21:45 server evcc[3077778]: HTTP/2.0 200 OK
Jul 26 12:21:45 server evcc[3077778]: Content-Type: application/json
Jul 26 12:21:45 server evcc[3077778]: Date: Mon, 26 Jul 2021 10:21:45 GMT
Jul 26 12:21:45 server evcc[3077778]: Server: nginx
Jul 26 12:21:45 server evcc[3077778]: 
Jul 26 12:21:45 server evcc[3077778]: {"status":"success","data":{"resultType":"vector","result":[{"metric":{"device":"1 EMH00 9448209","instance":"192.168.178.104:9000","job":"sml","vendor":"EMH"},"value":[1627294905.323,"-1207.9999999701977"]}]}}
</pre></details>

Und hier der Leistungsverlauf während dieses Ladezyklus.

[image]

andig, 2021-07-26:
Ohne jetzt alles durchzuschauen- hast Du mal 2 Timestamps im Logfile bei denen die Regelung nicht passt?

mweinelt, 2021-07-26:
> Jul 26 12:21:35 server evcc[3077778]: [lp-1  ] DEBUG 2021/07/26 12:21:35 charger disable

Hier hat er beschlossen abzuschalten.

Laut Graph war die Erzeugung stabil bei rund 8kW

> Jul 26 12:21:35 server evcc[3077778]: [lp-1  ] DEBUG 2021/07/26 12:21:35 site power 552W >= disable threshold 0W

Also weil er gesehen hat, dass wir in Bezug gegangen sind.

Nicht sicher wo ich sehe welche Leistung er in diesem Moment hatte, und ob er noch weiter hätte runterregeln können.

KanteTaete, 2021-07-26:
Hallo,

kann essein, dass die Timings nicht zueinander passen?
evcc hat ein 10s Intervall konfiguriert. Im Screenshot scheinen die Messwerte nur jede Minute aktualisiert zu werden.

mweinelt, 2021-07-26:
Prometheus hat ein 15s Intervall in dem neue Daten reinkommen. Sollte ich das Intervall angleichen?

andig, 2021-07-26:
> Sollte ich das Intervall angleichen?

Also auf jeden Fall solltest Du uns nicht mit "evcc tut komisch" in die Wüste schicken wenn es keinen guten Beleg dafür gibt!

mweinelt, 2021-07-26:
Entschuldige, falls ich euch da verwirrende Informationen bereitgestellt habe.

Was mich irritiert ist folgendes: Ich sehe im Webinterface, dass mit 15s Intervall etwa 4 Minuten lang in "PV Only" geladen wird, obwohl 2/3 der Energie importiert werden muss.

[image]

Grafana und EVCC greifen beide auf ungesampelte Werte vom Prometheus zurück.

andig, 2021-07-26:
> Was mich irritiert ist folgendes: Ich sehe im Webinterface, dass mit 15s Intervall etwa 4 Minuten lang in "PV Only" geladen wird, obwohl 2/3 der Energie importiert werden muss.

Deine Config hat keine enable/disable timer- ich bin grad nicht sicher, ob es da einen default gibt. Im Logfile kann ich den Zeitpunkt leider nicht finden.

andig, 2021-07-26:
>  Im Logfile kann ich den Zeitpunkt leider nicht finden.

Hast Du nochmal ein komplettes Logfile, level `debug` reicht?

mweinelt, 2021-07-26:
Auto ist leider unterwegs, morgen dann wieder. Danke für den Tipp mit den enable/disable delay/thresholds, das habe ich voll übersehen, weil ich mich am Beispiel in evcc-config orientiert hatte.

andig, 2021-07-28:
Meld Dich wenns wieder auftaucht...

closed 2021-07-28 by andig