evcc-io/evcc #1245: Wallbe modbus readings wrong

andig, 2021-07-27:
> Kann das sein?

Ich denke solange wie wir daran geschraubt und interpretiert haben auf jeden Fall. Ggf. auch bei Phoenix.

@premultiply ich bin immer noch unsicher, ob wir über Modbus in den Registern die Rohwerte, oder die gem. Wallbe Einstellungen interpretierten Werte sehen. Falls Ersteres müssten wir die Transformation darauf noch anwenden- aktuell haben wir ja nur eine Standardtransformation implementiert.

@lorbetzki kannst Du mal bitte as ModBus Log vom Auslesen der Leistung teilen- entweder aus evcc (log: trace) oder per mbmd? Ich würde gerne verstehen, was da physikalisch über den Draht geht.

Details sollten wir dann in einem Ticket diskutieren

_Originally posted by @andig in https://github.com/andig/evcc/discussions/1219#discussioncomment-1060414_

premultiply, 2021-07-27:
Laut Schnittstellenbeschreibung sollte in den Registern eigentlich der schon normierte Wert als 32 bit LE integer stehen.

andig, 2021-07-27:
/cc @lorbetzki das müssten wir uns mal im Logfile anschauen

premultiply, 2021-07-27:
Aus dem Phoenix-Manual:

> Sie können mehrere Input- und Holding-Register zusammenfassen, um 32-Bit-Daten 
> zu übertragen. Die Kodierung für solche Daten ist im Little-Endian-Format. Das Wort 
> mit dem niederwertigsten Element wird zuerst genannt.

lorbetzki, 2021-07-27:
Hi,

here we go:

Ergebnis mit auskommentierter meter in loadpoints

```
root@lokiv4:/usr/src# evcc --log trace
[main  ] INFO 2021/07/27 10:37:42 evcc 0.60 (986d734)
[main  ] INFO 2021/07/27 10:37:42 using config file /etc/evcc.yaml
[main  ] INFO 2021/07/27 10:37:42 listening at 0.0.0.0:7070
[mqtt  ] INFO 2021/07/27 10:37:42 connecting evcc-410836550 at localhost:1883
[mqtt  ] DEBUG 2021/07/27 10:37:42 localhost:1883 connected
[lp-1  ] WARN 2021/07/27 10:37:42 poll mode '{always 1m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2021/07/27 10:37:42 poll interval '1m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[cache ] TRACE 2021/07/27 10:37:42 title: Garage
[cache ] TRACE 2021/07/27 10:37:42 minCurrent: 6
[cache ] TRACE 2021/07/27 10:37:42 maxCurrent: 20
[cache ] TRACE 2021/07/27 10:37:42 phases: 1
[cache ] TRACE 2021/07/27 10:37:42 activePhases: 1
[cache ] TRACE 2021/07/27 10:37:42 hasVehicle: true
[cache ] TRACE 2021/07/27 10:37:42 mode:
[cache ] TRACE 2021/07/27 10:37:42 targetSoC: 100
[cache ] TRACE 2021/07/27 10:37:42 minSoC: 0
[cache ] TRACE 2021/07/27 10:37:42 socTitle: Zoe
[cache ] TRACE 2021/07/27 10:37:42 socCapacity: 52
[wallbe] TRACE 2021/07/27 10:37:42 modbus: send 00 01 00 00 00 06 ff 01 01 90 00 01
[wallbe] TRACE 2021/07/27 10:37:42 modbus: recv 00 01 00 00 00 04 ff 01 01 01
[lp-1  ] DEBUG 2021/07/27 10:37:42 max charge current: 6
[wallbe] TRACE 2021/07/27 10:37:42 modbus: send 00 02 00 00 00 06 ff 06 02 10 00 3c
[wallbe] TRACE 2021/07/27 10:37:42 modbus: recv 00 02 00 00 00 06 ff 06 02 10 00 3c
[site  ] INFO 2021/07/27 10:37:42 site config:
[site  ] INFO 2021/07/27 10:37:42   meters:    grid ? pv ? battery ?
[site  ] INFO 2021/07/27 10:37:42     grid:    power ? energy ? currents ?
[cache ] TRACE 2021/07/27 10:37:42 siteTitle: Zuhause
[cache ] TRACE 2021/07/27 10:37:42 chargeCurrent: 6
[cache ] TRACE 2021/07/27 10:37:42 gridConfigured: true
[cache ] TRACE 2021/07/27 10:37:42 pvConfigured: true
[site  ] INFO 2021/07/27 10:37:42     pv:      power ? energy ? currents ?
[cache ] TRACE 2021/07/27 10:37:42 batteryConfigured: true
[site  ] INFO 2021/07/27 10:37:42     battery: power ? energy ? currents ? soc ?
[lp-1  ] INFO 2021/07/27 10:37:42 loadpoint 1:
[lp-1  ] INFO 2021/07/27 10:37:42   mode:
[lp-1  ] INFO 2021/07/27 10:37:42   charger:   power ? energy ? currents ? timer ?
[lp-1  ] INFO 2021/07/27 10:37:42   meters:    charge ?
[lp-1  ] INFO 2021/07/27 10:37:42     charge:  power ? energy ? currents ?
[lp-1  ] INFO 2021/07/27 10:37:42   vehicles:  ?
[lp-1  ] INFO 2021/07/27 10:37:42     car 0:   range ? finish ? status ? climate ?
[cache ] TRACE 2021/07/27 10:37:42 prioritySoC: 80
[cache ] TRACE 2021/07/27 10:37:42 chargeConfigured: true
[site  ] DEBUG 2021/07/27 10:37:42 ----
[script] TRACE 2021/07/27 10:37:42 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=PV: 2283
[site  ] DEBUG 2021/07/27 10:37:42 pv power: 2283W
[cache ] TRACE 2021/07/27 10:37:42 pvPower: 2283
[script] TRACE 2021/07/27 10:37:42 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=gridPower: -45
[site  ] DEBUG 2021/07/27 10:37:42 grid power: -45W
[cache ] TRACE 2021/07/27 10:37:42 gridPower: -45
[script] TRACE 2021/07/27 10:37:42 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batteryPower: 1752
[site  ] DEBUG 2021/07/27 10:37:42 battery power: 1752W
[cache ] TRACE 2021/07/27 10:37:42 batteryPower: 1752
[script] TRACE 2021/07/27 10:37:42 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batterySOC: 46
[site  ] DEBUG 2021/07/27 10:37:42 battery soc: 46%
[site  ] DEBUG 2021/07/27 10:37:42 site power: 1707W
[wallbe] TRACE 2021/07/27 10:37:42 modbus: send 00 03 00 00 00 06 ff 04 00 78 00 02
[cache ] TRACE 2021/07/27 10:37:42 batterySoC: 46
[cache ] TRACE 2021/07/27 10:37:42 mode:
[wallbe] TRACE 2021/07/27 10:37:42 modbus: recv 00 03 00 00 00 07 ff 04 04 04 73 00 00
[lp-1  ] DEBUG 2021/07/27 10:37:42 charge power: 74645504W
[wallbe] TRACE 2021/07/27 10:37:42 modbus: send 00 04 00 00 00 06 ff 04 00 72 00 02
[cache ] TRACE 2021/07/27 10:37:42 chargePower: 7.4645504e+07
[... 468 lines cut ...]
in der evcc.yaml habe ich unter meter zusätzlich
```
- name: lademenge
  type: custom
  power:
    source: modbus
    uri: 192.168.147.4:502
    id: 255
    register:
      address: 120
      type: holding
      decode: int32s
```
angelegt und in loadpoints unter meter angegeben.

```
loadpoints:
- title: Garage # display name for UI
  charger: wallbe # charger
  vehicle: default
  meters:
    charge: lademenge
```

Das problem exisiterit mit register 120 (Lademenge) und 128 (geladene menge).

andig, 2021-07-27:
Ich denke Power ist das hier (Reg. 120=0x78):

    10:37:42 modbus: send 00 03 00 00 00 06 ff 04 00 78 00 02
    10:37:42 modbus: recv 00 03 00 00 00 07 ff 04 04 04 73 00 00

Muss man lesen als

    04 73 00 00

Decoding: https://play.golang.org/p/cc6pO1_vvta entspricht int32 **swapped** --> 1139

Aktuell machen wir bei der Wallbe das "swapped" nicht!

> Das problem exisiterit mit register 120 (Lademenge) und 128 (geladene menge).

Was mir überhaupt nicht in den Kopf will ist warum das nur bei 2 Registern ein Problem sein soll?

andig, 2021-07-27:
@lorbetzki kannst Du in der `wallbe.go` mal bitte folgende Änderung machen:

	func (wb *Wallbe) decodeReading(b []byte) float64 {
		switch wb.encoding {
		case encodingSDM:
			// assuming high register first
			bits := uint32(b[3])<<0 | uint32(b[2])<<8 | uint32(b[1])<<16 | uint32(b[0])<<24
			f := math.Float32frombits(bits)
			return float64(f)

		default:
			return rs485.RTUUint32ToFloat64Swapped(b)
		}
	}

geht es dann mit der "normalen" Konfiguration?

premultiply, 2021-07-27:
Ist `return rs485.RTUUint32ToFloat64Swapped(b)` nicht uint32s? Müsste dann eher int32s sein, oder?
Die Werte können auch negativ werden. Siehe Screenshot.

lorbetzki, 2021-07-27:
ich habe die Änderung durchgeführt, musste aber noch die Zeile
durch ersetzen, da ich sonst nicht kompilieren konnte.

```
[cache ] TRACE 2021/07/27 12:50:33 chargedEnergy: 1000
[cache ] TRACE 2021/07/27 12:50:33 chargeDuration: 6m38s
[wallbe] TRACE 2021/07/27 12:50:33 modbus: recv 01 2a 00 00 00 05 ff 04 02 00 43
[cache ] TRACE 2021/07/27 12:50:33 connected: true
[cache ] TRACE 2021/07/27 12:50:33 charging: true
[cache ] TRACE 2021/07/27 12:50:33 enabled: true
[cache ] TRACE 2021/07/27 12:50:33 socCharge: 71
[cache ] TRACE 2021/07/27 12:50:33 chargeEstimate: 4h5m37s
[wallbe] TRACE 2021/07/27 12:50:33 modbus: send 01 2b 00 00 00 06 ff 01 01 90 00 01
[cache ] TRACE 2021/07/27 12:50:33 chargeRemainingEnergy: 16755.555555555555
[wallbe] TRACE 2021/07/27 12:50:33 modbus: recv 01 2b 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/07/27 12:50:38 modbus: closing connection due to idle timeout: 5.000777268s
[site  ] DEBUG 2021/07/27 12:50:43 ----
[script] TRACE 2021/07/27 12:50:43 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=PV: 2712
[site  ] DEBUG 2021/07/27 12:50:43 pv power: 2712W
[cache ] TRACE 2021/07/27 12:50:43 pvPower: 2712
[script] TRACE 2021/07/27 12:50:43 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=gridPower: 27
[site  ] DEBUG 2021/07/27 12:50:43 grid power: 27W
[cache ] TRACE 2021/07/27 12:50:43 gridPower: 27
[script] TRACE 2021/07/27 12:50:43 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batteryPower: 2379
[site  ] DEBUG 2021/07/27 12:50:43 battery power: 2379W
[cache ] TRACE 2021/07/27 12:50:43 batteryPower: 2379
[script] TRACE 2021/07/27 12:50:43 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batterySOC: 57
[site  ] DEBUG 2021/07/27 12:50:43 battery soc: 57%
[site  ] DEBUG 2021/07/27 12:50:43 site power: 2406W
[cache ] TRACE 2021/07/27 12:50:43 batterySoC: 57
[cache ] TRACE 2021/07/27 12:50:43 mode: now
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 2c 00 00 00 06 ff 04 00 78 00 02
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 2c 00 00 00 07 ff 04 04 0f fd 00 00
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 2d 00 00 00 06 ff 04 00 72 00 02
[cache ] TRACE 2021/07/27 12:50:43 chargePower: 4093
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 2d 00 00 00 07 ff 04 04 00 13 00 00
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 2e 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 2e 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 2f 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 2f 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 30 00 00 00 06 ff 04 00 80 00 02
[cache ] TRACE 2021/07/27 12:50:43 chargeCurrents: [19 0 0]
[cache ] TRACE 2021/07/27 12:50:43 activePhases: 1
[cache ] TRACE 2021/07/27 12:50:43 chargeCurrent: 20
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 30 00 00 00 07 ff 04 04 06 25 00 00
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 31 00 00 00 06 ff 04 00 66 00 02
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 31 00 00 00 07 ff 04 04 01 98 00 00
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 32 00 00 00 06 ff 04 00 64 00 01
[cache ] TRACE 2021/07/27 12:50:43 chargedEnergy: 1000
[cache ] TRACE 2021/07/27 12:50:43 chargeDuration: 6m48s
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 32 00 00 00 05 ff 04 02 00 43
[cache ] TRACE 2021/07/27 12:50:43 connected: true
[cache ] TRACE 2021/07/27 12:50:43 charging: true
[cache ] TRACE 2021/07/27 12:50:43 enabled: true
[cache ] TRACE 2021/07/27 12:50:43 socCharge: 71
[cache ] TRACE 2021/07/27 12:50:43 chargeEstimate: 4h5m37s
[wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 33 00 00 00 06 ff 01 01 90 00 01
[cache ] TRACE 2021/07/27 12:50:43 chargeRemainingEnergy: 16755.555555555555
[wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 33 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/07/27 12:50:48 modbus: closing connection due to idle timeout: 5.000409126s
![2021-07-27 12_51_25-Configuration](https://user-images.githubusercontent.com/57233317/127142182-4763ab3d-ecaf-45cd-893e-f189d248a28f.png)

```
Die Anzeige im evcc Webif entspricht in etwa die des Wallbe Webif. Register 128 wird aber scheinbar nicht ausgewertet, bleibt bei evcc auf 0

[image]

Ich habe auf der Wallbe die Firmware Version SL-01.04.13, diese wurde mir von E3DC vor ca 2 Monaten installiert. Vielleicht liegt es daran?

andig, 2021-07-27:
> Die Anzeige im evcc Webif entspricht in etwa die des Wallbe Webif. Register 128 wird aber scheinbar nicht ausgewertet, bleibt bei evcc auf 0

Was meinst Du damit? Die geladene Energie? Wir ausgelesen wenn Du sie per Config aktiviert hast (`energy: true`).

andig, 2021-07-27:
> Ist `return rs485.RTUUint32ToFloat64Swapped(b)` nicht uint32s? Müsste dann eher int32s sein, oder?
> Die Werte können auch negativ werden. Siehe Screenshot.

Dann brauchen wir 2 Funktionen je nachdem was wir auslesen (Energie: uint, Leistung int). Alle Werte die wir auslesen müssten sollten egtl positiv sein?

premultiply, 2021-07-27:
Ja, trotzdem sind das wohl eigentlich signed int Werte.
Solange nur 31 bit und der positive Zahlenbereich genutzt wird fällt das ja nicht auf.

andig, 2021-07-27:
> Ja, trotzdem sind das wohl eigentlich signed int Werte.

Jetzt nochmal. Alle??? Sonst brauchen wir 2 Funktionen. Welche sind Deiner Meinung nach signed?

premultiply, 2021-07-27:
Ich denke ALLE Rückgabewerte sind int32s.

lorbetzki, 2021-07-27:
aus meinem Laienhaften verständnis würde ich sagen es geht nur um die beiden Register 120/128, und die wären auch nur positiv. Alles andere scheint soweit zu stimmen.

> 
> 
> > Die Anzeige im evcc Webif entspricht in etwa die des Wallbe Webif.
> 
> Nice, tut also erstmal 👍🏻
> 
> > Register 128 wird aber scheinbar nicht ausgewertet, bleibt bei evcc auf 0
> 
> Was meinst Du damit? Die geladene Energie? Wir ausgelesen wenn Du sie per Config aktiviert hast (`energy: true`).

ist bei mir aktiv. Ich füge mal die komplette evcc.yaml an, die derzeit bei mir aktiv ist und der Register  120 auch korrekt angezeigt wird.
```
root@lokiv4:~# cat /etc/evcc.yaml
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: error
  lp-1: error
  lp-2: error

meters:
- name: grid
  type: default
  power:
    type: script
    cmd: /bin/sh -c "/usr/bin/curl http://loki.lorbi.local/hook/evcc?state=gridPower" # actual command
    timeout: 3s # kill script after 3 seconds
- name: pv
  type: default
  power:
    type: script
    cmd: /bin/sh -c "/usr/bin/curl http://loki.lorbi.local/hook/evcc?state=PV" # actual command
    timeout: 3s # kill script after 3 seconds
- name: battery
  type: default
  power:
    type: script
    cmd: /bin/sh -c "/usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batteryPower" # actual command
    timeout: 3s # kill script after 3 seconds
  soc:
    type: script
    cmd: /bin/sh -c "/usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batterySOC" # actual command
    timeout: 3s # kill script after 3 seconds
- name: lademenge
  type: custom
  power:
    source: modbus
    uri: 192.168.147.4:502
    id: 255
    register:
      address: 120
      type: holding
      decode: int32s

chargers:
- name: wallbe
  type: wallbe # Wallbe charger
  uri: 192.168.147.4:502 # ModBus address
  meter:
    power: true
    energy: true
    currents: true

vehicles:
- name: default
  type: default
  title: Zoe
  capacity: 52 # kWh
  charge:
    type: script # use script plugin
    cmd: /bin/sh -c "/usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batteryCar" # actual command
    timeout: 3s # kill script aft
  cache: 5m

site:
  title: Zuhause # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter
    battery: battery # battery meter
  prioritySoC: 80 # give home battery priority up to this soc (0 to disable)

loadpoints:
- title: Garage # display name for UI
  charger: wallbe # charger
  vehicle: default
  meters:
    charge: lademenge
  mode: off
  soc:
    poll:
      mode: always
      interval: 1m
    min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 100 # always charge to 100%
    estimate: false # set true to interpolate between api updates
  onDisconnect: # set defaults when vehicle disconnects
    mode: off # switch back to pv mode
  phases: 1 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 20 # maximum charge current (default 16A)

# mqtt message broker
mqtt:
   broker: localhost:1883
   topic: evcc # root topic for publishing, set empty to disable
   user: mqtt

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

andig, 2021-07-27:
> aus meinem Laienhaften verständnis würde ich sagen es geht nur um die beiden Register 120/128, und die wären auch nur positiv. Alles andere scheint soweit zu stimmen.

Also ausgelesen wird die 128:

    [wallbe] TRACE 2021/07/27 12:50:43 modbus: send 01 30 00 00 00 06 ff 04 00 80 00 02
    [wallbe] TRACE 2021/07/27 12:50:43 modbus: recv 01 30 00 00 00 07 ff 04 04 06 25 00 00
    [cache ] TRACE 2021/07/27 12:50:43 chargedEnergy: 1000

Macht:

    06 26 -> 1573

Stimmt!

Mach mal bitte:

    evcc charger --log trace

Prinzipiell ist das Reg. 128 ziemlich hirntot weil es eine Auflösung von nur 1kWh hat. Liegts daran? Eigentlich müssten wir stattdessen das andere Register nehmen- ich vermute, das ist die 3376 mit sagenhaften 128bit?

premultiply, 2021-07-27:
904 steht sowohl in der Wallbe als auch in der Phoenix-Doku.

Bei Phoenix steht sogar bei 128 noch dabei:

> Integer [kWh]
Anzeige in [Wh]: siehe 
Register 904 (Holding)

andig, 2021-07-27:
Bei Bei mir sieht das so aus (Wallbe Modbus Register 5.0):

[image]
[image]

16bit  nutzen jedenfalls nichts, es sei denn das wären die unteren 16bit der kWh, also die Nachkommastellen???

@lorbetzki könntest Du nochmal das hier probieren:

```
// totalEnergy implements the api.MeterEnergy interface
func (wb *Wallbe) totalEnergy() (float64, error) {
	b, err := wb.conn.ReadInputRegisters(wbRegEnergy, 2)
	if err != nil {
		return 0, err
	}

	d, err := wb.conn.ReadHoldingRegisters(904, 1)
	if err != nil {
		return 0, err
	}

	res := wb.decodeReading(b) + float64(binary.BigEndian.Uint16(d))/1e3

	return res, nil
}
```

/cc @goebelmeier wäre auch spannend was da mit Deinem SDM passiert...

lorbetzki, 2021-07-27:
Hier mit wallbe
```
root@lokiv4:~# /tmp/evcc-new charger --log trace
[main  ] INFO 2021/07/27 20:39:20 evcc  ()
[main  ] INFO 2021/07/27 20:39:20 using config file /etc/evcc.yaml
[mqtt  ] INFO 2021/07/27 20:39:20 connecting evcc-1068120374 at localhost:1883
[mqtt  ] DEBUG 2021/07/27 20:39:20 localhost:1883 connected
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 01 00 00 00 06 ff 04 00 78 00 02
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 02 00 00 00 06 ff 04 00 80 00 02
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 02 00 00 00 07 ff 04 04 06 25 00 00
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 03 00 00 00 06 ff 04 00 72 00 02
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 03 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 04 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 04 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 05 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 05 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 06 00 00 00 06 ff 04 00 64 00 01
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 06 00 00 00 05 ff 04 02 00 42
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 07 00 00 00 06 ff 01 01 90 00 01
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 07 00 00 00 04 ff 01 01 00
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 08 00 00 00 06 ff 04 00 66 00 02
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 08 00 00 00 07 ff 04 04 00 48 00 00
Power:          0W
Energy:         1573.0kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Duration:       1m12s
Diagnostic dump:
[wallbe] TRACE 2021/07/27 20:39:20 modbus: send 00 09 00 00 00 06 ff 04 00 95 00 06
[wallbe] TRACE 2021/07/27 20:39:20 modbus: recv 00 09 00 00 00 0f ff 04 0c 4c 53 30 2d 2e 31 34 30 31 2e 00 33
Firmware:       SL-01.04.13
```
und hier mit phoenix-ev-eth

```
root@lokiv4:~# /tmp/evcc-new charger --log trace
[main  ] INFO 2021/07/27 20:45:05 evcc  ()
[main  ] INFO 2021/07/27 20:45:05 using config file /etc/evcc.yaml
[mqtt  ] INFO 2021/07/27 20:45:05 connecting evcc-473981400 at localhost:1883
[mqtt  ] DEBUG 2021/07/27 20:45:05 localhost:1883 connected
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 01 00 00 00 06 ff 04 00 78 00 02
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 00
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 02 00 00 00 06 ff 03 03 88 00 02
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 02 00 00 00 07 ff 03 04 01 22 00 00
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 03 00 00 00 06 ff 04 00 72 00 02
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 03 00 00 00 07 ff 04 04 00 00 00 00
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 04 00 00 00 06 ff 04 00 74 00 02
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 04 00 00 00 07 ff 04 04 00 00 00 00
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 05 00 00 00 06 ff 04 00 76 00 02
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 05 00 00 00 07 ff 04 04 00 00 00 00
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 06 00 00 00 06 ff 04 00 64 00 01
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 06 00 00 00 05 ff 04 02 00 42
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 07 00 00 00 06 ff 01 01 90 00 01
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 07 00 00 00 04 ff 01 01 01
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: send 00 08 00 00 00 06 ff 04 00 66 00 02
[ev-eth] TRACE 2021/07/27 20:45:05 modbus: recv 00 08 00 00 00 07 ff 04 04 00 48 00 00
Power:          0W
Energy:         0.3kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        true
Duration:       1m12s
```
ABER, mit der phoenix-ev-eth bekomme ich die Ladung nicht gestartet.

Ich habe mal versucht in der wallbe.go das Register 128 durch 904 zu ersetzen
```
root@lokiv4:~# /tmp/evcc-new charger --log trace
[... 524 lines cut ...]
[wallbe] TRACE 2021/07/27 20:35:30 modbus: send 00 55 00 00 00 06 ff 04 03 88 00 02
[cache ] TRACE 2021/07/27 20:35:30 chargeCurrents: [19 0 0]
[cache ] TRACE 2021/07/27 20:35:30 activePhases: 1
[cache ] TRACE 2021/07/27 20:35:30 chargeCurrent: 20
[wallbe] TRACE 2021/07/27 20:35:30 modbus: recv 00 55 00 00 00 07 ff 04 04 01 20 00 00
[wallbe] TRACE 2021/07/27 20:35:30 modbus: send 00 56 00 00 00 06 ff 04 00 66 00 02
[wallbe] TRACE 2021/07/27 20:35:30 modbus: recv 00 56 00 00 00 07 ff 04 04 00 47 00 00
[wallbe] TRACE 2021/07/27 20:35:30 modbus: send 00 57 00 00 00 06 ff 04 00 64 00 01
[cache ] TRACE 2021/07/27 20:35:30 chargedEnergy: 68000
[cache ] TRACE 2021/07/27 20:35:30 chargeDuration: 1m11s
[wallbe] TRACE 2021/07/27 20:35:30 modbus: recv 00 57 00 00 00 05 ff 04 02 00 43
[cache ] TRACE 2021/07/27 20:35:30 connected: true
[cache ] TRACE 2021/07/27 20:35:30 charging: true
[cache ] TRACE 2021/07/27 20:35:30 enabled: true
[cache ] TRACE 2021/07/27 20:35:30 socCharge: 68
[cache ] TRACE 2021/07/27 20:35:30 chargeEstimate: 4h28m44s
[wallbe] TRACE 2021/07/27 20:35:30 modbus: send 00 58 00 00 00 06 ff 01 01 90 00 01
[cache ] TRACE 2021/07/27 20:35:30 chargeRemainingEnergy: 18488.888888888887
[wallbe] TRACE 2021/07/27 20:35:30 modbus: recv 00 58 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/07/27 20:35:30 modbus: send 00 59 00 00 00 06 ff 05 01 90 00 00
[wallbe] TRACE 2021/07/27 20:35:30 modbus: recv 00 59 00 00 00 06 ff 05 01 90 00 00
[cache ] TRACE 2021/07/27 20:35:30 chargeCurrent: 0
```

[image]

lorbetzki, 2021-07-27:
@andig 
> @lorbetzki könntest Du nochmal das hier probieren:
[image]

[image]

```
root@lokiv4:~# /tmp/evcc-new charger --log trace
[main  ] INFO 2021/07/27 20:57:22 evcc  ()
[main  ] INFO 2021/07/27 20:57:22 using config file /etc/evcc.yaml
[mqtt  ] INFO 2021/07/27 20:57:22 connecting evcc-1016389013 at localhost:1883
[mqtt  ] DEBUG 2021/07/27 20:57:22 localhost:1883 connected
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 01 00 00 00 06 ff 04 00 78 00 02
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 01 00 00 00 07 ff 04 04 10 41 00 00
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 02 00 00 00 06 ff 04 00 80 00 02
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 02 00 00 00 07 ff 04 04 06 25 00 00
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 03 00 00 00 06 ff 03 03 88 00 01
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 03 00 00 00 05 ff 03 02 01 c9
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 04 00 00 00 06 ff 04 00 72 00 02
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 04 00 00 00 07 ff 04 04 00 13 00 00
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 05 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 05 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 06 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 06 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 07 00 00 00 06 ff 04 00 64 00 01
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 07 00 00 00 05 ff 04 02 00 43
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 08 00 00 00 06 ff 01 01 90 00 01
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 08 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 09 00 00 00 06 ff 04 00 66 00 02
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 09 00 00 00 07 ff 04 04 00 a1 00 00
Power:          4161W
Energy:         1573.5kWh
Current L1..L3: 19A 0A 0A
Charge status:  C
Enabled:        true
Duration:       2m41s
Diagnostic dump:
[wallbe] TRACE 2021/07/27 20:57:22 modbus: send 00 0a 00 00 00 06 ff 04 00 95 00 06
[wallbe] TRACE 2021/07/27 20:57:22 modbus: recv 00 0a 00 00 00 0f ff 04 0c 4c 53 30 2d 2e 31 34 30 31 2e 00 33
Firmware:       SL-01.04.13
```

```
[site  ] DEBUG 2021/07/27 20:55:58 grid power: 740W
[cache ] TRACE 2021/07/27 20:55:58 gridPower: 740
[script] TRACE 2021/07/27 20:55:58 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batteryPower: 4683
[site  ] DEBUG 2021/07/27 20:55:58 battery power: 4683W
[cache ] TRACE 2021/07/27 20:55:58 batteryPower: 4683
[script] TRACE 2021/07/27 20:55:58 /bin/sh -c /usr/bin/curl http://loki.lorbi.local/hook/evcc?state=batterySOC: 63
[site  ] DEBUG 2021/07/27 20:55:58 battery soc: 63%
[site  ] DEBUG 2021/07/27 20:55:58 site power: 5423W
[cache ] TRACE 2021/07/27 20:55:58 batterySoC: 63
[cache ] TRACE 2021/07/27 20:55:58 mode: now
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 53 00 00 00 06 ff 04 00 78 00 02
[wallbe] TRACE 2021/07/27 20:55:58 modbus: recv 00 53 00 00 00 07 ff 04 04 10 2c 00 00
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 54 00 00 00 06 ff 04 00 72 00 02
[cache ] TRACE 2021/07/27 20:55:58 chargePower: 4140
[wallbe] TRACE 2021/07/27 20:55:58 modbus: recv 00 54 00 00 00 07 ff 04 04 00 13 00 00
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 55 00 00 00 06 ff 04 00 74 00 02
[wallbe] TRACE 2021/07/27 20:55:58 modbus: recv 00 55 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 56 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/07/27 20:55:58 modbus: recv 00 56 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 57 00 00 00 06 ff 04 00 80 00 02
[cache ] TRACE 2021/07/27 20:55:58 chargeCurrents: [19 0 0]
[cache ] TRACE 2021/07/27 20:55:58 activePhases: 1
[cache ] TRACE 2021/07/27 20:55:58 chargeCurrent: 20
[wallbe] TRACE 2021/07/27 20:55:58 modbus: recv 00 57 00 00 00 07 ff 04 04 06 25 00 00
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 58 00 00 00 06 ff 03 03 88 00 01
[wallbe] TRACE 2021/07/27 20:55:58 modbus: recv 00 58 00 00 00 05 ff 03 02 01 68
[wallbe] TRACE 2021/07/27 20:55:58 modbus: send 00 59 00 00 00 06 ff 04 00 66 00 02
[... 36 lines cut ...]
[wallbe] TRACE 2021/07/27 20:56:08 modbus: send 00 5f 00 00 00 06 ff 04 00 76 00 02
[wallbe] TRACE 2021/07/27 20:56:08 modbus: recv 00 5f 00 00 00 07 ff 04 04 00 00 00 00
[wallbe] TRACE 2021/07/27 20:56:08 modbus: send 00 60 00 00 00 06 ff 04 00 80 00 02
[cache ] TRACE 2021/07/27 20:56:08 chargeCurrents: [19 0 0]
[cache ] TRACE 2021/07/27 20:56:08 activePhases: 1
[cache ] TRACE 2021/07/27 20:56:08 chargeCurrent: 20
[wallbe] TRACE 2021/07/27 20:56:08 modbus: recv 00 60 00 00 00 07 ff 04 04 06 25 00 00
[wallbe] TRACE 2021/07/27 20:56:08 modbus: send 00 61 00 00 00 06 ff 03 03 88 00 01
[wallbe] TRACE 2021/07/27 20:56:08 modbus: recv 00 61 00 00 00 05 ff 03 02 01 74
[wallbe] TRACE 2021/07/27 20:56:08 modbus: send 00 62 00 00 00 06 ff 04 00 66 00 02
[wallbe] TRACE 2021/07/27 20:56:08 modbus: recv 00 62 00 00 00 07 ff 04 04 00 57 00 00
[wallbe] TRACE 2021/07/27 20:56:08 modbus: send 00 63 00 00 00 06 ff 04 00 64 00 01
[cache ] TRACE 2021/07/27 20:56:08 chargedEnergy: 82.00000000010732
[cache ] TRACE 2021/07/27 20:56:08 chargeDuration: 1m27s
[wallbe] TRACE 2021/07/27 20:56:08 modbus: recv 00 63 00 00 00 05 ff 04 02 00 43
[cache ] TRACE 2021/07/27 20:56:08 connected: true
[... cut ...] rue
[cache ] TRACE 2021/07/27 20:56:08 enabled: true
[cache ] TRACE 2021/07/27 20:56:08 socCharge: 68
[cache ] TRACE 2021/07/27 20:56:08 chargeEstimate: 4h27m11s
[wallbe] TRACE 2021/07/27 20:56:08 modbus: send 00 64 00 00 00 06 ff 01 01 90 00 01
[cache ] TRACE 2021/07/27 20:56:08 chargeRemainingEnergy: 18488.888888888887
[wallbe] TRACE 2021/07/27 20:56:08 modbus: recv 00 64 00 00 00 04 ff 01 01 01
[wallbe] TRACE 2021/07/27 20:56:13 modbus: closing connection due to idle timeout: 5.000993558s
```

andig, 2021-07-28:
> sieht gut aus.

Supi :)

@premultiply das scheint tatsächliche abweichend als bei Phoenix zu sein:

[image]

closed 2021-07-28 by andig

premultiply, 2021-07-28:
Woraus schließt du das?
Und auch hier Integer, nicht unsigned Integer.