evcc-io/evcc #1250: SMA Home Manager 2 Konfigurationsprobleme

Terrible-Toaster, 2021-07-28:
Hallo,
Ich habe hier folgende Konfiguration:
1. SMH 2.0
2. 2x SunnyBoy Inverter
3. SunnyBoy Storage
4. Wallbe Eco2.0s

habe evcc auf einem Raspberry Pi laufen.
Nun habe ich Probleme beim konfigurieren des SMH2.0.
Beim ausführen von evcc detect bekomme ich nur die inverter, das Storage Modul sowie die Wallbox angezeigt.
Wallbox Konfiguration über „evcc charger -l debug“ liefert auch die korrekte Antwort .
Jedoch gibt es Probleme beim SMH.
Habe jetzt, wie in der Wiki beschrieben, den SMH folgender Maßen angelegt:

meters:
- name: sma-grid
  type: sma
  serial: 301147xxx 
  uri: 192.168.178.100

Bei Verwendung von Seriennummer, bekomme ich „Device Not found“. Verwende ich uri kriege ich „no Ping response from 192.168.178.100“
Manuelles pingen von 192.168.178.100 funktioniert einwandfrei und laut DHCP stimmt die IP Adresse auch.
Wo könnte hier der Fehler liegen?

andig, 2021-07-28:
/cc @bboehmke @premultiply

bboehmke, 2021-07-28:
Wenn die URI angegeben wird kann auf `serial` setting verzichtet werden. (die wird eh ignoriert in diesem fall).

Befinden sich der Raspberry im gleichen Subnetz wie der SHM? Wenn nicht könnte es Probleme mit der Multicast Kommunikation geben.
Laut SMA können auch bestimmte Switche Probleme machen.

Im Zweifel mal `evcc --log trace meter` ausführen und den log hier posten (eventuell die Seriennummern entfernen).

andig, 2021-07-28:
Über interface kann noch die „korrekte“ Netzwerkschnittstelle konfiguriert werden

Terrible-Toaster, 2021-07-28:
Vielen Dank @bboehmke 
Der Switch wars.
Aus bequemlichkeitsgründen hatte ich den raspi im Büro angeschlossen. Die Route zum SMH war dann:
Raspi -> D-Link Switch -> FritzBox -> D-Link Switch (nur für PV) -> SMH
Entweder der erste D-Link Switch oder die Fritzbox war das Problem. Vermutlich eher die Fritzbox, da der Switch Dumm ist und nicht einschränken sollte.
Jetzt hängt der Raspi direkt am Switch der für die PV ist und über evcc detect hat er direkt alle SMA Geräte gefunden!
Also, am besten keine Fritzbox dazwischen haben.

closed 2021-07-28 by Terrible-Toaster