evcc-io/evcc #1251: detection of loadpoint's second car fails after connection

Caibuk, 2021-07-30:
**Describe the bug**
I have a loadpoint with two configured cars.  Both have perfectly working online API access with indication of connected and charging states (Skoda Enyaq using `type: enyaq` as first in vehicles list and VW e-up using `type: id` as second).

_Prerequisites:_
- loadpoint is "idle" (no car connected) with the first car (Enyaq) assigned
- second car (e-up) gets plugged in

_(faulty) behaviour:_
Even though the second car might start charging, the loadpoint sticks to the first car.
Probable cause seems to me, that the second car gets polled too early and only once after getting plugged in (before the connected state  gets updated in VW's backend). I think this behaviour started with #1163 .

 _known workaround:_
Restart of evcc container while the second car is connected leads to detection of the correct car.

**To Reproduce**
Steps to reproduce the behavior:
1. configure evcc with a loadpoint with two vehicles
2. start evcc without a car connected (so that the first car in vehicles list gets assigned to the loadpoint)
3. plug in the second car in vehicles list
4. second car won't get detected and loadpoint sticks to the first car

**Expected behavior**
After connection of the second car in vehicles list (plus `cache`) this car gets detected and shown in GUI.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.60 (986d734)
```

<details><summary>Show output of <code>evcc dump -c configfile</code>:<br>
This is after restart of evcc and hence with correctly detected second car!</summary><br><pre>
[main  ] INFO 2021/07/30 10:25:14 evcc 0.60 (986d734)
[main  ] INFO 2021/07/30 10:25:14 using config file /config/evcc.yaml
[mqtt  ] INFO 2021/07/30 10:25:14 connecting evcc-668543700 at 10.0.1.97:1883
[mqtt  ] DEBUG 2021/07/30 10:25:14 10.0.1.97:1883 connected
[lp-1  ] WARN 2021/07/30 10:25:19 poll mode '{always 2m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2021/07/30 10:25:19 poll interval '2m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
config
======

grid: grid
----------
Power:  0W
Energy: 5509551.0kWh

pv: pv-dc-kostal
----------------
Power: 6978W

battery: battery-pv
-------------------
Power: -2371W
SoC:   35%

loadpoint 1
===========

charge: charge
--------------
Power:  3808W
Energy: 7688.2kWh

charger: wallbe
---------------
Charge status: C
Enabled:       true
Duration:      1h26m52s
Diagnostic dump:

vehicle 0
---------
SoC:           70%
Charge status: A
Capacity:      58kWh
Range:         236km
Finish time:   not available

vehicle 1
---------
SoC:            45%
Charge status:  C
Capacity:       32kWh
Range:          101km
Finish time:    2021-07-30 12:29:00 +0000 UTC
Climate active: false
Outside temp:   21.0°C
Target temp:    21.0°C
</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: debug
levels:
  #core: debug
  lp-1: debug
  #lp-2: debug
  #cache: trace
  #vw: trace
  id: debug
  enyaq: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
  - name: grid
    type: modbus
    model: kostal
    uri: 10.0.1.57:502
    id: 255
    power: Power
    energy: Export
  - name: pv-dc-kostal
    type: custom
    power:
      source: calc
      add: # The add plugin sums up both string values
        - source: modbus
          model: kostal
          value: 160:1:DCW # string 1
          uri: 10.0.1.37:1502
          id: 71 # Configured Modbus Device ID
        - source: modbus
          model: kostal
          value: 160:2:DCW # string 2
          uri: 10.0.1.37:1502
          id: 71 # Configured Modbus Device ID
  - name: battery-pv
    type: modbus
    power: 802:W
    soc: 802:SoC
    uri: 10.0.1.37:1502
    id: 71
  - name: charge
    type: custom
    power:
      source: mqtt
      topic: mbmd/sdm2301-10/Power
      timeout: 30s
    energy:
      source: mqtt
      topic: mbmd/sdm2301-10/Import
      timeout: 30s

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
  - name: wallbe
    type: wallbe # Wallbe charger
    uri: 10.0.1.67:502 # ModBus address
    legacy: true

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
  - name: e-up
    type: id
    title: Uppsala
    capacity: 32.3
    user: xxxxxxxxxxxxxxxx@googlemail.com # user
    password: xxxxxxxxxxxxxxxx # password
    vin: WVWxxxxxxxxxxxxxx
    cache: 2m
  - name: enyaq
    type: enyaq
    title: Ladmila
    capacity: 58
    user: xxxxxxxxxxxxxxxx@googlemail.com # user
    password: xxxxxxxxxxxxxxxx # password
    vin: TMBxxxxxxxxxxxxxx
    cache: 2m

# site describes the EVU connection, PV and home battery
site:
  title: Home
  #  residualpower: 250
  meters:
    grid: grid # grid meter
    pv: pv-dc-kostal # pv meter
    battery: battery-pv # battery meter
  prioritySoC: 30 # give home battery priority up to this soc (0 to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Ladesäule
    charger: wallbe # charger
    meters:
      charge: charge # charge meter
    vehicles:
      - enyaq
      - e-up
    mode: pv
    soc:
      min: 30 # immediately charge to 30% regardless of mode unless "off" (disabled)
      target: 80 # always charge to 80%
      estimate: false # set true to interpolate between api updates
      poll:
        mode: always
        interval: 2m
    onDisconnect:
      mode: pv # switch back to pv mode
    #    targetSoC: 80 # charge to xx%
    phases: 1 # ev phases (default 3)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # minimum export power (W). If zero, export must exceeds minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 10m # threshold must be exceeded for this long
      threshold: 500 # maximum import power (W)
    guardduration: 5m # switch charger contactor not more often than this (default 10m)
    mincurrent: 6 # minimum charge current (default 6A)
    maxcurrent: 16 # maximum charge current (default 16A)

# mqtt message broker
mqtt:
  broker: 10.0.1.97:1883
  user: mbmd
  password: xxxxxxxxxxxxxxxx

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:<br>
This is the first log, where detection of the second car fails on plugging in.</summary><br><pre>
See attached file 
"2021-07-30_evcc_Autowechsel_01.log".
Second car gets connected at:
[lp-1  ] INFO 2021/07/30 08:49:38 car connected
</pre></details>

[2021-07-30 evcc Autowechsel 01.log](https://github.com/andig/evcc/files/6906338/2021-07-30.evcc.Autowechsel.01.log)

<details><summary>Show evcc log output with <code>--log debug</code>:<br>
This is the second log, where the second car gets detected while plugged in after restart of evcc.</summary><br><pre>
See attached file "2021-07-30_evcc_Autowechsel_02.log".

</pre></details>

[2021-07-30 evcc Autowechsel 02.log](https://github.com/andig/evcc/files/6906352/2021-07-30.evcc.Autowechsel.02.log)

andig, 2021-07-30:
Also im ersten Fall sagen alle Autos "nicht verbunden":

    [lp-1  ] DEBUG 2021/07/30 08:38:23 vehicle status: A (Ladmila)
    [lp-1  ] DEBUG 2021/07/30 08:38:27 vehicle status: A (Uppsala)
    [lp-1  ] DEBUG 2021/07/30 08:49:38 vehicle status: A (Ladmila)
    [lp-1  ] DEBUG 2021/07/30 08:49:40 vehicle status: A (Uppsala)

auf der Basis lässt sich halt nix erkennen :(

andig, 2021-07-30:
Für den Anwender klarer würde das Ganze falls wir ein UI hätten in dem der gesamte Parkplatz z.B. einschließlich Status sichtbar wäre, vllt. auch einschl. der Möglichkeit den Status zu aktualisieren oder ein Fzg direkt auszuwählen.

/cc @DerAndereAndi @naltatis

andig, 2021-07-30:
Idee: solang _kein_ Auto identifiziert werden kann sollten die Abfragen- für definierte Zeit- weiter erfolgen um den langsameren APIs eine Chance zu geben aufzuholen.

Caibuk, 2021-07-30:
Das fände ich gut (oder einfach die ganze Zeit alle zum loadpoint gehörenden Autos abfragen - wie `poll: mode: always` einen glauben lassen könnte).

In log 1 sieht man ja auch, dass das vehicle 0 (Enyaq / "Ladmila") weiterhin regelmäßig gepollt wird, *obwohl* es ebenfalls nicht verbunden ist (sondern nur dem LP1 zugeordnet).
Das finde ich im Hinblick auf #1163 zumindest ein bisschen unlogisch.

Caibuk, 2021-07-30:
Ich habe jetzt einmal - einem Gefühl folgend - bei beiden vehicles das `cache: 2m` entfernt (keine Ahnung, warum ich das - schon immer? - da stehen hatte).
Nach einem ersten evcc-Neustart ohne angeschlossenem Auto (vehicle 0 = "Ladmila" ist zugeordnet) habe ich daraufhin vehicle 1 (= "Uppsala") angeschlossen, welches sofort erkannt wurde.

Ich hoffe, es funktioniert jetzt immer.
(Dann könnte hier zu - falls kein Bedarf besteht, den Algorithmus etwas treffsicherer zu machen.)

Caibuk, 2021-07-30:
Zu früh gefreut. Jetzt kam das große Auto (vehicle 0) wieder heim. Dem Loadpoint war noch vehicle 1 zugeordnet (welches aber schon länger nicht mehr angeschlossen war) und vehicle 0 wurde nicht als angeschlossen erkannt (wahrscheinlich auch durch zu frühes, einmaliges Polling, während die Skoda-API noch nicht aktualisiert war). Schade.

andig, 2021-07-30:
> Das fände ich gut (oder einfach die ganze Zeit alle zum loadpoint gehörenden Autos abfragen - wie poll: mode: always einen glauben lassen könnte).

Das geht nicht. Entweder Du nimmst "always" oder nicht. In letzterem Fall können wir nicht abfragen solange die Ladung von der Wallbox nicht freigegeben ist da auch das wieder zu battery drain führen würde...

**Idee**: Tatsächlich könnten wir aber auch nochmal überlegen, das Kriterium von "lädt" (Status=C) auf "Ladung ist freigegeben" (Wallbox=enabled) zu ändern. Würde Dein Problem aber nur abmildern, nicht lösen.

> Ich hoffe, es funktioniert jetzt immer.

Das ist Zufall. Tatsächlich setzen wir alle Fahrzeugcaches bei "vehicle connect" zurück, so dass alle 1x abgefragt werden.

> **Idee:** solang kein Auto identifiziert werden kann sollten die Abfragen- für definierte Zeit- weiter erfolgen um den langsameren APIs eine Chance zu geben aufzuholen.

Sagen wir für 15m?

andig, 2021-07-30:
Nochmal ein Gedanke: Log 1 zeigt, dass das API auch 10min später den Status noch nicht aktualisiert hat. Das stellt sich also schon die Frage, ob und wie lange eine wiederholte Abfrage sinnvoll sein kann...

Caibuk, 2021-07-30:
Beim ersten Abruf um 08:38 Uhr war "WVWxxx..." doch noch gar nicht verbunden... 🤔

Im gesamten Log 1 wird der VW zweimal abgefragt:
- beim evcc-Start 08:38 Uhr (kein Auto angeschlossen)
- und dann kurz nach `08:49:38 car connected` mit `08:49:40 
GET /vehicles/WVWxxxxxxxxxxxxxx/status HTTP/1.1`

In den 2 Sekunden hat es die Info "Auto angeschlossen" nicht bis in die Cloud geschafft. In meinen Augen kann das eigentlich nur unter glücklichen Umständen klappen.

In der Folge wird weiter brav alle zwei Minuten vehicle 0 ("TMBxxxx...") gepollt. Der zwar lp1 zugewiesen, aber nicht angeschlossen ist.

andig, 2021-07-30:
> Der zwar lp1 zugewiesen, aber nicht angeschlossen ist.

Das ist das Fallback. Alternative wäre so zu tun als wäre dann "unknown". Könnten wir auch diskutieren. Z.B. nach den 15min.

andig, 2021-07-30:
Anbei mal ein Versuch die Erkennung für 15min laufen zu lassen _oder_ solange, bis ein Fahrzeug als "matching" erkannt wurde.

andig, 2021-07-30:
- [ ] Erkennung für 15min wiederholen bis Fahrzeug über API erkannt
- [ ] Fahrzeug-APIs immer abfragen falls Wallbox enabled, nicht nur falls Fahrzeug lädt
- [ ] letztes aktives Fahrzeug entfernen falls nicht erneut erkannt

DerAndereAndi, 2021-07-30:
Wenn die Erkennung ob das Vehicle angeschlossen ist nach 10-15 Minuten nicht geht, wird auch der SoC Wert nicht korrekt sein. Von daher muss man sicher eher generell überlegen was da passieren soll. Denn die Zielladung (SoC und/oder Zeit) wird nicht funktionieren

Caibuk, 2021-07-31:
Ich habe die #1254 ausprobiert. Es scheint, dass vehicle 1 beim Anschließen immer noch *nur einmal* kurz nach dem Anstecken abgefragt wird. Leider ist da die Cloud noch nicht so weit (die braucht zwischen 10 und 30 Sekunden).
Da ich jetzt nur wenig Zeit habe, den Trace-log zu "putzen", schicke ich ihn @andig per Slack DN.

andig, 2021-07-31:
Nach Test von @Caibuk:
- PR aktualisiert: aktives Fahrzeug wird jetzt entfernt wenn es nicht erneut erkannt werden kann
- Voraussetzung: es hat überhaupt ein API dafür 
- Ausnahme: es ist das einzige konfigurierte Fahrzeug

Zu entscheiden: soll das konfigurierte Fahrzeug immer entfernt werden falls es das entsprechende API hat- auch wenn es das einzige Fahrzeug am Ladepunkt ist? Konfigurierbar machen (z.B. `anonymousVehicle: true`?)

Problem identifiziert: per Default steht der Cache für Fahrzeugabfragen auf 15m. Wenn wir nur 15m lang versuchen das aktuelle Fzg. zu finden dann ist das zu kurz. Alternativen:
- default cache kleiner (5min) - das könnte zu API Misuse führen
- länger als 15min neue Autos suchen - ziemlich träge
- während der 15min Suche alle 5min den Fahrzeug-Cache resetten und die Suche damit neu anstarten

Was meint ihr?

andig, 2021-07-31:
> Es scheint, dass vehicle 1 beim Anschließen immer noch _nur einmal_ kurz nach dem Anstecken abgefragt wird. Leider ist da die Cloud noch nicht so weit (die braucht zwischen 10 und 30 Sekunden).

s.o. Du kannst das über `cache` am Fzg ändern. Default sind 15m, das passt hier nicht.

andig, 2021-07-31:
> während der 15min Suche alle 5min den Fahrzeug-Cache resetten und die Suche damit neu anstarten

Ist im letzten Commit umgesetzt. Bitte nochmal testen.

andig, 2021-08-01:
@DerAndereAndi hatte noch eine interessante Anmerkung: die Erkennung muss Fahrzeuge ausschließen, die an anderen Ladepunkten schon erkannt wurden. Damit stellt sich die generelle Frage, ob eine Konfiguration von Fahrzeugen je Ladepunkt eigentlich sinnvoll ist oder diese nicht völlig flexibel sein sollten (mit den entsprechenden Nachteilen für Fahrzeuge ohne die notwendigen APIs). Ich schlage vor das Thema separat zu behandeln.

premultiply, 2021-08-01:
Wie wäre es mit folgender Herangehensweise:
Bei einem connect-Event (ein Fahrzeug wurde an einen LP angeschlossen) aktives Update des Fahrzeugstatus (sofern für die jeweilige API erforderlich oder möglich) bei allen konfigurierten und nicht bereits erkannten Fahrzeugen anfordern.
Nachfolgend maximal 5 mal im Minutenabstand (also für maximal 5 Minuten) API-Polling bis ein passendes Fahrzeug gefunden wurde.
Sobald ein passendes Fahrzeug gefunden und zugeordnet ist werden die Abfragen nur noch auf die erkannten und verbundenen Fahrzeuge beschränkt.
Bei jedem weiteren Zustandswechsel eines LP eines verbundenen Fahrzeug wird eine erneutes aktives Update an die entsprechende API versendet (sofern seitens der API nötig) und anschließend wieder bis zu 5 mal im Minutenabstand API-Polling bis Update erkannt wurde (z. B. via Timestamp oder Statusänderung).
Das sonstige Polling erfolgt dann gecached nur während des Ladevorgangs bzw. wenn Verbunden mit pro Fahrzeug konfigurierbaren Zyklen (z. B. 10 / 60 Minuten). Hierzu könnte man auch in der jeweiligen Implementierung dafür sinnvolle Default-Werte hinterlegen die sich ja je nach Hersteller unterscheiden können.
Darüber hinaus könnte man dort über eine Eigenschaft auch festlegen ob die entsprechende API auch darüber hinaus abgefragt werden darf (und wie oft) um Battery-Drain bei manchen Herstellern und Modellen zu vermeiden.

andig, 2021-08-01:
Du hast jetzt bis auf andere Konstanten genau den PR beschrieben ;)

premultiply, 2021-08-01:
Ausgezeichnet ;)

mark-sch, 2021-08-02:
Die Trägheit der Hersteller Apis führt den SoC Mehrwert leider gelegentlich ad absurdum. Bei Peugeot hatte ich das im Bereich von Stunden beobachtet, wohin will man da noch optimieren?

Sobald der zweite Loadpoint mit zweitem Wagen hinzukommt, bei dem ein Wagen an lp1 und lp2 laden können soll, gibt es aktuell eigentlich gar keine sinnvollen Zuordnungen mehr.

Mit Blick auf Zielladen, dem Matching Problem von vehicle und loadoint, battery drain, wechselnden, unzuverlässigen Apis und der Grauzone im Bereich legal, scheint es mir sinnvoll auch alternative Herangehensweisen auszuloten. EVnotivy und auch OpenWB SoC Module lesen die notwendigen Daten in Bruchteilen von Sekunden direkt vom Fahrzeug über OBD aus.

premultiply, 2021-08-02:
Bis auf wenige Ausnahmen und gelegentliche Totalausfälle der Herstellerinfrastruktur funktioniert das bei den meisten Fahrzeugen im Rahmen der Möglichkeiten mit den APIs eigentlich recht gut.

Ja, theoretisch könnte man mit einem Freematics-OBD-Modul o. ä. einen eigenen Zugang zum Fahrzeug-Bus schaffen.
Aber auch damit benötigt es noch viel mehr sehr individuellere Anpassungen und Programmierungen für noch mehr Varianten und größtenteils echtes Reverse-Engineering auf Bitebene.
Und da ist da eventuell auch noch eine Haftungsfrage wenn es durch das dauerhaft gestecke Modul zu Problemen kommen sollte. Darüber hinaus gibt es auch dort Battery-Drain usw.
Das ist daher auch keine bessere Lösung. Bestenfalls in Einzelfällen eine Alternative.

DerAndereAndi, 2021-08-02:
Der OBD Stecker löst auch das Problem nicht wenn der Mobilfunkempfang schlecht ist.

Für alle bisherigen Varianten gibt es Problemfälle, bei manchen wird es gut funktionieren, bei anderen nicht. Egal was wir tun, es wird keine 100% Lösung sondern abhängig von der persönlichen Konstellation.

Die einzige „richtige“ Lösung kommt erst noch, wenn die Verbindung vom EV zum Charger auch bei AC über ISO15118 (-20) erfolgt, dort wird dann auch wie an den DC Säulen der SoC übertragen. Es gibt bereits vereinzelte Ansätze dafür, also herstellerpsezifische Erweiterungen für ISO15118 (-2) die das machen. Aber es ist eben auch keine Lösung für bestehende EVs und Charger.

Also wird es hier keinen allgemein funktionierenden Ansatz geben. Das ist nun mal so.

andig, 2021-08-02:
> EVnotivy und auch OpenWB SoC Module lesen die notwendigen Daten in Bruchteilen von Sekunden direkt vom Fahrzeug über OBD aus.

@mark-sch Du meinst OBD ist schneller als API? Für mich ist das keine Massenlösung und ich persönlich will keinen Bastelstecker in meinem Auto. Es ist ja kein Problem EVNotify in evcc zu integrieren falls das jemand will- bei Bedarf fehlt einfach ein Beispiel für https://github.com/andig/evcc-config, oder?

mark-sch, 2021-08-02:
> @mark-sch Du meinst OBD ist schneller als API? 

Ja, das ist viel schneller. Von mehreren SoC pro Sekunde bis zu neuen Werten alle drei Sekunden ist das was ich bisher gesehen habe - je nachdem ob zwischen verschiedenen CAN-IDs umkonfiguriert wird, um Daten von verschiedenen Steuergeräten zu empfangen, z.B. BMS und OBCM. Das Nadelöhr wird dann vielmehr die aktuelle soc Implementierung in Bezug auf mqtt, welches aktuell mit polling und cache Mechanismen eingearbeitet ist, hier wäre dann ein event basierter Ansatz wünschenswert, was ja auch besser zu der Natur von mqtt passt.

> Der OBD Stecker löst auch das Problem nicht wenn der Mobilfunkempfang schlecht ist.

Die günstigen OBD Dongle arbeiten über Bluetooth und Wifi, denke hier kann der Anwender ganz gut sicherstellen, dass im Bereich der Wallbox ein Wifi Empfang ist. Wenn Daten kommen ist es auch gleichzeitig ein Indiz, dass der Wagen vor der Haustüre steht und keinen StatusB von einem Fremdlader bekommt.

> Ja, theoretisch könnte man mit einem Freematics-OBD-Modul o. ä. einen eigenen Zugang zum Fahrzeug-Bus schaffen.
Aber auch damit benötigt es noch viel mehr sehr individuellere Anpassungen und Programmierungen für noch mehr Varianten und größtenteils echtes Reverse-Engineering auf Bitebene.

Die Herangehensweise hat ihre ganz eigenen Herausforderungen, dies kann ich bestätigen. Auf der anderen Seite ist die Vielfalt gar nicht so groß wie man annehmen möchte, egal ob e-Golf, MEB id.3 id.4 usw. sowie die Palette an PHEVs (Cupra, Skoda, VW usw.) arbeiten alle mit den gleichen CAN-IDs, um auf das BMS zugreifen zu können. Tesla klappt super und die Hyundai/Kia Palette. Wahrscheinlich nimmt es sich zum apk reversen der Hersteller Apps nicht viel.
Problematisch finde ich im aktuellen SoC Ansatz, dass es bei zwei Loadpoints+Wagen nicht funktioniert und auch ein Lösungsansatz schwer vorstellbar ist, oder gibt es hierzu schon technische Ideen?

premultiply, 2021-08-02:
Bei "OBD-Steckern" muss man zwischen drei Varianten unterscheiden.
1) Die verbreiteten winzigen Billig-Stecker auf ELM327-Clon-Basis mit WiFi/BT
2) die teureren professionellen Diagnose-Tools mit STM-Chips
3) Stand-alone-Module mit eigenem Prozessor wie Freematics

Erstere sind ein absolter Security-Albtraum, da sie ungeschütze APs erstellen mit denen sich jeder nach Lust und Laune jederzeit verbinden kann und vollen Zugriff aufs Fahrzeug hat. Das ist die EVNotify-Variante. Kommt mir persönlich ausser in der geschlossenen Garage temporär zu Diagnosezwecken niemals während der Fahrt oder länger/unbeaufsichtigt in Fahrzeug.
Haben zudem auch Probleme mit Standby/Wakeup und Battery-Drain wenn sie immer gesteckt bleiben sollten unter völliger Ignoranz des Gefährdungspotenzials.

Zweite sind in der Regel relativ groß und vglw. teuer und brauchen aber auch noch weiterhin eine externe Logik die die eigentlich Kommunikation übernimmt.

Dritte brauchen eine eigenen Programmierung laufen dann aber völlig autark und können die Daten theoretisch je nach Variante per Wifi, LTE, LoRaWAN usw. absetzen.

mark-sch, 2021-08-02:
Die Unterteilung passt gut. Wobei ELM327 nicht kategorisch ein Security Problem bedeutet. Die besseren Wifi Dongles lassen sich mit WPA2 betreiben und neben dem AP auch mit STA mode, d.h. sie sind von aussen Wifi mäßig unsichtbar und verbinden sich in Reichweite mit dem Wifi Deines Routers (WPA2). Bei Tesla konnte ich es z.B. so beobachten: Geht der Wagen in den deep sleep so wird auch die 12V Verbindung gecutted, der Dongle ist stromlos. Zudem vehindert ein CAN-Bus polling zuvor nicht den deep sleep des Tesla. Wacht der Wagen auf, kommen kurz drauf frische Daten weil der Dongle sich neu mit dem Wifi verbindet.
Wenn der Wagen den Dongle (wie bei Tesla) nicht schlafen legt, dann gibt es einen Haufen von Konfigurationsmöglichkeiten am ELM327 wie mit sleep/wakeup umgegangen werden soll, von can bus traffic monitoring, Überwachung von Spannbungsschwankungen und ignition status. Nicht jeder Dongle ist da gleich gut, aber es gibt die guten und günstigen.

andig, 2021-08-02:
> Problematisch finde ich im aktuellen SoC Ansatz, dass es bei zwei Loadpoints+Wagen nicht funktioniert und auch ein Lösungsansatz schwer vorstellbar ist, oder gibt es hierzu schon technische Ideen?

Genau in diesem Issue und PR?!

premultiply, 2021-08-02:
> Die besseren Wifi Dongles lassen sich mit WPA2 betreiben und neben dem AP auch mit STA mode, d.h. sie sind von aussen Wifi mäßig unsichtbar und verbinden sich in Reichweite mit dem Wifi Deines Routers (WPA2).

Beispiel? Link? Ich kenne bisher keine die nicht in Kategorie 2 fallen lerne aber sehr gerne dazu.

mark-sch, 2021-08-02:
> Beispiel? Link? Ich kenne bisher keine die nicht in Kategorie 2 fallen lerne aber sehr gerne dazu.

https://www.sqrt.io/blog/obd2-wifi-adapter-absichern

DerAndereAndi, 2021-08-02:
Der OBD Stecker funktioniert auch nicht immer, in unserem ID.3 hat der nach ner halben Stunde keinen Strom mehr, auch wenn das EV am AC laden ist. Nochmal, auch das mag/könnte bei manchen Konstellationen besser funktionieren, aber nicht bei allen.

mark-sch, 2021-08-02:
> Der OBD Stecker funktioniert auch nicht immer, in unserem ID.3 hat der nach ner halben Stunde keinen Strom mehr, auch wenn das EV am AC laden ist. Nochmal, auch das mag/könnte bei manchen Konstellationen besser funktionieren, aber nicht bei allen.

Das mit dem id.3 ist zu lösen, du musst dem Dongle die richtige Initialisierungssequenz schicken damit er nicht mehr einschläft. Im Moment sehe ich das nach längerem Testing in bisher allen Konstellationen lösbar. Die Wifi Dongle muss entsprechend vorkonfiguriert werden, perspektivisch wäre es ja auch denkbar, wenn es für unter 50,- Euro einen fertig konfigurierten evcc soc Dongle gibt, plug&play :)
Die Riskobewertung Open Source "without warranty" an sein Fahrzeug ran zu lassen muss am Ende jeder für sich entscheiden, machst du ja auch jetzt schon wenn bei Zappelstrom die contactors der HV Batterie potentiell bei schlechter Konfig hin- und herschalten. Bei modernen Fahrzeugen hängt die OBD vor einem Can-Gateway, ohne einen speziellen Routing request kommt vom und vorallem zum eigentlichen powertrain can erstmal nichts störendes an.

andig, 2021-08-06:
> Es ist ja kein Problem EVNotify in evcc zu integrieren falls das jemand will- bei Bedarf fehlt einfach ein Beispiel für https://github.com/andig/evcc-config, oder?

ist dank @xerion3800 auch drin

closed 2021-08-07 by andig