evcc-io/evcc #1255: OCPP support available/panned?

daviid5, 2021-07-30:
Hello together,

I'm looking for a solution to manage the charging with severral charging points (10 or more).
The charging stations use Bender CC613 controller, which support OCPP 1.6J.
As far as I understand evcc can only act as an OCPP "client", but cannot communicate via OCPP with charging stations, is this right?

Currently I started to implement my own solution, using this OCPP implementation: https://github.com/lorenzodonini/ocpp-go which seems pretty straight forward.

andig, 2021-07-31:
> As far as I understand evcc can only act as an OCPP "client", but cannot communicate via OCPP with charging stations, is this right?

See https://github.com/andig/evcc/issues/1139. As of today, nobody has made a test device available for implementing server mode.

> The charging stations use Bender CC613 controller, which support OCPP 1.6J.

Does it support Smart charging? If not it cannot be controlled by evcc for purpose of PV-controlled charging.

> using this OCPP implementation

It's also used by evcc. Would be nice if we could do this together and improve evcc. Shall we continue in German?

andig, 2021-08-03:
ping @daviid5

andig, 2021-08-06:
Ich fände es ja zumindest höflich, auf eine ausführliche Antwort ein Lebenszeichen zu bekommen :(

closed 2021-08-06 by andig

daviid5, 2021-08-06:
Sorry, ist total untergegangen.

> See #1139. As of today, nobody has made a test device available for implementing server mode.
Ich nehme an du meinst physischen Zugriff auf einen Laderegler?
Ausleihen kann ich leider auch nichts.
Die ocpp-go Implementierung bietet aber auch die Möglichkeit OCPP-Clients abzubilden. Ist ein Example dabei, damit habe ich bisher meine Software getestet.

> Does it support Smart charging? If not it cannot be controlled by evcc for purpose of PV-controlled charging.
Tut es, der Bender CC613 Controller hat sogar ein integriertes (rudimentäres) Last-Management in dem ein Controller als Master.

> It's also used by evcc. Would be nice if we could do this together and improve evcc. Shall we continue in German?
Für was wird das denn schon verwendet?
Um evcc als OCPP Client/Chargepoint zu nutzen?

andig, 2021-08-09:
> Sorry, ist total untergegangen.

Kein Problem, danke für die Antwort. Wie gesagt- gerne auch engerer Austausch bei Slack.

> > See #1139. As of today, nobody has made a test device available for implementing server mode.
> 
> Ich nehme an du meinst physischen Zugriff auf einen Laderegler?

Ja genau. Bisher ist das alles Theorie.

> Die ocpp-go Implementierung bietet aber auch die Möglichkeit OCPP-Clients abzubilden. Ist ein Example dabei, damit habe ich bisher meine Software getestet.

Ja, hab ich auch- sagt aber nix über das Verhalten bei echter HW aus. Die "wirkliche" Interaktion würde mich schon interessieren.

> > It's also used by evcc. Would be nice if we could do this together and improve evcc. Shall we continue in German?
> 
> Für was wird das denn schon verwendet?
> Um evcc als OCPP Client/Chargepoint zu nutzen?

Genau. evcc kann beliebige Wallboxen ggü. dem Netzbetreiber/CS als Chargepoint anbieten. Einige NB fordern das, tatsächlich scheint aber dennoch keiner von denen ein CS zu haben so dass auch der Teil nie in der Realität genutzt wurde.

Der zweite Teil wäre jetzt eben genau mal selbst als CS zu agieren und einen CP zu steuern.

xantalor, 2022-01-02:
Hallo zusammen,

gab es her schon weitere Bemühungen? 
Ich würde / kann gerne auch bei der Programmierung unterstützen und an meiner Elvi von EVBox testen. 

Könntet Ihr mich ggf. abholen wie der aktuelle stand ist was ich mir anschauen könnte um das voran zu treiben?

Beste Grüße

andig, 2022-01-02:
Mir fehlt weiter eine Testbox im Zugriff

xantalor, 2022-01-02:
Du meinst eine Box bei dir vor Ort?

andig, 2022-01-02:
Zugriff reicht. Wobei es für OCPP notwendig ist, dass die Box sich beim CentralSystem anmeldet.

daviid5, 2022-01-02:
Stand von mir:
Mit evcc-io habe ich bisher nichts weiter gemacht.

Ich habe an unserer internen Anwendung etwas weiter gearbeitet die direkt die ocpp-go Implementierung nutzt.
Ich habe eine MariaDB Backend, mit der Authorisierung der RFID-Karten, logging der Zählerstände, Speicherung des aktuellen Status der Ladepunkte.
Die ocpp-go lib funktioniert wunderbar, out of the box mit unseren Ladesäulen (intern Bender CC613).
Next Step ist die Begrenzung der Ladeleistung mit den OCPP Charging-Profiles.
Last Management entsprechend abhängig vom zur Verfügung stehenden Überschussstrom, das dürfte das komplizierteste werden.
Zusätzlich muss auch noch beachtet werden dass eine Ladesäule (2 Ladepunkte) nur mit 50A abgesichert sind, sprich ich kann wenn nur ein Auto an der Säule hängt volle 32A zur Verfügung stellen, mit 2 angeschlossenen EVs nur 25A.
Außerdem gibt es nochmal weitere Vorsicherungen, z.B. 5 Säulen (10 Ladepunkte) die nur mit 150A vorgesichert sind, auch das muss natürlich mit beachtet werden ...

@andig  
Ich hatte auch mal mit Bender Kontakt, waren super nett.
Evtl. dort mal nachfragen ob sie zum Testen mal einen Laderegler zur Verfügung stellen würden?

andig, 2022-01-02:
> Die ocpp-go lib funktioniert wunderbar,

Die verwende ich auch

> Last Management entsprechend abhängig vom zur Verfügung stehenden Überschussstrom, das dürfte das komplizierteste werden.

Das hat evcc ja schon, jedenfalls für den PV-Teil

> Evtl. dort mal nachfragen ob sie zum Testen mal einen Laderegler zur Verfügung stellen würden?

Gute Idee- hast Du evtl. Kontaktdaten von einem Asp.? Gerne an cpuidle@gmx.de

xantalor, 2022-01-02:
Zum testen, kann ich sicherlich mal meine Box zur Verfügung stellen. Muss nur zu erst das die sehr schwache WLAN Anbindung beheben.

andig, 2022-01-02:
@daviid5 Du kannst Dich gerne bei Slack melden und wenn Du willst mal hier https://github.com/evcc-io/evcc/pull/1963 rein denken.

daviid5, 2022-01-02:
@andig wo find ich den euren Slack??

andig, 2022-01-02:
https://join.slack.com/t/evccgroup/shared_invite/zt-fw52e6lt-tdazCp1LPdPlYuKz3PvTAw

andig, 2022-01-05:
https://github.com/evcc-io/evcc/pull/1963 nimmt langsam Gestalt an. Wer mag und sich traut kann schonmal einen Blick drauf werfen.

afiller, 2022-03-13:
Hello zusammen... ich hätte auch eine EVBOX Elvi im Angebot die übers Netz erreichbar ist und dazu einen Polestar 2 der gerne per PV und SENEC.home immer bis 80% geladen werden soll. Könnte mich jemand ins Slack einladen? Der Link oben ist ausgelaufen und ich wäre ebenfalls einerseits Entwickler und hätte andererseits ein großes Interesse mit euch zusammen evcc mit OCPP zum Laufen zu bekommen.

andig, 2022-03-14:
@afiller meld dich bitte bei Slack, Link siehe Doku. Wenn das nicht klappt schick bitte eine Email zwecks Einladung an info@evcc.io