evcc-io/evcc #1267: Modbus Skalierungsfaktoren berücksichtigen können

sebnaf, 2021-08-03:
SolarEdge verwendet Skalierungsfaktoren, um Werte auszugeben.

Als User hätte ich gerne, dass ich Modbus-Register mit anderen Registern multiplizieren kann, um sinnvolle Werte für evcc zu erhalten.

Idealerweise erhalte ich eine Beispiel-Konfiguration für den aktuellen PV Ertrag und den Bezug/Einspeisung für eine SolarEdge-Konfiguration.
Sollte ich für das neue UI weitere Werte benötigen, mögen diese bitte ebenfalls evaluiert werden.

SunSpec ist leider kein Workaround, da die Implementation bei SolarEdge fehlerhaft/eigenwillig ist.

Da ich die ursprüngliche Konversation https://github.com/andig/evcc/discussions/1264 nicht in ein Issue wandeln kann (URL Fehler in github, "URL too long" hiermit ein neues Issue.

Danke sehr

andig, 2021-08-03:
> SunSpec ist leider kein Workaround, da die Implementation bei SolarEdge fehlerhaft ist.

was heisst das hier konkret?

sebnaf, 2021-08-03:
> was heisst das hier konkret?

Per SunSpec kann auf SolarEdge zwar der aktuelle PV-Ertrag ausgelesen werden, sämtliche angeschlossenen Smartmeter jedoch nicht - obwohl so dokumentiert.

Diese funktionieren jedoch per Modbus.

andig, 2021-08-03:
> Per SunSpec kann auf SolarEdge zwar der aktuelle PV-Ertrag ausgelesen werden, sämtliche angeschlossenen Smartmeter jedoch nicht - obwohl so dokumentiert.

Könntest Du mal bitte den Output von 

    mbmd inspect 

zeigen?

sebnaf, 2021-08-03:
@andig Gern.

```
pi@evcc:/opt $ sudo mbmd inspect -a 10.10.20.7:502 -d SUNS:1
2021/08/03 21:23:54 config: creating TCP connection for 10.10.20.7:502
--------- Model 1 common ---------
Mn         SolarEdge     string
Md   SE5000H-RW000BNN    string
Opt                      string
Vr           0004.001    string
SN           7401FC36    string
DA               1.00    uint16
Pad            0x0065    pad
--------- Model 101 inverter ---------
A                    0.00    uint16
AphA                 0.00    uint16
AphB                  n/a    uint16
AphC                  n/a    uint16
A_SF                   -2    sunssf
PPVphAB            234.20    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA                n/a    uint16
PhVphB                n/a    uint16
PhVphC                n/a    uint16
V_SF                   -1    sunssf
W                    0.00    int16
W_SF                    0    sunssf
Hz                  50.03    uint16
Hz_SF                  -3    sunssf
VA                   0.00    int16
VA_SF                   0    sunssf
VAr                  0.00    int16
VAr_SF                  0    sunssf
PF                   0.00    int16
PF_SF                   0    sunssf
WH             1923941.00    acc32
WH_SF                   0    sunssf
DCA                  0.01    uint16
DCA_SF                 -3    sunssf
DCV                  0.00    uint16
DCV_SF                 -1    sunssf
DCW                  0.00    int16
DCW_SF                  0    sunssf
TmpCab                n/a    int16
TmpSnk              41.35    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      2    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 1 common ---------
Mn          SolarEdge    string
Md   SE-MTR-3Y-400V-A    string
Opt          Export+I    string
Vr                 74    string
SN          606535102    string
DA               2.00    uint16
Pad            0x00cb    pad
--------- Model 203 ac_meter ---------
A                            5.70    int16
AphA                         1.90    int16
AphB                         2.00    int16
AphC                         1.80    int16
[... 125 lines cut ...]
TotVAhExpPhC                  NaN    acc32
TotVAhImp                     NaN    acc32
TotVAhImpPhA                  NaN    acc32
TotVAhImpPhB                  NaN    acc32
TotVAhImpPhC                  NaN    acc32
TotVAh_SF                     n/a    sunssf
TotVArhImpQ1                  NaN    acc32
TotVArhImpQ1PhA               NaN    acc32
TotVArhImpQ1PhB               NaN    acc32
TotVArhImpQ1PhC               NaN    acc32
TotVArhImpQ2                  NaN    acc32
TotVArhImpQ2PhA               NaN    acc32
TotVArhImpQ2PhB               NaN    acc32
TotVArhImpQ2PhC               NaN    acc32
TotVArhExpQ3                  NaN    acc32
TotVArhExpQ3PhA               NaN    acc32
TotVArhExpQ3PhB               NaN    acc32
TotVArhExpQ3PhC               NaN    acc32
TotVArhExpQ4                  NaN    acc32
TotVArhExpQ4PhA               NaN    acc32
TotVArhExpQ4PhB               NaN    acc32
TotVArhExpQ4PhC               NaN    acc32
TotVArh_SF                    n/a    sunssf
Evt                    0x00000000    bitfield32
```

andig, 2021-08-03:
Und was fehlt Dir jetzt genau bei den Metern? Sieht ziemlich korrekt und vollständig aus?

premultiply, 2021-08-03:
Ich denke es fehlt beim Meter nur der passende Index in der Konfig.

Die Batterie gibts bei SE aber leider immer noch nicht als SubSpec-Model...

sebnaf, 2021-08-03:
An den PV-Ertrag komme ich:

```
 - name: pv # pv power calculated from DC string and battery charge power
   type: custom
   power:
    source: modbus
    model: sunspec
    uri: 10.10.20.7:502 # Port 502 (SetApp) or 1502 (LCD)
    id: 1
    value: 101:W ```

Wie komme ich denn an die Grid-Info, das ist Meter 1/W

```
 - name: grid # pv power calculated from DC string and battery charge power
   type: custom
   power:
    source: modbus
    model: sunspec
    uri: 10.10.20.7:502 # Port 502 (SetApp) or 1502 (LCD)
    id: 1
    value: 203:W
    ```
ist es nicht: 
```
grid
----
Power: sunspec: 201:0:W not found
``` 

Die abschließende spannende Frage ist, ob diese Werte auch einen Scaling-Faktor brauchen. Das sehe ich, sobald ich wieder Licht habe…

andig, 2021-08-03:
> Die abschließende spannende Frage ist, ob diese Werte auch einen Scaling-Faktor brauchen.

den haben die schon. Deshalb solltest du ja Sunspec nehmen

andig, 2021-08-03:
In deiner Grid Konfig gehts zwischen 201 und 203 durcheinander. Es muss 203 sein, im Test hattest du aber 201?!

premultiply, 2021-08-03:
203:1:W ?

sebnaf, 2021-08-03:
> 203:1:W ?

Habe leider schon alle mir logischen Varianten durch

```
grid
----
Power: sunspec: 203:1:W not found

pv
--
Power: 0W
```

premultiply, 2021-08-03:
Mist. Dann muss @andig das Rätsel lösen 🤔

sebnaf, 2021-08-03:
> Mist. Dann muss @andig das Rätsel lösen 🤔

Definitiv schonmal Dank an Dich!

andig, 2021-08-04:
Ich glaube die Lösung liegt hier: https://github.com/andig/evcc/blob/master/util/modbus/modbus.go#L22. 
Du hast mehrere SunSpec (Sub)Devices (und die jeweils in unterschiedliche Models aufgeteilt). Jedes startet mit Model 1 (Common Block). 203:1:W bezeichnet den 1. Block in Model 203, Reading W(att), also Leistung.
Versuch mal subdevice: 1 oder 2 um an die Zähler zu kommen.

premultiply, 2021-08-04:
Das hatten wir ja eigentlich schon gerade zuvor erfolglos probiert. Und normalerweise würde ich aus Anwendersicht auch eigentlich erwarten dass er per default den ersten Eintrag nimmt wenn kein Index explizit angegeben ist.

Was sagt denn eigentlich `evcc detect`?

andig, 2021-08-04:
> Und normalerweise würde ich aus Anwendersicht auch eigentlich erwarten dass er per default den ersten Eintrag nimmt wenn kein Index explizit angegeben ist.

Genau das macht er. Deshalb ja `subdevice: 1/2` um an die Zähler zu kommen!

sebnaf, 2021-08-04:
> Was sagt denn eigentlich `evcc detect`?

Nichts, da der WR in einem anderen Netz ist - wie alle IoT-Geräte.

sebnaf, 2021-08-04:
> Versuch mal subdevice: 1 oder 2 um an die Zähler zu kommen.

Ich habe 
`id: 1`bis `id: 5`probiert in allen Kombinationen mit `value: 203:0:W`bis `value: 203:5:W`

andig, 2021-08-04:
Ich sagte `subdevice`, nicht id. Ist doch nicht so schwer wenn man mal zuhört:

```
device: ...
id: 1
subdevice: 1/2
power: 203:W
```

sebnaf, 2021-08-04:
> subdevice: 1 # oder 2

Subdevice:1 hat funktioniert, danke.

Ich lese wohl sehr aufmerksam mit, jedoch war der Parameter `subdevice`mir bisher nicht als solchiger bekannt. Entweder bin ich nicht clever genug oder die Doku nicht ausreichend. Oder beides.

In diesem Fall freue ich mich, dass wir jetzt eine Konfiguration für SolarEdge mit SunSpec haben, die wir für immer und ewig für die Nachwelt sauber dokumentiert hier versenken können.

Danke.

andig, 2021-08-04:
Freu mich, dass es klappt. Entschuldige meine Ungeduld.

> Entweder bin ich nicht clever genug oder die Doku nicht ausreichend. Oder beides.

Auf jeden Fall die Doku! Ein bekanntes Manko... Den PR um das "händisch" zu machen können wir damit ad acta legen?

premultiply, 2021-08-04:
> Genau das macht er. Deshalb ja `subdevice: 1/2` um an die Zähler zu kommen! Die 1 in `203:1:W` bezeichnet den Block im Model, nicht das Subdevice!

Oh, danke @andig für die Erklärung! Das war mir auch noch nicht klar und neu.
`subdevice` kannte ich noch überhaupt nicht. Ich glaub das haben wir vorher auch noch nirgends verwendet, oder?
Wie geht denn `evcc detect` damit theoretisch um? Mangels Gerät kann ich das ja auch nicht selbst probieren...

andig, 2021-08-04:
> Wie geht denn evcc detect damit theoretisch um? Mangels Gerät kann ich das ja auch nicht selbst probieren...

Keine Ahnung. Ich glaube wir finden erstmal nur die IDs. Damit ist ja schonmal klar, was für ein Gerät da horcht. Vllt könnte @sebnaf das mal ausprobieren.

sebnaf, 2021-08-04:
> Den PR um das "händisch" zu machen können wir damit ad acta legen?

Von mir aus gerne, ich poste meine Konfig mit Infos in das Wiki in den nächsten Tagen.

sebnaf, 2021-08-04:
> @sebnaf das mal ausprobieren.

Ja, gerne. Reicht das kommende Woche noch? Ich muss hier netzwerktechnisch ein paar Dinge ändern, damit das klappt. Diese Woche ist die Zeit etwas knapp für solche Dinge im laufenden Betrieb.

closed 2021-08-04 by andig