evcc-io/evcc #1295: easee Charger: Support for multiple circuits

simonstamm, 2021-08-06:
Hi @andig,

Thanks for your effort integrating all kind of chargers. I'm currently having two easee charger installed in my garage with the following settings:

[image]

When trying to start evcc it fails with:

```
cannot create charger 'Easee': cannot create charger 'easee': cannot determine circuit id, found: [{121841 126057 1 1 16 false 0} {121847 126057 2 2 16 false 0}]
```

That makes sense when I look into:

https://github.com/andig/evcc/blob/502a921a1d902f76fef0a4c28332d0522f141738/charger/easee.go#L100

Is there any way you can filter on the circuits response to the correct charger instead of using always `site.Circuits[0].ID`?

Example response I get on `GET https://api.easee.cloud/api/chargers/EH1/site`:

```json
{
    "vat": 0.0,
    "partnerId": null,
    "useDynamicMaster": false,
    "circuits": [
        {
            "chargers": [
                {
                    "userRole": null,
                    "isTemporary": false,
                    "id": "EH1",
                    "name": "Easee Home 1",
                    "color": 1,
                    "createdOn": "2020-11-26T11:03:09.468882",
                    "updatedOn": "2021-08-05T06:45:54.365364",
                    "backPlate": {
                        "id": "xxx",
                        "masterBackPlateId": "xxx",
                        "name": "Easee Home 1"
                    },
                    "levelOfAccess": null,
                    "productCode": null
                }
            ],
            "masterBackplate": null,
            "useDynamicMaster": false,
            "parentCircuitId": null,
            "id": 121841,
            "siteId": 126057,
            "circuitPanelId": 1,
            "panelName": "1",
            "ratedCurrent": 16.0
        },
        {
            "chargers": [
                {
                    "userRole": null,
                    "isTemporary": false,
                    "id": "EH2",
                    "name": "Easee Home 2",
                    "color": 1,
                    "createdOn": "2020-12-16T19:23:53.248163",
                    "updatedOn": "2021-08-05T07:16:51.59802",
                    "backPlate": {
                        "id": "xxx",
                        "masterBackPlateId": "xxx",
                        "name": "Easee Home 2"
                    },
                    "levelOfAccess": null,
                    "productCode": null
                }
            ],
            "masterBackplate": null,
            "useDynamicMaster": false,
            "parentCircuitId": null,
            "id": 121847,
            "siteId": 126057,
            "circuitPanelId": 2,
            "panelName": "2",
            "ratedCurrent": 16.0
        }
    ],
    "equalizers": [],
    "createdOn": "2021-08-05T06:31:16.301938",
    "updatedOn": "2021-08-05T07:54:52.539581",
    "contactInfo": {
        "installerName": null,
        "installerPhoneNumber": null,
        "ownerName": "Simon",
        "ownerPhoneNumber": "+xxx",
        "company": null
    },
    "costPerKWh": 0.0,
    "costPerKwhExcludeVat": 0.0,
    "currencyId": "NOK",
    "siteType": 1,
    "ratedCurrent": 50.0,
    "id": 126057,
    "siteKey": "LGUS-Y222",
    "name": "Site",
    "levelOfAccess": null,
    "address": {
        "street": "",
        "buildingNumber": "",
        "zip": "",
        "area": "",
        "country": null,
        "latitude": null,
        "longitude": null,
        "altitude": null
    }
}
```

Thanks!

andig, 2021-08-06:
> Is there any way you can filter on the circuits response

Not yet but we can easily change that, will do a PR for you

andig, 2021-08-06:
PR ist da- mich würde interessieren, wie sich das dann in der Praxis äußert wenn beide Boxen auf voller Last laufen wollen. Da müssten einzelne Stromvorgaben ja dynamisch ignoriert werden. Bin gespannt wie das läuft!

simonstamm, 2021-08-06:
Das ging fix, lieben Dank dir! Dein PR funktioniert prima:

```
[lp-1  ] INFO 2021/08/06 17:41:57 loadpoint 1:
[lp-1  ] INFO 2021/08/06 17:41:57   mode:      minpv
[lp-1  ] INFO 2021/08/06 17:41:57   charger:   power ✓ energy ✗ currents ✓ timer ✗
```

Bzgl. der beiden Stromkreise: Das stimmt, aktuell hab ich die 2. Wallbox nur installiert und eingebunden, das E-Auto meiner Frau kommt jedoch erst im nächsten Jahr. Das Testen der einzelnen Stromvorgaben wird daher noch ein bisschen dauern. :-)

closed 2021-08-06 by andig

andig, 2021-08-15:
@simonstamm hättest Du Lust mal https://github.com/andig/evcc/pull/1213 zu testen?