evcc-io/evcc #1307: Support vehicle identification across multiple loadpoints

andig, 2021-08-08:
> Ja bitte, denn jetzt ist es mit zwei Fahrzeugen und zwei LP nicht sinnvoll nutzbar

na jetzt wollen wir uns doch erstmal freuen, dass es an einem Ladepunkt schön läuft ;)

_Originally posted by @andig in https://github.com/andig/evcc/issues/1254#issuecomment-894385188_

andig, 2021-08-08:
PR ist nicht viel getestet, sollte aber dafür sorgen, dass bereits zugewiesene Fahrzeuge nicht nochmal erkannt werden und damit am jeweiligen Ladepunkt "kleben" bleiben. Prinzipiell nicht lösen lassen sich race conditions wenn z.B. 2 Fahrzeuge gleichzeitig eingesteckt werden und die Fahrzeugsuche auf beiden APIs dann los läuft.

xerion3800, 2021-08-09:
Habe den PR (686296b338f03fc2b7ba62c7e4d95bcb339e6150) mal kurz in meiner Testumgebung angetestet. Die Erkennung an sich funktioniert aber wenn die Erkennung durch ist und das andere (nicht verbundene) Auto dann den Status "A" liefert, wechselt der verbundene LP ständig zwischen "unknown" und den Namen des verbundenen Autos hin und her, kann man auch gut im Log sehen.
[evcc.log](https://github.com/andig/evcc/files/6952700/evcc.log)

andig, 2021-08-10:
Sollte behoben sein. Ein Hinweis noch: falls während der Suchephase ein zweites Fahrzeug gefunden wird bewerten wir die Situation als "unklar" und schmeissen beide raus. Das ist notwendig um auch langsamen APIs eine Chance zu geben.

xerion3800, 2021-08-10:
Perfekt :-)  jetzt funktioniert es wie es soll, das einzige was noch vorhanden war ist das hier https://github.com/andig/evcc/issues/1306#issue-963494752 aber das ist ja bekannt ;-)

closed 2021-08-10 by andig

tadschik, 2021-08-12:
Da passt glaube ich noch was nicht.

loadpoint.go

if _, ok := lp.vehicles[0].(api.ChargeState); !ok {
			lp.setActiveVehicle(lp.vehicles[0])
		}

liefert beim enyaq leider kein "ok". Der ursprüngliche Code funktioniert.

PR #1308 (2021-08-10): Fix vehicle detection across multiple loadpoints
Fix #1307