evcc-io/evcc #1312: ist it possible to use PSA Api without clientid/secret?

mamaretti, 2021-08-09:
Hi andig,

as written in https://github.com/andig/evcc/issues/586#issuecomment-895057237 I'm asking if its possible to use the PSA-API with username and password?

I'm using this on iobroker which working great just need username and password.
There is no server "flobz/psa_car_controller" needed.

https://github.com/TA2k/ioBroker.psa

https://forum.iobroker.net/topic/45951/test-adapter-psa-v0-0-x-peugot-opel-citroen

thx

andig, 2021-08-09:
You need client id and secret in the configuration. The linked repo has these hard-coded instead of extracted per user from the PSA app. Could you check if the values from https://github.com/TA2k/ioBroker.psa/blob/109f152c020baedb7d6626f1ed1345baecb9f1e2/main.js#L37 Work for you? Then I can add these to evcc.

mamaretti, 2021-08-09:
Yes its working. just install the iobroker adapter, and enter my PSA username and password. then I get some useful informations of my car. (battery level etc....)

andig, 2021-08-09:
> Yes its working. just install the iobroker adapter

That was not my question ;). I would like to know if the client id/secret plus your username/password work with evcc?

mamaretti, 2021-08-09:
sorry. 
haven't tested it, will do next days.... waiting for PIE.

andig, 2021-08-09:
If you could share your login credentials for testing please email me at cpuidle@gmx.de

closed 2021-08-10 by andig