evcc-io/evcc #1325: rfid support for go-echarger

tobintax, 2021-08-11:
It would be nice to have the rfid feature for the go-echarger wallbox. I would like to start the charging with my rfid dongle. 
This could be implemented through the go-echarger api with parameter "unlocked_by". That shows the number which legitimated the last charging. If this number is higher than 0, charging is allowed and evcc is set to a predefined value (e.g. charge now or charge PV only). 

In additional the charged kwh per rfid card would be nice. This could be implemented through the followed go-echarger api parameter (equal to 1-10 rfid card): 

eca
ecr
ecd
ec4
ec5
ec6
ec7
ec8
ec9
ec1

Thanks in advance!

andig, 2021-08-11:
Kannst Du mal bitte eine API response von Dir zeigen wenn per RFID freigeschaltet wurde? Wie sieht das aus wenn *nicht* per RFID freigegeben wurde?
Was brauchst du zum matchen: die RFID ID oder den Namen (Letzteres)?

tobintax, 2021-08-11:
Hi Andi, schön wieder mit dir zu arbeiten ;-)

Hier ohne rfid, "uby" also "unlocked by" steht auf 0. 

_{"version":"B","tme":"1108211913","rbc":"16","rbt":"1232621891","car":"1","amp":"16","err":"0","ast":"1","alw":"0","stp":"0","cbl":"32","pha":"56","tmp":"7","tma":[25.50,25.00,24.88,24.63],"amt":"32","dws":"0","dwo":"0","adi":"1","uby":"0","eto":"7100","wst":"3","txi":"0","nrg":[220,220,219,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"032671","wss":"wlan-ap","wke":"*************************","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"50","aho":"3","afi":"7","azo":"0","ama":"16","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"65535","cfi":"65280","lse":"1","ust":"2","wak":"192ce7f308","r1x":"2","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"5250","ecr":"1790","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"046D907A","rcr":"044DC2BA","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"user-1","rnm":"User2","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":16,"lom":6,"lop":1,"log":"SKKMKPVF","lon":0,"lof":0,"loa":0,"lch":30736,"mce":1,"mcs":"192.168.3.138","mcp":1883,"mcu":"","mck":"","mcc":0}_                                    

Hier mit rfid freigeschaltet: "uby" also "unlocked by" steht auf 2, da es der rfid user 2 war. 

_{"version":"B","tme":"1108211913","rbc":"32","rbt":"538425552","car":"4","amp":"6","err":"0","ast":"1","alw":"0","stp":"0","cbl":"20","pha":"56","tmp":"42","tma":[30.13,29.25,29.50,30.00],"amt":"32","dws":"10977553","dwo":"0","adi":"1","uby":"2","eto":"10390","wst":"3","txi":"0","nrg":[220,220,219,1,0,0,0,0,0,0,0,0,0,0,0,0],"fwv":"040.0","sse":"032628","wss":"wlan-ap2","wke":"****************","wen":"1","cdi":"0","tof":"101","tds":"1","lbr":"93","aho":"3","afi":"7","azo":"0","ama":"16","al1":"6","al2":"8","al3":"10","al4":"12","al5":"16","cid":"255","cch":"65535","cfi":"65280","lse":"1","ust":"2","wak":"c31d8370e6","r1x":"0","dto":"0","nmo":"0","sch":"AAAAAAAAAAAAAAAA","sdp":"0","eca":"440","ecr":"9410","ecd":"0","ec4":"0","ec5":"0","ec6":"0","ec7":"0","ec8":"0","ec9":"0","ec1":"0","rca":"046D907A","rcr":"044DC2BA","rcd":"","rc4":"","rc5":"","rc6":"","rc7":"","rc8":"","rc9":"","rc1":"","rna":"user-1","rnm":"user-2","rne":"","rn4":"","rn5":"","rn6":"","rn7":"","rn8":"","rn9":"","rn1":"","loe":0,"lot":16,"lom":6,"lop":1,"log":"SKKMKPVF","lon":0,"lof":0,"loa":0,"lch":701,"mce":1,"mcs":"192.168.3.138","mcp":1883,"mcu":"","mck":"","mcc":0}_

tobintax, 2021-08-11:
> Das passt erstmal nicht in die Struktur. Wozu brauchst Du die?

Was man hat, das hat man :-) Ich lese es aktuell aus und stelle es in einem Grafana Dashboard dar. So wichtig ist das nicht. Wichtiger wäre mir dass man das Laden per RFID am go-echarger starten kann. Das wäre cool

andig, 2021-08-11:
Was für ein krankes API. Probiers gerne mal...

andig, 2021-08-11:
> Wichtiger wäre mir dass man das Laden per RFID am go-echarger starten kann. Das wäre cool

Neues Ticket...

tobintax, 2021-08-11:
Sorry nochmal, was muss ich nun konkret tun :)

andig, 2021-08-11:
Master auschecken und compilieren oder auf Release warten. Wie immer ;)

closed 2021-08-11 by andig

tobintax, 2021-08-11:
ok, dann warte ich (leider). Habe keine Ahnung vom compilieren. 
Nur noch eine Frage: Was hast Du nun genau implementiert? Wenn "uby" auf 1-10 steht, dann startet er das Laden? Und wenn ja, kann man dann in der config unterscheiden zwischen PV Ladung und Sofortladung?

andig, 2021-08-11:
Bitte oben lesen. Ladestart = neues Ticket. Nicht alles durcheinanderwerfen...