evcc-io/evcc #1326: Nissan Leaf 2021 - unexpected Status 401

tobintax, 2021-08-11:
**I have a strange Issue. We have a nissan leaf Ze1 from 2021. This is my vehicles section of evcc:** 

_# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- name: nissan
  type: nissan
  title: Leaf # display name for UI
  capacity: 39 # kWh
  user: xxxxx@xxx.xxx
  password: xxxxxx_

**Starting evcc stops suddenly:** 

_root@ubuntu:/tmp# evcc -c /tmp/evcc-config.yaml vehicle
SoC:        vehicle not available: login failed: unexpected status: 401
Capacity:   0kWh
Identifier: vehicle not available: login failed: unexpected status: 401_

**Changing the log level to "debug" and starting evcc, looks pretty:** 

_root@ubuntu:/tmp# evcc -c /tmp/evcc-config.yaml vehicle
[main  ] INFO 2021/08/11 19:10:30 evcc 0.61 (02c8729)
[main  ] INFO 2021/08/11 19:10:30 using config file /tmp/evcc-config.yaml
[nissan] DEBUG 2021/08/11 19:10:33 found vehicle: SJNFAAZE1Uxxxxxxx
......SoC:     90%
Charge status: A
Capacity:      39kWh
Range:         244km
Finish time:   not available_

tobintax, 2021-08-11:
Was heißt das? Sorry, ich bin kein echter Entwickler und nur selten auf Github.

andig, 2021-08-11:
Der Fehler von Nissan ist behoben, aber noch in keinem Release- das wird diesmal auch noch ein bisschen dauern. Da hilft nur selbst compilieren :/

andig, 2021-08-11:
Als workaround könntest Du das Fzg per `type: cloud` abfragen, da konnte ich den Fix schon ausrollen. Braucht einen Sponsortoken...

tobintax, 2021-08-11:
ok, danke für die Erklärung und danke fürs fixen!