evcc-io/evcc #1333: PSA config example

mamaretti, 2021-08-12:
Hi Andi 

can you share a PSA-config-example?

`vehicles:
- name: peugeot
  type: peugeot
  title: 208
  capacity: 50 # kWh
  user: xxx
  password: xxx
  vin: xxx`

gives error:
SoC:        vehicle not available: missing client id and secret (see https://github.com/flobz/psa_car_controller)
Capacity:   0kWh
Identifier: vehicle not available: missing client id and secret (see https://github.com/flobz/psa_car_controller)

`vehicles:
- name: peugeot
  type: peugeot
  title: 208
  capacity: 50 # kWh
  user: xxx
  password: xxx
  vin: xxx
  ID: 1eebc2d5-5df3-459b-a624-20abfcf82530
  Secret: T5tP7iS0cO8sC0lA2iE2aR7gK6uE5rF3lJ8pC3nO1pR7tL8vU1`

gives 
* '' has invalid keys: ID, Secret

evcc version 0.61 (02c8729)

mamaretti, 2021-08-12:
my fault

ist working with 

`vehicles:

name: peugeot
type: peugeot
title: 208
capacity: 50 # kWh
user: xxx
password: xxx
vin: xxx
ClientID: 1eebc2d5-5df3-459b-a624-20abfcf82530
ClientSecret: T5tP7iS0cO8sC0lA2iE2aR7gK6uE5rF3lJ8pC3nO1pR7tL8vU1`

result 
`SoC:            65%
Charge status:  not available
Capacity:       50kWh
Range:          184km
Finish time:    not available
Climate active: false
Outside temp:   20.0°C
Target temp:    20.0°C`

mamaretti, 2021-08-12:
but config is not working without
  ClientID: 1eebc2d5-5df3-459b-a624-20abfcf82530
  ClientSecret: T5tP7iS0cO8sC0lA2iE2aR7gK6uE5rF3lJ8pC3nO1pR7tL8vU1

andig, 2021-08-12:
For current master you don't need id/secret at all.

closed 2021-08-12 by andig