evcc-io/evcc #1334: Tesla + Wallbe: charger logic error: disabled but charging

hopfi2k, 2021-08-12:
Charging a Tesla with a Wallbe charger works fine.

Changing mode while charging to "Stop" gives me a: `charger logic error: disabled but charging` warning which I don't understand.

This is my EVCC.yaml:
```
chargers:
- name: wallbe
  type: wallbe
  uri: 192.168.1.83:502

meters:
- name: pv-dc-kostal
  type: default
  power:
    type: calc
    add:
    - type: modbus
      model: sunspec
      value: 160:1:DCW # string 1
      uri: 192.168.1.170:1502
      id: 71
    - type: modbus
      model: sunspec
      value: 160:2:DCW # string 2
      uri: 192.168.1.170:1502
      id: 71
- name: netz
  type: modbus
  model: ksem
  uri: 192.168.1.27:502
  id: 1
  power: Power
  energy: Export

vehicles:
- name: tesla
  type: tesla
  title: 100D
  capacity: 100	# Akkugröße in kWh
  tokens:
    access: <removed>
    refresh: <removed>
  cache: 1m

loadpoints:
- title: Garage
  vehicle: tesla
  charger: wallbe
  phases: 3 # ev phases (default 3)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

site:
- title: Zuhause
  meters:
    grid: netz
    pv: pv-dc-kostal
```

I'd like to understand where the logic error here is?

andig, 2021-08-12:
The warning may not make much sense. Can you share a logfile (level: debug)?

hopfi2k, 2021-08-12:
Sure - just made one (see attachment)

One other observation: changing mode to STOP doesn't actually stop charging the connected Tesla. It only stops charging when ending charging from the Tesla-App...

[debug.log](https://github.com/andig/evcc/files/6976614/debug.log)

premultiply, 2021-08-12:
DIP 10 switched to allow remote control?

hopfi2k, 2021-08-12:
Yes

premultiply, 2021-08-12:
Are you sure you are are still running a Wallbe Firmware on the Phoenix OEM Controller or did you "upgrade" to original Phoenix Firmware?

hopfi2k, 2021-08-12:
100% Original. Brand new Wallbe Box, taken out of the box today, no Firmware upgrade or switch to the Original Phoenix Firmware done.

Only thing I did was to enable DHCP in network settings. Everything else is stock.

andig, 2021-08-12:
> One other observation: changing mode to STOP doesn't actually stop charging the connected Tesla. It only stops charging when ending charging from the Tesla-App...

Benutzt Du evcc erst seit kurzem oder ist das ein neues Problem?

andig, 2021-08-12:
Dein Problem ist hier:

    [lp-1  ] DEBUG 2021/08/12 16:18:06 charger disable
    [lp-1  ] DEBUG 2021/08/12 16:18:13 charger status: C
    [cache ] DEBUG 2021/08/12 16:18:13 enabled: false
    [lp-1  ] WARN 2021/08/12 16:18:13 charger logic error: disabled but charging

Wie kann eine abgeschaltete Wallbox laden? Kann ich Dir nicht sagen, aber ist offensichtlich falsch!

hopfi2k, 2021-08-12:
Hmm..einleuchtend. Aber das verrückte ist, dass das Auto ja lädt (wunderbar mit PV Überschuss und perfekter Regelung des Ladestroms in Abhängigkeit vom Überschuss) - auch wenn im Log steht, dass die "disabled" ist...

hopfi2k, 2021-08-13:
Simple Lösung: der Schlüsselschalter an der Wallbe muss auf "0" stehen - dann geht's...🙈

andig, 2021-08-13:
Also halten wir fest: wenn evcc solche Warnungen ausgibt liegt das Problem beim Anwender ;)

hopfi2k, 2021-08-13:
Touché ;-)

closed 2021-08-13 by andig