evcc-io/evcc #1335: Add plausi check for positive enable threshold

andig, 2021-08-12:
Anhand des Logfiles könnte es ein Problem mit dem Enable Threshold sein. Hier konfiguriert auf 300 W:

```
enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 300 # minimum export power (W). If zero, export must exceed minimum charge power to enable
```

Sobald die Site Power unter 300 W fällt, wird der Enable Timer gestartet und nach einer Minute dann das Laden aktiviert (Site Power solange unter 300 W):

```
Aug 10 01:50:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:50:15 max charge current: 0A = 0A + -0.62A (285W @ 2p)
Aug 10 01:50:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:50:15 site power 285W < enable threshold 300W
Aug 10 01:50:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:50:15 pv enable timer remaining: 50s
...
Aug 10 01:51:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:51:15 max charge current: 0A = 0A + -0.62A (284W @ 2p)
Aug 10 01:51:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:51:15 site power 284W < enable threshold 300W
Aug 10 01:51:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:51:15 pv enable timer elapsed
Aug 10 01:51:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:51:15 pv max charge current: 6A
Aug 10 01:51:15 smarthomepi evcc[30250]: [lp-1  ] DEBUG 2021/08/10 01:51:15 charger enable
```

Also wird bei unter 300 W Netzbezug der Enable Timer schon getriggert, wobei laut Beschreibung des Parameters man von 300 W Einspeisung ausgehen sollte:

> ... export must exceed minimum charge power to enable

Bestätigt auch die entsprechende Stelle im Code:

https://github.com/andig/evcc/blob/7ac1a9a962372ed2b18632cb4360cdcc16cd6cd1/core/loadpoint.go#L829

Demnach müsste das Vorzeichen vom Enable Threshold negativ sein, damit es korrekt funktioniert, oder?

_Originally posted by @rivengh in https://github.com/andig/evcc/discussions/1318#discussioncomment-1163971_

andig, 2021-08-12:
Eigentlich haben wir die Warnung sogar:

		log.WARN.Printf("PV mode enable threshold (%.0fW) is larger than disable threshold (%.0fW)", lp.Enable.Threshold, lp.Disable.Threshold)

Lesson learned: bitte komplette Logfile liefern!

closed 2021-08-12 by andig

rivengh, 2021-08-12:
Die Warnung wird aber wohl in dem beschriebenen Fall  nicht ausgelöst, da der Disable Threshold auf 500 W konfiguriert ist:
```
disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 500 # maximum import power (W)
```
 
Und daher immer noch größer und die Condition für die Warning false:

https://github.com/andig/evcc/blob/7ac1a9a962372ed2b18632cb4360cdcc16cd6cd1/core/loadpoint.go#L209

premultiply, 2021-08-12:
Ich finde die Warnung und den Parameter selbst auch schwer verständlich beschrieben. Ich glaube ein knappes Beispiel könnte es besser erklären.
Und "lager" sollte vermutlich eher "higher" oder "lower" sein.

andig, 2021-08-13:
Fixed in https://github.com/andig/evcc/commit/e6f4ba02295e2ca90790d480fdf95a9af432c715

rivengh, 2021-08-13:
Im Configfile erschließt sich meines Erachtens für den Nutzer noch nicht, dass ein negativer Wert für den Enable Threshold bei Einspeisung (export power) erwartet wird:

https://github.com/andig/evcc/blob/821886637777c3dc197c686e0c3aa9a14fc656b5/evcc.dist.yaml#L99-L101

Vielleicht könnte man den Kommentar noch etwas anpassen, zum Beispiel so:

`threshold: -0 # minimum export power (W), as negative value. If zero, export must exceed minimum charge power to enable.`