evcc-io/evcc #1336: Vehicle detection does not work for Skoda

bicaluv, 2021-08-12:
**Describe the bug**
With the changes introduced with "Fix vehicle detection across multiple loadpoints #1308", my vehicle does not get set and the SoC etc. ist not correctly displayed in the UI.

**To Reproduce**
Draw change #1308 from Git and got the error in the UI. after changing the following lines back to the state before, everything works fine again:
```
	if len(lp.vehicles) > 0 {
		lp.setActiveVehicle(lp.vehicles[0])
		// associate first vehicle if it cannot be auto-detected
		// if _, ok := lp.vehicles[0].(api.ChargeState); !ok {
		//	lp.setActiveVehicle(lp.vehicles[0])
		// }

		lp.startVehicleDetection()
	}
```

**Expected behavior**
Correct read out of vehicle.

**EVCC details:**
UI shows always the same connected vehicle.
Wrong:
[image]

Correct:
[image]

andig, 2021-08-12:
Config?

bicaluv, 2021-08-12:
Klar.
Brauchst du noch Config und/oder Log?

PR #1337 (2021-08-12): Fix vehicle connection status not displayed
Fix #1336
closed 2021-08-12 by andig