evcc-io/evcc #1347: go-e Charger liefert "charger: alw update failed: 0" und weitere Fehler

bicaluv, 2021-08-13:
**Describe the bug**
Wird das Laden von evcc gestartet kommt es mit dem aktuellen Git-Stand (13.08.2021) zu den folgenden Fehlermeldungen, die mit der Original 0.61 Version nicht vorhanden waren (siehe Log):
```
...
Aug 13 12:06:07 raspberrypi evcc[15572]: [lp-1  ] ERROR 2021/08/13 12:06:07 max charge current 6: amp update failed: 13
Aug 13 12:06:07 raspberrypi evcc[15572]: [lp-1  ] ERROR 2021/08/13 12:06:07 amp update failed: 13
...
Aug 13 12:06:08 raspberrypi evcc[15572]: [lp-1  ] ERROR 2021/08/13 12:06:08 charger enable: alw update failed: 0
Aug 13 12:06:08 raspberrypi evcc[15572]: [lp-1  ] ERROR 2021/08/13 12:06:08 alw update failed: 0
```

Interessanterweise funktioniert das Aktivieren und Deaktivieren, des Chargers.

**To Reproduce**
Die Fehler kommen nur dann, wenn der Charger angesteuert wird (klar)

**Expected behavior**
Keine Fehler :-)

**EVCC details:**
Show output of <code>evcc -v</code>:
```
0.61 + aktuellen Git Hub Code

```

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>

uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

# log settings
log: error
levels:
  core: error
  lp-1: error

# meter definitions
meters:
- name: sma-grid
  type: sma
  uri: XX.XX.XX.XX0

- type: sma
  name: sma-pv
  uri: XX.XX.XX.XX1
  password: "123456"

hems:
  type: shm
  allowcontrol: false

# charger definitions
chargers:
- name: go-eCharger
  type: go-e
  token: 1234567 # go-e cloud API token
  cache: 10s # go-e cloud API cache duration

# vehicle definitions
vehicles:
- name: Citigo
  type: skoda
  title: CITIGOe iV
  capacity: 36 # kWh
  user: xxxx
  password: 123456
  vin: 345678

# site describes the EVU connection, PV and home battery
site:
  title: Haus # display name for UI
  meters:
    grid: sma-grid # grid meter
    pv: sma-pv # pv meter

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: go-eCharger # charger
  vehicle: Citigo
  mode: pv
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: charging
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 30 # immediately charge to 30% regardless of mode unless "off" (disabled)
    target: 80 # always charge to 80%
    estimate: true # set true to interpolate between api updates
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 80 # charge to 80%
  phases: 2 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 10m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
See attachment
[syslog-2.zip](https://github.com/andig/evcc/files/6981769/syslog-2.zip)

</pre></details>

andig, 2021-08-13:
Liegt das evtl. an https://github.com/andig/evcc/pull/1292? Der wurde allerdings schon vor 2 Tagen gemerged und sah im Test gut aus?

bicaluv, 2021-08-13:
Ich schau mir die Änderungen mal an und setzte nur diese zurück, mal schauen...

andig, 2021-08-13:
Und was für einen GoE hast du egtl- v3? Der scheint eh zu spinnen...

sebnaf, 2021-08-13:
Das klingt identisch. Delay muss auf über 7s sein.
Ja, betrifft nur Hardware v3 Charger.

premultiply, 2021-08-13:
Läuft super bei go-e...

sebnaf, 2021-08-13:
Wer ne Idee für eine bezahlbare und ähnlich kleine/unauffällige, funktionierende WB hat… gerne her damit.

premultiply, 2021-08-13:
Alles schwierig. Vielleicht noch SmartWB?
Wallbe Eco 2.0+Zähler, Tinkerforge Warp Charger Smart+gescheiter Zähler? Keba P30? Heidelberg EC? Alfen? cFos?
Wer weiss...

sebnaf, 2021-08-13:
Mmmh. Ich hoffe darauf, dass go-e das bald behebt. Die v2 kann das ja alles. Dann ist es schon ein durchaus interessantes Produkt.

bicaluv, 2021-08-14:
> Und was für einen GoE hast du egtl- v3? Der scheint eh zu spinnen...

v2.
Ich habe nun meine gesamte Infrastruktur neu gebooten (auch die Router und die WB). Der Fehler ist weg. Bitte entschuldig den damit verbundenen Aufwand.

bicaluv, 2021-08-14:
Misst, zu früh gefreut!
Also, die Fehler
```
amp update failed: 8
max charge current 9: amp update failed: 8
```
kommt dann, wenn ich die Änderungen der mit #1292 gemachten Commits drin lasse und die WB geregelt werden soll. Setzte ich den Code in go-e.go auf Vor-#1292 zurück, kommen die Fehler nicht mehr.
Hinweis: ich nutzte die Token-basierte Variante zum Ansteuern der go-e.

bicaluv, 2021-08-14:
Reopen

andig, 2021-08-14:
Ich dreh den zurück, würde es aber gerne verstehen. Hättest du ein log trace?

bicaluv, 2021-08-14:
Warten mal noch etwas, ich bin gerade noch am Testen und auch mit zurückgesetzten #1292 kam der Fehler noch. Bin jetzt auf #1247 .
Welche Traces? Einfach nur DEBUG oder spezielles für lp-1 oder charger?

bicaluv, 2021-08-14:
Also bis jetzt kam der Fehler nicht mehr, trotz diverser Regeleingriffe. Bin jedoch auf dem Stand vor den #1247 Commit.
Welche Traces helfen dir mit dem aktuellen Stand?

andig, 2021-08-14:
Charger damit ich sehe was da hin und her geht

bicaluv, 2021-08-14:
Habe dir das Log direkt gesendet, dann muss ich es nicht "bereinigen".

andig, 2021-08-14:
Hast du gefunden welcher commit es kapitt gemacht hat?

bicaluv, 2021-08-14:
Es geht nur dann ohne die Fehler, wenn ich auf #1247 gehe. 
welche Commits da alles drin waren und welcher es genau war, habe ich noch nicht ermittelt

andig, 2021-08-14:
Der Fehler beim Cloud API aus https://github.com/andig/evcc/pull/1292 sollte mit https://github.com/andig/evcc/commit/5e667299839f455e20be118c5638ae7dd8563066 behoben sein und jetzt sogar ein paar weniger Abfragen machen.

closed 2021-08-14 by andig