evcc-io/evcc #1348: onIdentify für unterschiedliche Fahrzeuge?

hopfi2k, 2021-08-13:
Ich versuche an einer Wallbox (Loadpoint) `onIdentify ` für unterschiedliche Fahrzeug zu nutzen:

```
loadpoints:
- title: Garage
  vehicles:
  - tesla
  - id4
  charger: wallbe
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 85
    estimate: true
  onDisconnect:
    mode: pv
    targetSoC: 100
  onIdentify:
  - id4:
    mode: pv
    targetSoC: 100
  - tesla:
    mode: pv
    targetSoC: 85
  enable:
    delay: 1m
    threshold: 100
  disable:
    delay: 5m
    threshold: 1000
  phases: 3
  guardduration: 5m
  mincurrent: 6
  maxcurrent: 16
```

was aber zu folgender Fehlermeldung führt:
```
[main  ] FATAL 2021/08/13 11:58:58 failed configuring loadpoint: 2 error(s) decoding:
Aug 13 11:58:58 raspberrypi evcc[2285]: * 'onIdentify[0][<interface {} Value>]' expected a map, got 'int'
Aug 13 11:58:58 raspberrypi evcc[2285]: * 'onIdentify[0][<interface {} Value>]' expected a map, got 'string'
```

Der "Plan" war, für unterschiedliche (identifizierte) Fahrzeuge einen unterschiedlichen Target-SoC festzulegen...

andig, 2021-08-13:
Na dann mach das doch ;)

```
  onIdentify:
    id4:
    - mode: pv
    - targetSoC: 100
...
````

Geht aber aktuell nur wenn Dein Charger das Fahrzeug identifiziert. Für das Statusmatching ist es noch nicht implementiert.

hopfi2k, 2021-08-13:
Mach ich ja, aber funktioniert nicht:

loadpoints:
```
 ...
  onIdentify:
    id4:
    - mode: pv
    - targetSoC: 100
    tesla:
    - mode: pv
    - targetSoC: 85
  ...
```

gibt die Fehlermeldung: 

```
[main  ] FATAL 2021/08/13 12:14:26 failed configuring loadpoint: 2 error(s) decoding:
Aug 13 12:14:26 raspberrypi evcc[2570]: * 'onIdentify[<interface {} Value>]' expected a map, got 'slice'
Aug 13 12:14:26 raspberrypi evcc[2570]: * 'onIdentify[<interface {} Value>]' expected a map, got 'slice'
```

andig, 2021-08-13:
korrigiert

closed 2021-08-13 by andig