evcc-io/evcc #1353: go-e Warning "charger logic error: disabled but charging"

bicaluv, 2021-08-14:
**Describe the bug**
Ich nutze den go-e V2 Charger mit den Internet API (Token).
Startet das Laden (z.B. weil ich den Soll-SoC hochsetze) und beenden es wieder, kommt recht häufig die Warning "charger logic error: disabled but charging". Diese Warning kommt nur einmal.
Scheint ein Timing Problem zu sein (evtl. in Kombination mit dem beim Internet-API genutzten Cache) und dem Poll-Intervall.

**To Reproduce**
Laden starten.
Laden beenden.
Warning "charger logic error: disabled but charging"

**Expected behavior**
Keine Warnung

**EVCC details:**
Show output of <code>evcc -v</code>:
```
0.61 mit alle GitHub Änderungen bis 14.08.2021 (inkl. #1347 )

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
uri: 0.0.0.0:7070 # uri for ui

interval: 10s # control cycle interval

log settings

log: error

levels:

core: error

lp-1: error

meter definitions

meters:

name: sma-grid

type: sma

uri: XX.XX.XX.XX0

type: sma

name: sma-pv

uri: XX.XX.XX.XX1

password: "123456"

hems:

type: shm

allowcontrol: false

charger definitions

chargers:

name: go-eCharger

type: go-e

token: 1234567 # go-e cloud API token

cache: 10s # go-e cloud API cache duration

vehicle definitions

vehicles:

name: Citigo

type: skoda

title: CITIGOe iV

capacity: 36 # kWh

user: xxxx

password: 123456

vin: 345678

site describes the EVU connection, PV and home battery

site:

title: Haus # display name for UI

meters:

grid: sma-grid # grid meter

pv: sma-pv # pv meter

loadpoint describes the charger, charge meter and connected vehicle

loadpoints:

title: Carport # display name for UI

charger: go-eCharger # charger

vehicle: Citigo

mode: pv

soc:
polling defines usage of the vehicle APIs

Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery

or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.

poll:

# poll mode defines under which condition the vehicle API is called:

#   charging: update vehicle ONLY when charging (this is the recommended default)

#   connected: update vehicle when connected (not only charging), interval defines how often

#   always: always update vehicle regardless of connection state, interval defines how often

mode: charging

# poll interval defines how often the vehicle API may be polled if NOT charging

interval: 60m

min: 30 # immediately charge to 30% regardless of mode unless "off" (disabled)

target: 80 # always charge to 80%

estimate: true # set true to interpolate between api updates

onDisconnect: # set defaults when vehicle disconnects

mode: pv # switch back to pv mode

targetSoC: 80 # charge to 80%

phases: 2 # ev phases (default 3)

enable: # pv mode enable behavior

delay: 1m # threshold must be exceeded for this long

threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable

disable: # pv mode disable behavior

delay: 5m # threshold must be exceeded for this long

threshold: 200 # maximum import power (W)

guardduration: 10m # switch charger contactor not more often than this (default 10m)

mincurrent: 6 # minimum charge current (default 6A)

maxcurrent: 16 # maximum charge current (default 16A)

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
<- paste here

</pre></details>

andig, 2021-08-14:
Die spannende Frage ist, wie schnell nach einem Update wir den neuen Status in *aussagekräftiger Form* wieder abholen dürfen. Bisher haben wir den alten einfach behalten und nur so getan, was sicher falsch war. Eigentlich hätte es den Fehler schon immer geben müssen, zumindest seitdem die Warnung drin ist?

bicaluv, 2021-08-14:
Die ist auch schon länger drin, bin halt nie dazu gekommen mir die Zeit zu nehmen. Sorry.

bicaluv, 2021-08-14:
Idee: Zumindest so lange warten, wie der Cache-Interval bei Token API warten, bevor die Warning kommt.

andig, 2021-08-14:
Bleibt die Frage: wie lange ist "so lange" bis ein vernünftiger Status kommt?

bicaluv, 2021-08-14:
Go-e API Cache + 1s

andig, 2021-08-17:
@bicaluv wir sollten da vllt gar nichts mehr ändern sondern Du einfach die Zykluszeit hochdrehen bis sie langsam genug für den Go-E ist

premultiply, 2021-08-17:
Ja, bitte! Und alle Warte-Changes zurückdrehen. Aktives warten auf irgendwas kann nur eine Notlösung sein und halte ich hier für völlig unnötig.

bicaluv, 2021-08-17:
Bin bei euch. 
lassen wir das bleiben.

andig, 2021-08-17:
Manchmal ist die einfachste Lösung die Beste

closed 2021-08-17 by andig