evcc-io/evcc #1386: Potentielles Vehicle SoC Problem

DerAndereAndi, 2021-08-19:
Von @andig gemeldet https://github.com/andig/evcc/pull/1283#issuecomment-901645968 :

Bitte hier rein schauen: https://github.com/andig/evcc/blame/master/core/soc/estimator.go#L114

Da sind m.E. 2 Dinge fehlerhaft:

- wenn ein Fehler auftritt wird mit den Daten weiter gearbeitet, die sind aber nicht verlässlich
- der Fehler wird erst geloggt und dann zurück gegeben, so dass am Ende 2x geloggt wird, siehe Add 1p3p support #1213  im Logfile

DerAndereAndi, 2021-08-19:
@premultiply der Code war von dir wenn ich das richtig sehe. Was meinst du dazu?

premultiply, 2021-08-19:
Fehler heisst die betroffene Fahrzeug-API hat einen temporären Ausfall. Daher wird einfach mit dem Estimator weiter gemacht und es gibt temporär keine Neujustierung der linearen Steigungsrate. Die aktuelle Steigungsrate wird weiterhin beibehalten.
So sollte es funktionieren.

Tut es das nicht?

premultiply, 2021-08-19:
An Logmeldungen hänge ich wirklich nicht. Wenn du meinst dass es anders besser wäre: Nur zu.

DerAndereAndi, 2021-08-19:
@andig was sagst du?

andig, 2021-08-19:
> Tut es das nicht?

Ich denke das tut nicht weil der Soc- da auch der Fehler weiter gegeben wird- ignoriert wird. Außerdem gilt bei Go immer: Fehler behandeln. Logging ist eine Behandlung- aber dann muss man ihn nicht beim Aufrufer durch Ergebnis verwerfen nochmal behandeln.

closed 2021-08-22 by andig