evcc-io/evcc #1400: Go-E und ZOE

bacanol, 2021-08-21:
Hallo,

Vorab: ich finde das Projekt toll!

Ich habe es jetzt mit einer Go-E Ladestation und einer Zoe eingerichtet, habe aber festgestellt, dass EVCC bei etwa 1kw PV schon versucht mit ~ 4kw das Auto zu laden im PV Modus. Es flipfloppt immer zwischen den 6A und 7A hin und her: 
[image]
Und
[image]
Meine Frage, habe ich etwas falsch eingestellt oder ist das Verhalten bekannt?

Viele Grüße 😉 Nico

premultiply, 2021-08-21:
Irgendwas geht da mit 3P laden und 1P-Konfig wohl bei dir durcheinander.
Schalt mal L2 und L3 ab. Dann sollte es passen.

bacanol, 2021-08-21:
> Irgendwas geht da mit 3P laden und 1P-Konfig wohl bei dir durcheinander.
> Schalt mal L2 und L3 ab. Dann sollte es passen.

Ich habe folgende Config:
`loadpoints:
- title: Carport Rechts # display name for UI
  charger: mycharger # charger
#  meters:
#    charge: charge # charge meter
  vehicle: renault
  # vehicles: # use if multiple vehicles allowed to charge on this loadpoint
  # - ID.3
  # - e-Up
  mode: pv
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: connected
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    min: 30 # immediately charge to 0% regardless of mode unless "off" (disabled)
    target: 80 # always charge to 100%
    estimate: true # set true to interpolate between api updates
    levels: # target soc levels for UI
    - 30
    - 50
    - 80
    - 100
  onDisconnect: # set defaults when vehicle disconnects
    mode: pv # switch back to pv mode
    targetSoC: 80 # charge to 100%
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: 400 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)
`

Ich soll also hart die phases auf 1 setzen?

premultiply, 2021-08-21:
Das ist am einfachsten und normalerweise für den Hausgebrauch völlig ausreichend und normalerweise passend.
Wie groß ist deine PV? Über 30 kWp wird sie wohl nicht haben, oder?

Die ZOE kann man **dreiphasig** nicht sinnvoll mit niedrigen Phasenströmen laden. Ladeabbrüche und sehr hoher Blindleistungsanteil, der die Regelung erschwert. Das klappt erst bei über 10 kW gescheit.
Die `threshold`s würde ich rauswerfen und passend zu den abgeschalteten L2 und L3 `phases: 1` setzen.
Dann damit erstmal testen und weitere Erfahrungen sammeln.

Wenn die Wallbox und Elektrok es kann den max. Phasenstrom auf 20A setzen.

bacanol, 2021-08-21:
> Das ist am einfachsten und normalerweise für den Hausgebrauch völlig ausreichend und normalerweise passend.
> Wie groß ist deine PV? Über 30 kWp wird sie wohl nicht haben, oder?
> 
> Die ZOE kann man **dreiphasig** nicht sinnvoll mit niedrigen Phasenströmen laden. Ladeabbrüche und sehr hoher Blindleistungsanteil der die Regelung erschwert. Das klappt erst bei über 10 kW gescheit.
> Die `threshold`s würde ich rauswerfen und passend zu den abgeschalteten L2 und L3 `phases: 1` setzen.
> Dann damit erstmal testen und weitere Erfahrungen sammeln.
> 
> Wenn die Wallbox und Elektrik es kann den max. Phasenstrom auf 20A setzen.

Okay, habe es umgestellt. Allerdings hat sich nichts geändert, er schreibet im Log immer noch, dass er 3 Physik lädt:
`max charge current: 5.5A = 10.7A + -5.2A (3612W @ 3p)`

`max charge current: 10.7A = 6.6A + 4.1A (-2807W @ 3p)`

`detected phases: 3p [6.6 6.6 6.6]A`

Oder muss ich die stromseitig trennen?

premultiply, 2021-08-21:
Ja, die Phasen L2 und L3 zur Box abschalten. Auto vorher abstecken!

closed 2021-08-21 by andig