evcc-io/evcc #1430: EVCC regelt nicht mehr auf 0W (bzw. residualpower) sobald Wallbox-Current gelesen werden kann

Alestrix, 2021-08-24:
**Describe the bug**

Wenn EVCC nur die Leistung am Netzanschlusspunkt zur Verfügung hat, wird der an die Wallbox gesendete Soll-Strom korrekt geregelt, so dass am Netzübergang im Schnitt 0W bzw die eingestellte `residualpower` anliegt - im Bild unten Bereich (1) (`residualPower=100`).

Zugehöriges Log:

    [lp-1  ] DEBUG 2021/08/24 10:45:06 max charge current: 19.0A = 18.9A + 0.1A (-17W @ 1p)
    [lp-1  ] DEBUG 2021/08/24 10:45:06 pv max charge current: 19A
    [lp-1  ] DEBUG 2021/08/24 10:45:06 max charge current: 19
    [lp-1  ] DEBUG 2021/08/24 10:45:10 charge power: 4226W
    [lp-1  ] DEBUG 2021/08/24 10:45:10 detected phases: 1p (19.0A @ 4226W)
    [lp-1  ] DEBUG 2021/08/24 10:45:10 charger status: C
    [lp-1  ] DEBUG 2021/08/24 10:45:10 vehicle soc: 72%
    [lp-1  ] DEBUG 2021/08/24 10:45:10 vehicle range: 234km
    [lp-1  ] DEBUG 2021/08/24 10:45:12 max charge current: 19.0A = 19.0A + -0.0A (6W @ 1p)
    [lp-1  ] DEBUG 2021/08/24 10:45:12 pv max charge current: 19A
    [lp-1  ] DEBUG 2021/08/24 10:45:12 max charge current: 19
    [lp-1  ] DEBUG 2021/08/24 10:45:14 charge power: 4234W
    [lp-1  ] DEBUG 2021/08/24 10:45:14 detected phases: 1p (19.0A @ 4234W)
    [lp-1  ] DEBUG 2021/08/24 10:45:14 charger status: C
    [lp-1  ] DEBUG 2021/08/24 10:45:14 vehicle soc: 72%
    [lp-1  ] DEBUG 2021/08/24 10:45:14 vehicle range: 234km
    [lp-1  ] DEBUG 2021/08/24 10:45:16 max charge current: 18.9A = 19.0A + -0.0A (1W @ 1p)
    [lp-1  ] DEBUG 2021/08/24 10:45:16 pv max charge current: 18.9A
    [lp-1  ] DEBUG 2021/08/24 10:45:16 max charge current: 19
    [lp-1  ] DEBUG 2021/08/24 10:45:19 charge power: 4217W
    [lp-1  ] DEBUG 2021/08/24 10:45:19 detected phases: 1p (18.9A @ 4217W)
    [lp-1  ] DEBUG 2021/08/24 10:45:19 charger status: C
    [lp-1  ] DEBUG 2021/08/24 10:45:19 vehicle soc: 72%
    [lp-1  ] DEBUG 2021/08/24 10:45:19 vehicle range: 234km

Sobald aber ein Charger in der Konfiguration definiert ist, der EVCC den tatsächlich zum Auto fließenden Strom mitteilt, wird nicht mehr korrekt geregelt (Bereich (2), `residualPower=100`)

Zugehöriges Log:

    [lp-1  ] DEBUG 2021/08/24 10:46:39 max charge current: 18.0A = 17.1A + 0.9A (-205W @ 1p)
    [lp-1  ] DEBUG 2021/08/24 10:46:39 pv max charge current: 18A
    [lp-1  ] DEBUG 2021/08/24 10:46:39 max charge current: 18
    [lp-1  ] DEBUG 2021/08/24 10:46:42 charge power: 4036W
    [lp-1  ] DEBUG 2021/08/24 10:46:42 charge currents: [17.1 0.05 0.02]A
    [lp-1  ] DEBUG 2021/08/24 10:46:42 detected phases: 1p [17.1 0.05 0.02]A
    [lp-1  ] DEBUG 2021/08/24 10:46:42 charger status: C
    [lp-1  ] DEBUG 2021/08/24 10:46:42 vehicle soc: 73%
    [lp-1  ] DEBUG 2021/08/24 10:46:42 vehicle range: 238km
    [lp-1  ] DEBUG 2021/08/24 10:46:44 max charge current: 18.0A = 17.1A + 0.9A (-206W @ 1p)
    [lp-1  ] DEBUG 2021/08/24 10:46:44 pv max charge current: 18A
    [lp-1  ] DEBUG 2021/08/24 10:46:44 max charge current: 18
    [lp-1  ] DEBUG 2021/08/24 10:46:47 charge power: 4021W
    [lp-1  ] DEBUG 2021/08/24 10:46:47 charge currents: [17 0.05 0.02]A
    [lp-1  ] DEBUG 2021/08/24 10:46:47 detected phases: 1p [17 0.05 0.02]A
    [lp-1  ] DEBUG 2021/08/24 10:46:47 charger status: C
    [lp-1  ] DEBUG 2021/08/24 10:46:47 vehicle soc: 73%
    [lp-1  ] DEBUG 2021/08/24 10:46:47 vehicle range: 238km
    [lp-1  ] DEBUG 2021/08/24 10:46:49 max charge current: 18.0A = 17.0A + 1.0A (-221W @ 1p)
    [lp-1  ] DEBUG 2021/08/24 10:46:49 pv max charge current: 18A
    [lp-1  ] DEBUG 2021/08/24 10:46:49 max charge current: 18
    [lp-1  ] DEBUG 2021/08/24 10:46:52 charge power: 4019W
    [lp-1  ] DEBUG 2021/08/24 10:46:52 charge currents: [17 0.05 0.02]A
    [lp-1  ] DEBUG 2021/08/24 10:46:52 detected phases: 1p [17 0.05 0.02]A
    [lp-1  ] DEBUG 2021/08/24 10:46:52 charger status: C
    [lp-1  ] DEBUG 2021/08/24 10:46:52 vehicle soc: 73%
    [lp-1  ] DEBUG 2021/08/24 10:46:52 vehicle range: 238km

[image]
Das Bild zeigt die Leistung am Netzabschlusspunkt mit unterschiedlichen evcc Konfigurationen.

**To Reproduce**

Man benötigt ein Auto, dass sich nicht exakt an die Vorgabe durch die Wallbox hält (das dürften wohl die meisten Autos sein, bei Zoe ist es aber wohl ein Extremfall) und einen Wallbox-Charger, der die Ströme liefern kann.
Dann beobachtet man die sitePower einmal mit konfigurierter Wallbox-Strommessung und mal ohne.
[... 150 lines cut ...]
    threshold: 0 # minimum grid power in Watts. Negative values denote export. If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 460 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 8 # minimum charge current (default 6A)
  maxcurrent: 21 # maximum charge current (default 16A)

mqtt:
  broker: pi:1883
  topic: evcc # root topic for publishing, set empty to disable

influx:
  url: http://pi:8086
  database: evcc

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
siehe oben

</pre></details>

**Mitigation**
Es könnte ausreichen, EVCC nicht über die Ströme zu informieren (wie oben durch die Auskommentierung getan). Ich kann mir aber vorstellen, dass z.B. die automatische Unterscheidung zwischen 1P und 3P zuverlässiger funktioniert, wenn EVCC Zugriff auf die Ströme hat. Ob EVCC die Wallbox-Ströme noch für andere Zwecke nutzen kann, ist mir nicht bekannt.

andig, 2021-08-24:
Mit anderen Worten: wenn ein Fahrzeug sich nicht Normkonform verhält liegt die Regelung um den Betrag der Abweichung daneben. 

Warum regeln wir auf Ist-Ströme: weil dadurch ein Aufschwingen der Regelung vermieden wird. Phantomabweichungen werden immer auf Basis der Echtwerte bewertet.

Einfacher Workaround: mit ´ResidualPower` kompensieren.

/cc @premultiply wie siehst Du das Thema Regelung?

Alestrix, 2021-08-24:
Danke für den Change!

Ein Kommentar zur Normkonformität sei mir noch erlaubt: meines Wissens wird über das PWM Signal dem Fahrzeug der maximal erlaubte Strom kommuniziert. Ob das Fahrzeug diesen voll ausnutzt oder noch einen Sicherheitspuffer abzieht, ist wahrscheinlich im Ermessen des OBCs.

Alestrix, 2021-08-24:
Versehentlich etwas zu früh geschlossen

premultiply, 2021-08-25:
Ich kann mich daran erinnern dass wir die implementierte Variante als das kleinere Übel angesehen hatten und damals schon klar war, dass es Probleme geben wird wenn eine größere Diskrepanz zwischen gewünschter Wirkleistung und tatsächlich durch den Ladepunkt geregelter Scheinleistung besteht.

Das ist bei der ZOE im 3P-Modus und zu niedrigen Ladestromvorgaben der Fall.

Alestrix, 2021-08-25:
In #877 steht leider nicht, warum man damals diese Entscheidung getroffen hat.

Ich verstehe, wenn aus irgendeinem Grund das Auto bei einem niedrigeren Strom "stehen bleibt" als möglich, wird der eingestellte Strom mit jedem Regelzyklus immer höher gehen bis zum konfigurierten `maxcurrent`. Wenn dann aus welchem Grund auch immer später der OBC entscheidet, nun doch der Vorgabe zu folgen (vielleicht weil die Batterie nun die gewünschte Temperatur hat oder warum auch immer), dann geht der Bezug plötzlich durch die Decke.

Daher mein Vorschlag in #1431,  den eingestellten Current und den gemessenen nie mehr als einen bestimmten Wert auseinanderlaufen zu lassen. @andig meinte, dadurch könnte die Regelung für manche Fahrzeuge verlangsamt werden, das kann ich aber bisher nicht nachvollziehen. In meinen Gedankenspielen/"Schreibtischtests" ist eher das Gegenteil der Fall. @andig, kannst Du einen Beispielablauf skizzieren, wann die Regelung durch den Vorschlag verlangsamt wird?

premultiply, 2021-08-25:
Ich glaube Andi hat das im Eifer des Gefechts missverstanden und mit einem Ramping verwechselt welches pro Regelzyklus die Schrittweite limitiert. Oder?

Dein Vorschlag wäre vielleicht durchaus mal noch genauer zu betrachten und in der Praxis zu testen. Er würde sozusagen einen limitierten Overshoot erlauben.

Allerdings muss man da viele Aspekte betrachten und abwägen, insbesondere wenn es um das (sichere) Verhalten bei mehreren Ladepunkten geht, insbesondere wenn dort zukünftig Lastmanagementfunktionen greifen sollten.

Alestrix, 2021-08-25:
> Allerdings muss man da viele Aspekte betrachten und abwägen, insbesondere wenn es um das (sichere) Verhalten bei mehreren Ladepunkten geht, insbesondere wenn dort zukünftig Lastmanagementfunktionen greifen sollten.

D.h. man müsste den "Overshoot" Wert - in meinem Beispiel 3A - konfigurierbar halten und in dem von Dir genannten Fall sicherheitshalber auf 0A einstellen. Dann wird als effectiveCurrent immer der kleinere von eingestelltem und gemessenem Current genommen. Das müsste ja eigentlich immer der gemessene sein (d.h. identisch mit bisherigem Verhalten), da das Auto ja nie mehr ziehen darf als eingestellt.

Alestrix, 2021-08-25:
> Das ist bei der ZOE im 3P-Modus und zu niedrigen Ladestromvorgaben der Fall.

Leider ist das nicht darauf beschränkt. Die Logs oben zeigen eine 1P Ladung mit 17A (18A eingestellt).

andig, 2021-08-25:
Hier geht es erstmal darum ob wir auf Soll- oder Ist-Strom regeln. Alles Andere ist ein weiteres Thema.

Alestrix, 2021-08-28:
Danke @andig für den Change und dass Du bereit bist, das durchzutesten! Hier mal ein erster Bericht.

Ich habe die latest (mit konfigurierter Wallbox Strommessung) verglichen mit der 0.61 (die hatte ich hier zuvor am laufen) mal ohne und mal mit konfigurierter Wallbox-Strommessung in der evcc.yaml. WB-Leistungsmessung war immer konfiguriert.

[image]

Es verhalten sich wie erwartet latest und 0.61 ohne Strommessung gleich, während 0.61 mit Strommessung zu viel PV Strom ungenutzt lässt. Davon abgesehen ist noch auffällig, dass die 0.61+Strom - Linie wesentlich ruhiger läuft als die anderen beiden, wo die Regelung auf Basis des Soll-Stroms für wesentlich mehr Unruhe sorgt. Das hätte ich so nicht erwartet, würde für mich aber kein Problem darstellen. Weiß nicht, wie das andere sehen.

In allen Fällen habe ich auf dem MQTT Bus gemonitort, dass auch wirklich mA-genaue Ströme an die WB gesendet werden, was auch der Fall war (das Log lässt erst mal anderes vermuten: `pv max charge current: 14.4A` gefolgt von `max charge current: 14`)

Für einen Test mit 3P ist leider nicht genug Sonne da...

So viel von mir. Jetzt wäre noch interessant, wie das bei anderen aussieht.

Viele Grüße
 Alex

PS: Die Ruhigere Regelung bei "0.61+Strom" mag auch daran liegen, dass in dem Moment die PV-Erzeugung wesentlich konstanter war. Etwas später sah es dann so aus:
[image]

andig, 2021-08-28:
> Davon abgesehen ist noch auffällig, dass die 0.61+Strom - Linie wesentlich ruhiger läuft als die anderen beiden, wo die Regelung auf Basis des Soll-Stroms für wesentlich mehr Unruhe sorgt. Das hätte [...]

Danke für die Info. Wird es ruhiger, wenn Du das Regelintervall verlängerst?

Wir können das ja immer noch zurück drehen wenn es keine gesamtheitlich überzeugende Lösung gibt.

andig, 2021-08-28:
> In allen Fällen habe ich auf dem MQTT Bus gemonitort, dass auch wirklich mA-genaue Ströme an die WB gesendet werden, was auch der Fall war (das Log lässt erst mal anderes vermuten: pv max charge current: 14.4A gefolgt von max charge current: 14)

Das sieht blöd aus- gerne neues Ticket.

Alestrix, 2021-08-28:
> Danke für die Info. Wird es ruhiger, wenn Du das Regelintervall verlängerst?

Da auch bei einem 15s Intervall immer nur der Momentanwert und kein Durchschnitt genommen wird, glaube ich aber eigentlich nicht an eine Glättung. Die Unruhe stört mich aber eigentlich nicht und es ist schon toll, wie gut die Regelung den Strom ausregelt:
[image]

Ich setze es aber trotzdem mal auf 15s, wobei mir die schnelle Reaktion schon gut gefällt. Wegen Schei$$wetter kann ich aber jetzt nicht mehr testen.

andig, 2021-08-28:
Mein Punkt ist: wir optimieren nicht auf Deinen Einzelfall, sondern auf die Masse.

Alestrix, 2021-08-28:
> Mein Punkt ist: wir optimieren nicht auf Deinen Einzelfall, sondern auf die Masse.

Das ist schon klar, aber wenn es meinen (*) Fall löst, ohne andere Probleme zu erzeugen, dann spricht ja nichts dagegen. Wie schon erwähnt, wurde etwas später die "legacy" Variante ja auch unruhig. Wenn man dann noch den erlaubten "Schlupf" konfigurierbar machen würde, könnte man durch Setzen des selben auf 0 wieder das vorherige Verhalten herbeiführen. Wenn mich Frau und Kinder mal ausreichend lange los lasen, würde ich mich auch mal an einem PR für die Konfigurierbarkeit heranwagen. Fragt sich halt, wann das mal der Fall sein wird :-/

(*) Wie schon an anderer Stelle erwähnt, gibt es keine Norm, die dem Auto vorschreibt, den gesamten zur Verfügung gestellten Strom auszunutzen. Daher denke ich nicht, dass meine Zoe ein Einzelfall ist. Auch ist vielleicht bei einigen der Effekt nur geringer und daher die Regelung zwar nicht so genau, aber genau genug.

Alestrix, 2021-08-30:
Keine "Beruhigung" durch 15s Intervall:
[image]

Für mehr Tests haben leider Zeit und Sonne nicht gereicht. Ich nehme gerne Vorschläge für weitere Tests an. Z.B. wollte ich auch die Glättung per `decay` probieren, habe es aber nicht zum Laufen gebracht ("[main  ] FATAL 2021/08/30 15:43:03 cannot create meter 'grid': cannot create meter 'custom': 1 error(s) decoding: * '' has invalid keys: decay").

andig, 2021-09-02:
Bleibt die Frage so lassen oder zurück drehen?

Alestrix, 2021-09-02:
Ich hab da nen Bias (bei mir läuft's mit dem Change ja besser), daher sollte das jemand anderes beantworten. @premultiply was denkst Du?

premultiply, 2021-09-02:
Das müsste mal jemand mit mehreren Ladepunkten bei paralleler Nutzung testen. Mit einem LP mit schneller und feingranularer PWM-Anpassung funktioniert es gut bzw. sogar vielleicht sogar tatsächlich besser. Mit NRGkick oder anderen lahmen Gerätschaften wird man aber da wohl weniger Freude als zuvor dran haben.
Ich würde es aber jetzt erstmal so lassen.

Im Rahmen des Lastmanagements muss man das aber nochmal ganz genau betrachten und vermutlich auch nochmal nachjustieren. Aber da wird man um einen Zähler für die Ist-Leistung und die Ist-Ströme an jedem Punkt ohnehin nicht herum kommen wenn man es möglichst effizient machen will.

closed 2021-09-02 by andig

Alestrix, 2021-09-02:
Habe mich mal an einer Konfiguraitonsoption versucht - #1472 .
Durch Setzen von maxoffsetcurrent auf 0 kann man das alte Verhalten wiederherstellen.