evcc-io/evcc #1454: Clarify PV current log

andig, 2021-08-28:
> In allen Fällen habe ich auf dem MQTT Bus gemonitort, dass auch wirklich mA-genaue Ströme an die WB gesendet werden, was auch der Fall war (das Log lässt erst mal anderes vermuten: pv max charge current: 14.4A gefolgt von max charge current: 14)

Das sieht blöd aus- gerne neues Ticket.

_Originally posted by @andig in https://github.com/evcc-io/evcc/issues/1430#issuecomment-907614750_

Alestrix, 2021-08-28:
Es scheinen bei `max charge current` immer nur zwei gültige Ziffern angezeigt zu werden:

    [lp-1  ] DEBUG 2021/08/28 13:02:42 pv max charge current: 10.1A
    [lp-1  ] DEBUG 2021/08/28 13:02:42 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:02:48 pv max charge current: 10.2A
    [lp-1  ] DEBUG 2021/08/28 13:02:48 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:02:52 pv max charge current: 10.2A
    [lp-1  ] DEBUG 2021/08/28 13:02:52 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:02:58 pv max charge current: 10A
    [lp-1  ] DEBUG 2021/08/28 13:02:58 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:03:02 pv max charge current: 9.94A
    [lp-1  ] DEBUG 2021/08/28 13:03:02 max charge current: 9.9
    [lp-1  ] DEBUG 2021/08/28 13:03:07 pv max charge current: 10.2A
    [lp-1  ] DEBUG 2021/08/28 13:03:07 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:03:13 pv max charge current: 9.88A
    [lp-1  ] DEBUG 2021/08/28 13:03:13 max charge current: 9.9
    [lp-1  ] DEBUG 2021/08/28 13:03:17 pv max charge current: 9.99A
    [lp-1  ] DEBUG 2021/08/28 13:03:17 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:03:22 pv max charge current: 9.91A
    [lp-1  ] DEBUG 2021/08/28 13:03:22 max charge current: 9.9
    [lp-1  ] DEBUG 2021/08/28 13:03:27 pv max charge current: 9.93A
    [lp-1  ] DEBUG 2021/08/28 13:03:27 max charge current: 9.9
    [lp-1  ] DEBUG 2021/08/28 13:03:32 pv max charge current: 10.2A
    [lp-1  ] DEBUG 2021/08/28 13:03:32 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:03:37 pv max charge current: 10.1A
    [lp-1  ] DEBUG 2021/08/28 13:03:37 max charge current: 10
    
    [...]
    
    [lp-1  ] DEBUG 2021/08/28 13:22:24 pv max charge current: 11.6A
    [lp-1  ] DEBUG 2021/08/28 13:22:24 max charge current: 12
    [lp-1  ] DEBUG 2021/08/28 13:22:29 pv max charge current: 11.3A
    [lp-1  ] DEBUG 2021/08/28 13:22:29 max charge current: 11
    [lp-1  ] DEBUG 2021/08/28 13:22:34 pv max charge current: 11.3A
    [lp-1  ] DEBUG 2021/08/28 13:22:34 max charge current: 11
    [lp-1  ] DEBUG 2021/08/28 13:22:39 pv max charge current: 10.8A
    [lp-1  ] DEBUG 2021/08/28 13:22:39 max charge current: 11
    [lp-1  ] DEBUG 2021/08/28 13:22:44 pv max charge current: 10.8A
    [lp-1  ] DEBUG 2021/08/28 13:22:44 max charge current: 11
    [lp-1  ] DEBUG 2021/08/28 13:22:49 pv max charge current: 10.6A
    [lp-1  ] DEBUG 2021/08/28 13:22:49 max charge current: 11
    [lp-1  ] DEBUG 2021/08/28 13:22:55 pv max charge current: 10.4A
    [lp-1  ] DEBUG 2021/08/28 13:22:55 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:22:59 pv max charge current: 10.1A
    [lp-1  ] DEBUG 2021/08/28 13:22:59 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:23:04 pv max charge current: 10.1A
    [lp-1  ] DEBUG 2021/08/28 13:23:04 max charge current: 10
    [lp-1  ] DEBUG 2021/08/28 13:23:09 pv max charge current: 9.84A
    [lp-1  ] DEBUG 2021/08/28 13:23:09 max charge current: 9.8
    [lp-1  ] DEBUG 2021/08/28 13:23:14 pv max charge current: 9.43A
    [lp-1  ] DEBUG 2021/08/28 13:23:14 max charge current: 9.4
    [lp-1  ] DEBUG 2021/08/28 13:23:19 pv max charge current: 9.36A
    [lp-1  ] DEBUG 2021/08/28 13:23:19 max charge current: 9.4
    [lp-1  ] DEBUG 2021/08/28 13:23:24 pv max charge current: 9.38A
    [lp-1  ] DEBUG 2021/08/28 13:23:24 max charge current: 9.4
    [lp-1  ] DEBUG 2021/08/28 13:23:29 pv max charge current: 9.34A
    [lp-1  ] DEBUG 2021/08/28 13:23:29 max charge current: 9.3
    [lp-1  ] DEBUG 2021/08/28 13:23:34 pv max charge current: 8.96A
    [lp-1  ] DEBUG 2021/08/28 13:23:34 max charge current: 9
    [lp-1  ] DEBUG 2021/08/28 13:23:39 pv max charge current: 9.1A
    [lp-1  ] DEBUG 2021/08/28 13:23:39 max charge current: 9.1
    [lp-1  ] DEBUG 2021/08/28 13:23:44 pv max charge current: 9.08A
    [lp-1  ] DEBUG 2021/08/28 13:23:44 max charge current: 9.1
    [lp-1  ] DEBUG 2021/08/28 13:23:49 pv max charge current: 8.83A
    [lp-1  ] DEBUG 2021/08/28 13:23:49 max charge current: 8.8

Alestrix, 2021-08-28:
`%.2g` --> `%.2f`?

andig, 2021-08-31:
Closed in 7f2edde

closed 2021-08-31 by andig