evcc-io/evcc #1473: Target charging cannot be stopped anymore

Alestrix, 2021-09-03:
**Describe the bug**
With latest master, target charging is always enabled.

**To Reproduce**
Start latest master, try to remove target charging time - log shows still "target charging active" and GUI shows high grid power draw despite PV mode.

**Expected behavior**
Target charging can be switched off to get back "regular" behavior.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
evcc version 0.62 (3986b75)

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
config
======

grid: grid
----------
Power:          -3421W
Current L1..L3: 5.08A 4.15A 5.28A

pv: pv
------
Power: 3794W

loadpoint 1
===========

charge: warpmeter
-----------------
Power:          2W
Current L1..L3: 0.03A 0.01A 0.01A

charger: warp
-------------
Charge status: B
Enabled:       false

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>

[lp-1  ] WARN 2021/09/03 15:18:19 poll mode '{always 2h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] INFO 2021/09/03 15:18:19 vehicle updated: unknown -> Zoe
[lp-1  ] DEBUG 2021/09/03 15:18:21 max charge current: 7A
[lp-1  ] INFO 2021/09/03 15:18:21 loadpoint 1:
[lp-1  ] INFO 2021/09/03 15:18:21   mode:      pv
[lp-1  ] INFO 2021/09/03 15:18:21   charger:   power ✗ energy ✗ currents ✗ timer ✗
[lp-1  ] INFO 2021/09/03 15:18:21   meters:    charge ✓
[lp-1  ] INFO 2021/09/03 15:18:21     charge:  power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2021/09/03 15:18:21   vehicles:  ✓
[lp-1  ] INFO 2021/09/03 15:18:21     car 0:   range ✓ finish ✓ status ✓ climate ✓
[lp-1  ] DEBUG 2021/09/03 15:18:32 charge power: 2W
[lp-1  ] DEBUG 2021/09/03 15:18:32 charge currents: [6.96 6.94 6.95]A
[lp-1  ] DEBUG 2021/09/03 15:18:32 charger status: C
[lp-1  ] INFO 2021/09/03 15:18:32 start charging ->
[push  ] ERROR 2021/09/03 15:18:32 invalid message template for start: wanted: [socCharge], got: [vehiclePresent minCurrent siteTitle connectedDuration gridCurrents chargeCurrents minSoC vehicleCapacity title pvPower enabled batteryConfigured charging hasVehicle gridPower chargeDuration vehicleTitle chargeCurrent chargeConfigured gridConfigured phases chargedEnergy pvConfigured activePhases chargePower maxCurrent connected targetSoC mode]
[lp-1  ] DEBUG 2021/09/03 15:18:33 vehicle soc: 77%
[lp-1  ] DEBUG 2021/09/03 15:18:33 vehicle range: 274km
[lp-1  ] DEBUG 2021/09/03 15:18:35 target charging active for 0001-01-01 00:00:00 +0000 UTC: projected 2021-09-03 15:19:00 +0200 CEST (0s remaining)
[lp-1  ] DEBUG 2021/09/03 15:18:35 target charging: speedup (23A)
[lp-1  ] DEBUG 2021/09/03 15:18:35 max charge current: 23A
[lp-1  ] DEBUG 2021/09/03 15:18:35 charge power: 2W
[lp-1  ] DEBUG 2021/09/03 15:18:35 charge currents: [6.96 6.94 6.95]A
[lp-1  ] DEBUG 2021/09/03 15:18:35 detected phases: 3p [6.96 6.94 6.95]A
[lp-1  ] DEBUG 2021/09/03 15:18:35 charger status: C
[lp-1  ] DEBUG 2021/09/03 15:18:35 next soc poll remaining time: 1h59m57s
[lp-1  ] DEBUG 2021/09/03 15:18:35 vehicle soc: 77%
[lp-1  ] DEBUG 2021/09/03 15:18:35 vehicle range: 274km
[lp-1  ] DEBUG 2021/09/03 15:18:37 target charging: speedup (23A)
[lp-1  ] DEBUG 2021/09/03 15:18:38 charge power: 12471W
[lp-1  ] DEBUG 2021/09/03 15:18:38 charge currents: [17.1 17.1 17]A
[lp-1  ] DEBUG 2021/09/03 15:18:38 detected phases: 3p [17.1 17.1 17]A
[lp-1  ] DEBUG 2021/09/03 15:18:38 charger status: C
[lp-1  ] DEBUG 2021/09/03 15:18:38 next soc poll remaining time: 1h59m57s
[lp-1  ] DEBUG 2021/09/03 15:18:38 vehicle soc: 77%
[lp-1  ] DEBUG 2021/09/03 15:18:38 vehicle range: 274km
[lp-1  ] DEBUG 2021/09/03 15:18:40 target charging: speedup (23A)
[lp-1  ] DEBUG 2021/09/03 15:18:41 charge power: 15723W
[lp-1  ] DEBUG 2021/09/03 15:18:42 charge currents: [21.9 21.9 21.8]A
[lp-1  ] DEBUG 2021/09/03 15:18:42 detected phases: 3p [21.9 21.9 21.8]A
[lp-1  ] DEBUG 2021/09/03 15:18:42 charger status: C
[lp-1  ] DEBUG 2021/09/03 15:18:42 next soc poll remaining time: 1h59m56s
[lp-1  ] DEBUG 2021/09/03 15:18:42 vehicle soc: 77%
[lp-1  ] DEBUG 2021/09/03 15:18:42 vehicle range: 274km
[lp-1  ] DEBUG 2021/09/03 15:18:44 target charging: speedup (23A)
[lp-1  ] DEBUG 2021/09/03 15:18:46 charge power: 15717W
[lp-1  ] DEBUG 2021/09/03 15:18:47 charge currents: [21.9 21.9 21.8]A
[lp-1  ] DEBUG 2021/09/03 15:18:47 detected phases: 3p [21.9 21.9 21.8]A
[lp-1  ] DEBUG 2021/09/03 15:18:47 charger status: C
[lp-1  ] DEBUG 2021/09/03 15:18:47 next soc poll remaining time: 1h59m55s
[lp-1  ] DEBUG 2021/09/03 15:18:47 vehicle soc: 77%
[lp-1  ] DEBUG 2021/09/03 15:18:47 vehicle range: 274km
[lp-1  ] INFO 2021/09/03 15:18:48 set target charge: 85 @ 1970-01-01 01:00:00 +0100 CET

--> at 2021/09/03 15:18:48 another attempt at deleting target charging date was made

</pre></details>

bacanol, 2021-09-03:
Hello,

I have exactly the same issue. I activated target mode yesterday evening for today 7:30 which worked fine, but from that point on the evcc started to charge with full power not recognizing the pv mode nor the power coming from pv. It pulled grid power for the rest.

andig, 2021-09-04:
Will take a look tomorrow. Goal is to fix this before the next release

schenlap, 2021-09-04:
I have a similar problem, when starting evcc  charching target 0001-01-01 is set and charging starts immedialty with max power.
```
[site  ] DEBUG 2021/09/04 10:52:31 pv power: 1608W                                                                                                                                    
[site  ] DEBUG 2021/09/04 10:52:31 grid power: -564W                                                                                                                                  
[site  ] DEBUG 2021/09/04 10:52:31 site power: -564W       <-- this is below pv start power, charching should not start                                                                                                                           
[lp-1  ] DEBUG 2021/09/04 10:52:31 charge power: 0W                                                                                                                                   
[lp-1  ] DEBUG 2021/09/04 10:52:31 charge currents: [0 0 0]A                                                                                                                          
[lp-1  ] DEBUG 2021/09/04 10:52:31 charger status: B                                                                                                                                  
[lp-1  ] DEBUG 2021/09/04 10:52:32 vehicle soc: 22%                                                                                                                                   
[lp-1  ] DEBUG 2021/09/04 10:52:32 vehicle range: 5km                                                                                                                                 
[lp-1  ] DEBUG 2021/09/04 10:52:32 target charging active for 0001-01-01 00:00:00 +0000 UTC: projected 2021-09-04 10:53:00 +0200 CEST (0s remaining)                     <--- default startup value is long ago             
[lp-1  ] DEBUG 2021/09/04 10:52:32 target charging: speedup (16A)                                                                                                                     
[lp-1  ] DEBUG 2021/09/04 10:52:32 max charge current: 16A                                                                                                                            
[lp-1  ] DEBUG 2021/09/04 10:52:32 charger enable                   <--- we are above min soc, below pv enable threshold, charger should not start!                                                                                                                  
[site  ] DEBUG 2021/09/04 10:52:41 ----                                                                                                                                               
[site  ] DEBUG 2021/09/04 10:52:41 pv power: 1603W                                                                                                                                    
[site  ] DEBUG 2021/09/04 10:52:41 grid power: -699W                                                                                                                                  
[site  ] DEBUG 2021/09/04 10:52:41 site power: -699W                                                                                                                                  
[lp-1  ] DEBUG 2021/09/04 10:52:41 charge power: 2870W                                                                                                                                
[lp-1  ] DEBUG 2021/09/04 10:52:41 charge currents: [12.7 0 0]A                                                                                                                       
[lp-1  ] DEBUG 2021/09/04 10:52:42 charger status: C                                                                                                                                  
[lp-1  ] DEBUG 2021/09/04 10:52:42 charge start energy: 128kWh
[lp-1  ] INFO 2021/09/04 10:52:42 start charging ->
[lp-1  ] DEBUG 2021/09/04 10:52:42 soc estimated: 22.00% (vehicle: 22.00%)
[lp-1  ] DEBUG 2021/09/04 10:52:42 vehicle soc: 22%
[lp-1  ] DEBUG 2021/09/04 10:52:42 vehicle range: 5km
[lp-1  ] DEBUG 2021/09/04 10:52:42 target charging: speedup (16A)    <- charges with max power, but should be disabled. Power does not decrease over time
[site  ] DEBUG 2021/09/04 10:52:51 ----
[site  ] DEBUG 2021/09/04 10:52:51 pv power: 1604W
[site  ] DEBUG 2021/09/04 10:52:51 grid power: 2115W
[site  ] DEBUG 2021/09/04 10:52:51 site power: 2115W
[lp-1  ] DEBUG 2021/09/04 10:52:52 charge power: 3300W
[lp-1  ] DEBUG 2021/09/04 10:52:52 charge currents: [14.5 0 0]A
[lp-1  ] DEBUG 2021/09/04 10:52:52 detected phases: 1p [14.5 0 0]A
[lp-1  ] DEBUG 2021/09/04 10:52:52 charger status: C
[lp-1  ] DEBUG 2021/09/04 10:52:52 soc estimated: 22.13% (vehicle: 22.00%)
```

andig, 2021-09-05:
Zielladen ist momentan kaputt. Ich muss mir das nochmal ganz in Ruhe anschauen...

andig, 2021-09-09:
Ich mache mal zu. Das akute Problem ist behoben, Zielladen bekommt nochmal einen Neustart wenn wir soweit sind.

closed 2021-09-09 by andig