evcc-io/evcc #1492: decoupling the vehicle library part

frzifus, 2021-09-06:
**Is your feature request related to a problem? Please describe.**
I would like to use the bluelink part externally. It would be great if we could decouple the vehicle part of the repository. Here I see the `helper` and `logger` package as a problem.

**Describe the solution you'd like**
I would suggest removing the dependencies of all vehicle packages.

andig, 2021-09-06:
Ich sehe folgende Punkte:
- helper ist unkritisch. Warum stört das?
- Logger braucht eine Entscheidung, welches Logging wir brauchen: leveled oder rein ein einfacher Logger?
- Falls einfacher Logger (mein Favorit) müssen ein paar wenige DEBUG/WARN Statements entfallen. Zur not per error channel ersetzen
- Evcc muss dann im Kern bei leveled logging bleiben, Module bekommen nur den einfachen „trace“ Logger
- Entscheidung was in die Lib gehört (NewFromConfig gehört zu evcc, ebenso wie die Registry)
- Ordnerstruktur: alle Dateien in Unterordner stecken. Fahrzeug selbst jeweils in <brand>.go?
- Sponsortoken verleibt in evcc

Weitere Ideen? Dann müsste es nur noch jemand machen ;)

andig, 2021-09-07:
Hier mal eine Liste aller Einsatzbereiche eines leveled Loggers innerhalb der Fahrzeug-APIs:

    19 results - 18 files

    evcc • vehicle/audi.go:
    72  		if err == nil {
    73: 			log.Debugf("found vehicle: %v", cc.VIN)
    74  		}

    evcc • vehicle/bmw.go:
    53  		if err == nil {
    54: 			log.Debugf("found vehicle: %v", cc.VIN)
    55  		}

    evcc • vehicle/carwings.go:
    93  		if err := v.session.Connect(v.user, v.password); err != nil {
    94: 			log.Errorln("login failed:", err)
    95  		}

    evcc • vehicle/enyaq.go:
    71  		if err == nil {
    72: 			log.Debugf("found vehicle: %v", cc.VIN)
    73  		}

    evcc • vehicle/fiat.go:
    62  		if err == nil {
    63: 			log.Debugf("found vehicle: %v", cc.VIN)
    64  		}

    evcc • vehicle/ford.go:
    84  		if err == nil {
    85: 			log.Debugf("found vehicle: %v", v.vin)
    86  		}

    220  		if elapsed := time.Since(lastUpdate); err == nil && elapsed > fordStatusExpiry {
    221: 			v.log.Debugf("vehicle status is outdated (age %v > %v), requesting refresh", elapsed, fordStatusExpiry)
    222  

    evcc • vehicle/hyundai.go:
    70  		vehicle = vehicles[0]
    71: 		log.Debugf("found vehicle: %v", cc.VIN)
    72  	} else {

    evcc • vehicle/id.go:
    68  		if err == nil {
    69: 			log.Debugf("found vehicle: %v", cc.VIN)
    70  		}

    evcc • vehicle/kia.go:
    70  		vehicle = vehicles[0]
    71: 		log.Debugf("found vehicle: %v", cc.VIN)
    72  	} else {

    evcc • vehicle/nissan.go:
    61  		if err == nil {
    62: 			log.Debugf("found vehicle: %v", api.VIN)
    63  		}

    evcc • vehicle/renault.go:
    148  		if err == nil {
    149: 			log.Debugf("found vehicle: %v", v.vin)
    150  		}

    evcc • vehicle/seat.go:
    68  		if err == nil {
    69: 			log.Debugf("found vehicle: %v", cc.VIN)
    70  		}

    evcc • vehicle/skoda.go:
    68  		if err == nil {
    69: 			log.Debugf("found vehicle: %v", cc.VIN)
    70  		}

    evcc • vehicle/volvo.go:
    126  		if err == nil {
    127: 			log.Debugf("found vehicle: %v", v.vin)
    128  		}

    evcc • vehicle/vw.go:
    68  		if err == nil {
    69: 			log.Debugf("found vehicle: %v", cc.VIN)
    70  		}

    evcc • vehicle/bluelink/api.go:
    171  			if err := v.DoJSON(req, &resp); err == nil && resp.RetCode != resOK {
    172: 				v.log.Errorf("unexpected response: %s", resp.RetCode)
    173  			}

    evcc • vehicle/bluelink/stamps.go:
    47  	if err != nil {
    48: 		log.Errorln(err)
    49  		return

    evcc • vehicle/porsche/identity.go:
    237  		if foundVehicle.VIN != "" {
    238: 			v.log.Debugf("found vehicle: %v", foundVehicle.VIN)
    239  

Sämtliche `v.log.Debugf("found vehicle` können aus meiner Sicht entfallen. Der Rest sind Informationen, die beim Debuggen helfen und damit auch in einen einfachen Logger geschrieben werden könnten. Hier würde ein einfachs Loginterface also völlig ausreichen.

andig, 2021-12-15:
Depends on #1493

andig, 2021-12-15:
> Sämtliche v.log.Debugf("found vehicle können aus meiner Sicht entfallen. Der Rest sind Informationen, die beim Debuggen helfen und damit auch in einen einfachen Logger geschrieben werden könnten. Hier würde ein einfachs Loginterface also völlig ausreichen.

Bzw. müssten wir für die Fälle des asynchronen loggens von Fehlern die entsprechenden Informationen per channel "raus reichen" und im aufrufenden Code loggen.

closed 2022-01-19 by github-actions[bot]