evcc-io/evcc #1493: replacing the logger

frzifus, 2021-09-06:
**Is your feature request related to a problem? Please describe.**
The currently modified logger provides e.g. the debug logging via a struct field. This causes that the logger can not be abstracted away by an interface.

**Describe the solution you'd like**
Replacing the current logger with e.g.: https://github.com/sirupsen/logrus

**Describe alternatives you've considered**
Adding a log wrapper that provides the methods for logging with different log levels.

andig, 2021-09-06:
Soweit ich mich erinnere, biete Logrus auch kein Interface -> keine Lösung. Ist das nicht per se auch doppelt zu https://github.com/evcc-io/evcc/issues/1492 wo es ohnehin Bestandteil sein muss?

frzifus, 2021-09-06:
Ich würde sagen es ist nur ein teil von https://github.com/evcc-io/evcc/issues/1492.

Der Logger selbst bietet verschiedene Methoden wie `Infof` oder `Debugf`an. Somit könnte jedes Package selbst definieren was es benötigt.

e.g.:
pkg1:
```golang
type myLogger interface {
	Debugf(format string, args ...interface{})
}
```

pkg2:
```golang
type myLogger interface {
	Infof(format string, args ...interface{})
}
```

Das selbe ließe sich durch einen [Notepad](https://github.com/spf13/jwalterweatherman/blob/ce7498392da6d96a17a449ce0e96172af864d822/notepad.go#L42) Wrapper erreichen. 

```golang
type Wrapper struct {
	intern *jww.Notepad
}

func (w *intern) Debug(args ...interface{}) {
	intern.DEBUG.Println(args...)
}
```

andig, 2021-09-07:
Das ist klar. Den leveled Logger durch ein Interface zu ergänzen wäre sofort möglich. Tatsächlich funktioniert das ja genau so wenn log.TRACE z.B. an die Modbus lib übergeben wird. Das zusätzliche Interface würde nur die Levels nochmal zusätzlich rausführen. 

Für den evcc core sehe ich darin aber keinen großen Mehrwert. Für die Module bleibt die Frage, welches Logging eigentlich notwendig ist. Solange Fehler immer zurück gegeben werden bleibt größtenteils nur die Verwendung von TRACE- uns dafür genügt ein einfacher Logger.

frzifus, 2021-09-12:
> Für die Module bleibt die Frage, welches Logging eigentlich notwendig ist. Solange Fehler immer zurück gegeben werden bleibt größtenteils nur die Verwendung von TRACE- uns dafür genügt ein einfacher Logger.

Mit Modulen meinst du das Package `vehicle`? Dort können wir unabhänig entscheiden wie das Logger interface aussehen soll. Wenn ich das aber richtig gesehen habe, sind >90% der `vehicle` Pakete http basiert. Da kommt man ggf. auch gut ohne direktes logging aus.

Hierzu ein beispiel:
https://github.com/frzifus/goblue/blob/master/example/query_status/main.go#L42

andig, 2021-09-13:
> > Für die Module bleibt die Frage, welches Logging eigentlich notwendig ist. Solange Fehler immer zurück gegeben werden bleibt größtenteils nur die Verwendung von TRACE- uns dafür genügt ein einfacher Logger.
> 
> Mit Modulen meinst du das Package `vehicle`? 

Ich meine alle implementierten Geräte aus den Packages `vehicle`,`charger` und `meter`. Sorry- bei OpenWB heissen die Module, hier nicht im Go-Sinne gemeint.

> Dort können wir unabhängig entscheiden wie das Logger interface aussehen soll. 

Mir ist wichtig, *überall* (in den genannten Packges) die gleiche Struktur durchzuziehen weil das den Code deutlich wartbarer macht.

> Wenn ich das aber richtig gesehen habe, sind >90% der `vehicle` Pakete http basiert. Da kommt man ggf. auch gut ohne direktes logging aus.
> 
> Hierzu ein beispiel: https://github.com/frzifus/goblue/blob/master/example/query_status/main.go#L42

Schöne Idee! Aber siehe oben: ich möchte einen Ansatz der überall funktioniert. Wenn `vehicle` einen Transport braucht soll es sich selbst drum kümmern. Im Sinne von evcc ist das opaque. Einen anderen Transport drunter schieben zu können mag hier funktionieren, bei anderen Geräten aber nicht. Von daher sollten wir das über den Logger lösen, nicht den Logger punktuell obsolet machen.

andig, 2021-12-08:
@frzifus ich hab einen neuen Ansatz gefunden, siehe https://github.com/evcc-io/evcc/pull/2002. Idee ist graduelle Migration und soviel Schritte wie nötig. Bisher sieht das ganz gut aus, wird aber noch dauern bis es fertig ist...

andig, 2021-12-08:
> Wenn vehicle einen Transport braucht soll es sich selbst drum kümmern. Im Sinne von evcc ist das opaque. Einen anderen Transport drunter schieben zu können mag hier funktionieren, bei anderen [...]

Die Idee finde ich auch nicht schlecht, wäre aber ein anderes Thema, nämlich custom HTTP client.

frzifus, 2021-12-08:
Schaue am Wochenende rein :)

closed 2022-01-19 by github-actions[bot]