evcc-io/evcc #1497: ABL emH1...invalid status?

hopfi2k, 2021-09-07:
**Software**:
Latest EVCC (compiled from Github)

**Hardware**:
ABL emH1
PW21 Serial RS-485 <-> TCP Server via LAN

**Config** (relevant part):
```
chargers:
- name: wallbox
  type: abl
  uri: 192.168.1.143:5555
  id: 1

```

**Error-Message**:
```
./evcc charger --log trace
[main  ] INFO 2021/09/07 16:47:51 evcc 0.62 (e0f68de)
[main  ] INFO 2021/09/07 16:47:51 using config file /home/pi/evcc.yaml
[abl   ] TRACE 2021/09/07 16:47:51 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030204F402\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030204F402\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030204F402\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">01030204F402\r\n"
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  invalid status: 70
Enabled:        true
Diagnostic dump:
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">010304014141561F\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2021/09/07 16:47:52 modbus: recv ">010304014141561F\r\n"
Firmware: 01 41 41 56
```

Error-Message `invalid status: 70` shows when no car is connected. What is the root cause of this error message? Maybe firmware related?

Running evcc with `./evcc  --log trace` gives (again with no car connected):

```
[modbus] TRACE 2021/09/07 16:57:13 Power: 2280.00W
[site  ] DEBUG 2021/09/07 16:57:13 grid power: 2280W
[site  ] DEBUG 2021/09/07 16:57:13 site power: 2280W
[abl   ] TRACE 2021/09/07 16:57:13 modbus: send ":0103002E0005C9\r\n"
[cache ] DEBUG 2021/09/07 16:57:13 gridPower: 2280
[cache ] DEBUG 2021/09/07 16:57:13 mode: pv
[abl   ] TRACE 2021/09/07 16:57:13 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[abl   ] TRACE 2021/09/07 16:57:13 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2021/09/07 16:57:13 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[lp-1  ] DEBUG 2021/09/07 16:57:13 charge power: 0W
[abl   ] TRACE 2021/09/07 16:57:13 modbus: send ":0103002E0005C9\r\n"
[cache ] DEBUG 2021/09/07 16:57:13 chargePower: 0
[abl   ] TRACE 2021/09/07 16:57:13 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[abl   ] TRACE 2021/09/07 16:57:13 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2021/09/07 16:57:13 modbus: recv ">01030A2EF4106403E803E803E89B\r\n"
[lp-1  ] DEBUG 2021/09/07 16:57:13 charge currents: [0 0 0]A
[abl   ] TRACE 2021/09/07 16:57:13 modbus: send ":010300040001F7\r\n"
[cache ] DEBUG 2021/09/07 16:57:13 chargeCurrents: [0 0 0]
[cache ] DEBUG 2021/09/07 16:57:13 chargeCurrent: 6
[cache ] DEBUG 2021/09/07 16:57:13 chargedEnergy: 0
[cache ] DEBUG 2021/09/07 16:57:13 chargeDuration: 0s
[abl   ] TRACE 2021/09/07 16:57:13 modbus: recv ">01030204F402\r\n"
[abl   ] TRACE 2021/09/07 16:57:13 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2021/09/07 16:57:14 modbus: recv ">01030204F402\r\n"
[lp-1  ] ERROR 2021/09/07 16:57:14 charger: invalid status: 70
[modbus] TRACE 2021/09/07 16:57:18 modbus: closing connection due to idle timeout: 5.000210951s
[modbus] TRACE 2021/09/07 16:57:18 modbus: closing connection due to idle timeout: 5.000882885s
[site  ] DEBUG 2021/09/07 16:57:23 ----
[modbus] TRACE 2021/09/07 16:57:23 modbus: send 00 0c 00 00 00 06 47 03 9c 88 00 32
[modbus] TRACE 2021/09/07 16:57:23 modbus: recv 00 0c 00 00 00 67 47 03 64 05 54 01 c6 01 c5 01 c8 ff fe ff ff ff ff ff ff 09 25 09 26 09 29 ff ff 0c 74 00 00 13 88 ff fe 0c 7d 00 00 00 fe 00 00 03 e8 ff fd 00 b5 08 63 00 00 ff ff 00 00 ff ff 00 00 0c c8 00 00 02 27 01 a6 01 c1 00 00 ff ff 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[modbus] TRACE 2021/09/07 16:57:23 Power: 3188.00W
[site  ] DEBUG 2021/09/07 16:57:23 pv power: 3188W
[cache ] DEBUG 2021/09/07 16:57:23 pvPower: 3188
[modbus] TRACE 2021/09/07 16:57:23 modbus: send 00 07 00 00 00 06 47 03 9c 87 00 69
[modbus] TRACE 2021/09/07 16:57:23 modbus: recv 00 07 00 00 00 d5 47 03 d2 80 00 01 2e 01 87 01 57 ff fe 80 00 5b b7 5b 8d 5b d5 80 00 80 00 80 00 80 00 ff fe 13 86 ff fe 00 e2 00 40 00 52 00 4f 00 01 00 ed 00 46 00 59 00 4f 00 01 ff b9 ff e5 ff dd ff f8 00 01 03 b9 03 96 03 98 03 e2 ff fd ff a4 1a 94 ff e0 d0 ee ff de 7e 9d ff df cd 2e 00 6a 4a 9d 00 29 92 85 00 1f cb 3d 00 25 ea b7 00 00 ff 9f 0e ce ff de ee a1 ff da fa 8f ff de eb 78 00 7a 91 69 00 2f 8c 64 00 2c 8f f6 00 28 2d 4a 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 80 00 00 00 00 00 00 00 00 00
```

From my understanding communication with the ABL seems to work, though...

andig, 2021-09-07:
Mach den ersten Test bitte mit aktuellem Master nochmal. Evtl ist da die Umrechnung des Statuswertes falsch oder ich schaue ins falsche Byte.

andig, 2021-09-07:
/cc @matty67 ich lese für Status die `0x04` statt `0x2E`- funktioniert das bei Dir?

andig, 2021-09-07:
@hopfi2k soweit ich erkenne sagt Deine Box:

    modbus: recv ">01030204F402\r\n"

`F4` wäre nach meiner Interpretation Status F, also Fehler. Welcher kann ich Dir leider nicht sagen...