evcc-io/evcc #1516: v0.63 not accepting power type default

mat9922, 2021-09-09:
Following configuration is not working in v0.63 - in v0.62 it works.

- name: pv-amount
  type: default
  power:
    type: calc
    add:
    - type: modbus
      value: Power
      model: sunspec
      uri: 192.168.178.38:502 # IP address, configured port (default is 502)
      id: 126 # Configured Modbus Device ID ("Ger  t")
    - source: http
      uri:  http://192.168.178.48:8087/getPlainValue/lg-ess-home.0.user.essinfo.home.statistics.pcs_pv_total_power
      scale: 1000

Error message when starting is: "wrong type default"

Did I miss something in my configuration or is it an issue?

premultiply, 2021-09-09:
default -> custom

closed 2021-09-09 by premultiply

mat9922, 2021-09-09:
thx for quick reply