evcc-io/evcc #1517: Invalid Meter in der 0.63

JBSullivan, 2021-09-09:
Mit jeder neuen Version von evcc (Win Version), tausche ich üblicher Weise immer nur die exe Datei aus und teste dann die üblichen Verdächtigen durch (meter, vehicle, usw.)

Mit der 0.63 habe ich die folgende Meldung bekommen.

C:\iobroker\evcc>evcc meter
[main  ] FATAL 2021/09/09 14:06:40 cannot create meter 'pv': invalid meter type: default

Ich habe dann wieder die 0.62 "scharf " geschaltet und die meter Abfrage leiferte sofort wieder die gwünschten daten.

C:\iobroker\evcc>evcc meter
pv
--
Power: 4635W

battery
-------
Power: -16W
SoC:   100%

grid
----
Power: -3436W

Hier die Daten aus der Config:

meters:
- name: pv
  type: default
  power:
    type: calc
    add:
    - type: modbus
      model: sunspec
      uri: 10.122.60.57:502
      id: 1
      value: 160:1:DCW # mpp 1 pv
    - type: modbus
      model: sunspec
      uri: 10.122.60.57:502
      id: 1
      value: 160:2:DCW # mpp 2 pv

premultiply, 2021-09-09:
Duplicate https://github.com/evcc-io/evcc/issues/1516

default -> custom

closed 2021-09-09 by premultiply

premultiply, 2021-09-09:
Ja, überall.

Wirf levels raus. Das ist in der GUI über den Slider gelöst ... aber eigentlich schon länger.

andig, 2021-09-09:
Habs in der Releasedoku ergänzt. Tatsächlich gibts dazu seit mehreren Versionen Warnungen...

JBSullivan, 2021-09-09:
Offensichtlich habe ich einige Versionen "verpennt", denn da scheint sich noch mehr verändert zu haben. So werden bei den Ladepunkten bei mir nur noch "Ladepunkt" in GUI angezeigt, obwohl in der Config der Name definiert ist. Naja, muss ich mir mal in Ruhe ansehen ........

andig, 2021-09-09:
Name als "title" definiert? Der sollte definitiv auftauchen. Weitere Diskussion zum Release gerne in Discussions!