evcc-io/evcc #1520: Support Keba without internal metering (DE Edition 440)

premultiply, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
The cheap Keba DE Edition 440 is lacking the internal metering for power, energy and currents as integrated in all(?) other Keba series but it looks like the current implementation of the Keba control interface assumes that the internal metering does always exist.

**Describe the solution you'd like**
Make the metering configuration optional for Keba devices so that the values can be estimated from control values if the metering is missing.

**Describe alternatives you've considered**
Accept that this device always returns all metering values as zero or overwrite it by an external meter.

**Additional context**
https://github.com/evcc-io/evcc/discussions/1519

andig, 2021-09-09:
Kann man die Situation erkennen? Falls nein müssen wir das wie immer machen- dann wäre das für die bestehenden Kebas extra zu enablen.

premultiply, 2021-09-09:
Speziell dazu gefunden habe ich bislang nichts. Vielleicht ist dort im "report 3"-Datensatz die Spannung auf L1 ja immer 0 und bei den anderen nicht - oder so... wer weiß.
Vielleicht kann @Pjg-Git ja in Kürze neue Erkenntnisse zu dieser Keba-Variante beitragen oder ggf. einen direkten Testzugang bereitstellen.

premultiply, 2021-09-10:
Feedback der DE-Box beim Laden siehe hier https://github.com/evcc-io/evcc/discussions/1519#discussioncomment-1305701
Leider wird die Spannung auf L1 fest mit 230 zurückgegeben.
Eine automatische Erkennung könnte man dann vielleicht auf `E total` > 0 stützen

Pjg-Git, 2021-09-10:
Hab nochmal bei einem anderen Versuch von mir mit einem Cupra Formentor Hybrid nachgesehen. 
Da war das wohl identisch. Also unabhängig der verwendeten Phasen.

Pjg-Git, 2021-09-11:
Ich habe den Ladepunkt bisher nur rudimentär konfiguriert und nun gesehen dass man dort ja auch die Anzahl der Phasen angeben kann. 
Dann könnte man das ja auch verwenden um die Leistung zu errechnen wenn kein Zähler (intern oder extern) vorhanden ist. 
Der aktuelle Strom wird ja von der Box gemeldet. 
Man weiß ja idR welches Fahrzeug man an seiner Box anschließt und mit wievielen Phasen dieses läd.

premultiply, 2021-09-11:
> Der aktuelle Strom wird ja von der Box gemeldet.

Huch?

Kannst du bitte mal die Ausgabe (als Text) von `evcc charger --log trace` während des Ladens hier einfügen?

Pjg-Git, 2021-09-11:
Sie meldet unter Maxcurr die jeweilige Amperezahl auf die sie eingestellt ist. Wenn ich das ändere (z.B. über OpenWB) ändert sich auch die Zahl in dem log. 
kann ich dann noch mal reinhängen z.B. mit 9A. Kann aber morgen werden.

premultiply, 2021-09-11:
Darum geht es ja nicht. Den Wert kennt ja die Steuerung selber und simuliert dann ohne Zähler die Werte.
Es ging um die Messwerte und die sind dann wohl weiterhin null.

Pjg-Git, 2021-09-11:
Klar. Ohne Zähler keine Messung. 
Deshalb soll er ja kalkulieren. 
eingestellte Ampere*Anzahl Phasen lt config*230 ergibt ja ungefähr die aktuelle Leistung. Mehr geht halt ohne Zähler nicht.

Pjg-Git, 2021-09-11:
Im Moment kalkuliert er da nichts.

premultiply, 2021-09-11:
Ja, weil die Box keinen Zähler hat aber die Werte trotzdem mit Wert 0 meldet.
Dazu braucht es in EVCC eben eine Sonderbehandlung da die anderen Keba-Boxen die korrekten Messwerte liefern.
Darum geht es in diesem Issue.
EVCC muss ja irgendwie erkennen können ob die Werte stimmen oder nicht. Notfalls muss man es manuell konfigurieren. So funktioniert das bei anderen Boxen auch.

Kannst daher bitte `evcc charger --log trace` ausführen und die Textausgabe hier einfügen damit wir in der Sache weiterkommen?

Pjg-Git, 2021-09-11:
Hier einmal auf 9A gestellt kein Fahrzeug gesteckt:

[keba  ] TRACE 2021/09/11 13:11:09 send to 192.168.178.106:7090 report 3
[keba  ] TRACE 2021/09/11 13:11:09 recv from 192.168.178.106:7090 {
"ID": "3",
"U1": 0,
"U2": 0,
"U3": 0,
"I1": 0,
"I2": 0,
"I3": 0,
"P": 0,
"PF": 0,
"E pres": 0,
"E total": 0,
"Serial": "21837467",
"Sec": 155733
}
[keba  ] TRACE 2021/09/11 13:11:09 send to 192.168.178.106:7090 report 2
[keba  ] TRACE 2021/09/11 13:11:09 recv from 192.168.178.106:7090 {
"ID": "2",
"State": 1,
"Error1": 0,
"Error2": 0,
"Plug": 3,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 0,
"Max curr %": 1000,
"Curr HW": 16000,
"Curr user": 10000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"Serial": "21837467",
"Sec": 155733
}
[keba  ] TRACE 2021/09/11 13:11:09 send to 192.168.178.106:7090 report 2
[keba  ] TRACE 2021/09/11 13:11:09 recv from 192.168.178.106:7090 {
"ID": "2",
"State": 1,
"Error1": 0,
"Error2": 0,
"Plug": 3,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 0,
"Max curr %": 1000,
"Curr HW": 16000,
"Curr user": 10000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"Serial": "21837467",
"Sec": 155733
}
[keba  ] TRACE 2021/09/11 13:11:09 send to 192.168.178.106:7090 report 3
[keba  ] TRACE 2021/09/11 13:11:09 recv from 192.168.178.106:7090 {
"ID": "3",
"U1": 0,
"U2": 0,
"U3": 0,
"I1": 0,
"I2": 0,
"I3": 0,
"P": 0,
"PF": 0,
"E pres": 0,
"E total": 0,
"Serial": "21837467",
"Sec": 155733
}
[keba  ] TRACE 2021/09/11 13:11:09 send to 192.168.178.106:7090 report 100
[keba  ] TRACE 2021/09/11 13:11:09 recv from 192.168.178.106:7090 {
"ID": "100",
"Session ID": 8,
"Curr HW": 16000,
"E start": 0,
"E pres": 0,
"started[s]": 1631268947,
"ended[s]": 1631269256,
"started": "2021-09-10 10:15:47.000",
"ended": "2021-09-10 10:20:56.000",
"reason": 1,
"timeQ": 3,
"RFID tag": "0000000000000000",
"RFID class": "00000000000000000000",
"Serial": "21837467",
"Sec": 155733
}
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        true
Charged:        0.0kWh
Identifier:     0000000000000000
Diagnostic dump:
[keba  ] TRACE 2021/09/11 13:11:09 send to 192.168.178.106:7090 report 100
[keba  ] TRACE 2021/09/11 13:11:09 recv from 192.168.178.106:7090 {
"ID": "100",
"Session ID": 8,
"Curr HW": 16000,
"E start": 0,
"E pres": 0,
"started[s]": 1631268947,
"ended[s]": 1631269256,
"started": "2021-09-10 10:15:47.000",
"ended": "2021-09-10 10:20:56.000",
"reason": 1,
"timeQ": 3,
"RFID tag": "0000000000000000",
"RFID class": "00000000000000000000",
"Serial": "21837467",
"Sec": 155733
}
{ID:100 Serial:21837467 SessionID:0 CurrHW:16000 EStart:0 EPres:0 Started:2021-09-10 10:15:47.000 Ended:2021-09-10 10:20:56.000 StartedS:1631268947 EndedS:1631269256 Reason:1 TimeQ:3 RFIDTag:0000000000000000 RFIDClass:00000000000000000000 Sec:155733}

Pjg-Git, 2021-09-11:
und jetzt 9A, Fahrzeug gesteckt und läd:
[keba  ] TRACE 2021/09/11 13:15:38 send to 192.168.178.106:7090 report 2
[keba  ] TRACE 2021/09/11 13:15:38 recv from 192.168.178.106:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 9000,
"Max curr %": 150,
"Curr HW": 16000,
"Curr user": 9000,
"Curr FS": 0,
"Tmo FS": 0,
"Curr timer": 0,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"Serial": "21837467",
"Sec": 156002
}
[keba  ] TRACE 2021/09/11 13:15:38 send to 192.168.178.106:7090 report 3
[keba  ] TRACE 2021/09/11 13:15:38 recv from 192.168.178.106:7090 {
"ID": "3",
"U1": 230,
"U2": 0,
"U3": 0,
"I1": 0,
"I2": 0,
"I3": 0,
"P": 0,
"PF": 0,
"E pres": 0,
"E total": 0,
"Serial": "21837467",
"Sec": 156002
}
[keba  ] TRACE 2021/09/11 13:15:38 send to 192.168.178.106:7090 report 100
[keba  ] TRACE 2021/09/11 13:15:38 recv from 192.168.178.106:7090 {
"ID": "100",
"Session ID": 9,
"Curr HW": 16000,
"E start": 0,
"E pres": 0,
"started[s]": 1631358880,
"ended[s]": 0,
"started": "2021-09-11 11:14:40.000",
"ended": "0",
"reason": 5,
"timeQ": 0,
"RFID tag": "0000000000000000",
"RFID class": "00000000000000000000",
"Serial": "21837467",
"Sec": 156002
}
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Charge status:  C
Enabled:        true
Charged:        0.0kWh
Identifier:     0000000000000000
Diagnostic dump:
[keba  ] TRACE 2021/09/11 13:15:38 send to 192.168.178.106:7090 report 100
[keba  ] TRACE 2021/09/11 13:15:38 recv from 192.168.178.106:7090 {
"ID": "100",
"Session ID": 9,
"Curr HW": 16000,
"E start": 0,
"E pres": 0,
"started[s]": 1631358880,
"ended[s]": 0,
"started": "2021-09-11 11:14:40.000",
"ended": "0",
"reason": 5,
"timeQ": 0,
"RFID tag": "0000000000000000",
"RFID class": "00000000000000000000",
"Serial": "21837467",
"Sec": 156002
}
{ID:100 Serial:21837467 SessionID:0 CurrHW:16000 EStart:0 EPres:0 Started:2021-09-11 11:14:40.000 Ended:0 StartedS:1631358880 EndedS:0 Reason:5 TimeQ:0 RFIDTag:0000000000000000 RFIDClass:00000000000000000000 Sec:156002}

premultiply, 2021-09-11:
Danke, das hilft weiter!

andig, 2021-09-11:
Meldet der U2=0 auch wenn 3p geladen wird?

premultiply, 2021-09-11:
Falls autodetect gewünscht und sinnvoll würde ich dazu dann tatsächlich `E total = 0` als Erkennung für eine fehlende Zählerfunktion heranziehen.

Pjg-Git, 2021-09-12:
> Meldet der U2=0 auch wenn 3p geladen wird?

Ja. Es hing ein Tesla dran. Der läd mir 3P

andig, 2021-09-12:
Gerne mal probieren. Ohne Messung der KEBA sollten jetzt die Vorgabewerte angesetzt werden.

Pjg-Git, 2021-09-12:
Sorry. Blick hier nicht durch. Wo bekomme ich denn jetzt ne neue evcc Datei mit der Änderung drin für Windows her?

andig, 2021-09-12:
Selbst compilieren wie in der README beschrieben

Pjg-Git, 2021-09-12:
Welcher Readme? Link?
In der readme.md steht nichts von komplilieren. 
Da steht übrigens auch unter Installation nur Download, wieder ohne Link wo man das File herholen soll. 
Ich Google da immer damit ich überhaupt ne Downloaddatei finde!
Also warte ich halt auf die nächste Version.

andig, 2021-09-12:
> Welcher Readme? Link?
> In der readme.md steht nichts von komplilieren.

Doch, tut es: https://github.com/evcc-io/evcc#installation

> Ich Google da immer damit ich überhaupt ne Downloaddatei finde!

Wie bei jedem anderen Projekt auf Github sonst auch: https://github.com/evcc-io/evcc/releases

> Also warte ich halt auf die nächste Version.

Das ist immer eine gute Option wenn man nicht selbst compilieren möchte.

andig, 2021-09-14:
Findet sich jemand der das testen kann?

Pjg-Git, 2021-09-15:
Mache ich gerne. 
Muss aber wie gesagt auf eine kompilierte Version warten da ich nicht weiß wie ich das selbst anstellen soll.

PR #1541 (2021-09-17): Keba: auto-detect optional meter
Fix #1520
closed 2021-09-17 by andig