evcc-io/evcc #1524: ABL invalid status: Outlet disabled

premultiply, 2021-09-10:
### Discussed in https://github.com/evcc-io/evcc/discussions/1523

<div type='discussions-op-text'>

<sup>Originally posted by **pmt272** September 10, 2021</sup>
Die ABL Kommunikation funktioniert nun super. Nach dem anschalten der Wallbox:

```
evcc charger -l trace

[abl   ] TRACE 2021/09/10 09:00:53 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 41 32 45 41 31 31 31 30 42 30 33 45 38 30 33 45 38 30 33 45 38 34 36 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 41 32 45 41 31 31 31 30 42 30 33 45 38 30 33 45 38 30 33 45 38 34 36 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 41 32 45 41 31 31 31 30 42 30 33 45 38 30 33 45 38 30 33 45 38 34 36 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 32 30 34 41 31 35 35 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 32 30 34 41 31 35 35 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 32 30 34 41 31 35 35 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 32 30 34 41 31 35 35 0d 0a
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        true
Diagnostic dump:
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 30 31 30 30 30 32 46 39 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 34 30 31 34 31 34 31 42 45 42 37 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: send 3a 30 31 30 33 30 30 30 31 30 30 30 32 46 39 0d 0a
[abl   ] TRACE 2021/09/10 09:00:54 modbus: recv 3e 30 31 30 33 30 34 30 31 34 31 34 31 42 45 42 37 0d 0a

```

Wenn ich dann evcc starte und 5 Minuten kein Auto verbinde, oder den Ladevorgang zum Beispiel über die GUI auf Stop setze, bekomme ich aber als Status einen "invalid status: Outlet disabled": 

```
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 41 32 45 45 30 31 30 36 34 30 33 45 38 30 33 45 38 30 33 45 38 41 46 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 41 32 45 45 30 31 30 36 34 30 33 45 38 30 33 45 38 30 33 45 38 41 46 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 41 32 45 45 30 31 30 36 34 30 33 45 38 30 33 45 38 30 33 45 38 41 46 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 32 45 30 30 30 35 43 39 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 41 32 45 45 30 31 30 36 34 30 33 45 38 30 33 45 38 30 33 45 38 41 46 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 32 30 34 45 30 31 36 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 32 30 34 45 30 31 36 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 32 30 34 45 30 31 36 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 30 34 30 30 30 31 46 37 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 32 30 34 45 30 31 36 0d 0a
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  invalid status: Outlet disabled
Enabled:        false
Diagnostic dump:
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 30 31 30 30 30 32 46 39 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 34 30 31 34 31 34 31 42 45 42 37 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: send 3a 30 31 30 33 30 30 30 31 30 30 30 32 46 39 0d 0a
[abl   ] TRACE 2021/09/10 09:04:44 modbus: recv 3e 30 31 30 33 30 34 30 31 34 31 34 31 42 45 42 37 0d 0a
Firmware: 01 41 41 be
```

Outlet disabled ist ja richtig - es ist dann aber doch kein invalid status.
Wenn ich evcc mit -l debug starte, wird für den Status auch immer ein Error ausgegeben:

`[lp-1  ] ERROR 2021/09/10 09:21:37 charger: invalid status: Outlet disabled`</div>

andig, 2021-09-10:
Closed in https://github.com/evcc-io/evcc/commit/d471eec155200a2c2901b5cf18a9da254563912c

closed 2021-09-10 by andig

premultiply, 2021-09-10:
Bist du sicher dass das reicht?
Was passiert wenn der Ladevorgang seitens EVCC pausiert wird?
Wird dann nicht das Fahrzeug quasi disconnected?

andig, 2021-09-10:
Lets find out…

premultiply, 2021-09-13:
Ist so überhaupt irgendeine Statuserkennung möglich wenn der Charger im disabled-Zustand keinen (anderen) Status zurück gibt?

andig, 2021-09-13:
Keine Ahnung. Aber bisher scheint das Problem ja nicht wieder aufgetaucht zu sein.

premultiply, 2021-09-13:
Ich glaube eher bislang hat das auch niemand getestet. Der typische ABL-Besitzer ist wohl eher ein Release-Downloader.

pmt272, 2021-09-13:
Ich bin wohl nicht ganz typisch und habe es auch ohne neuen Release getestet. 
Mit den Änderungen kommt der Fehler nicht mehr, aber die Box ist halt disabled. Da ich noch auf ein E-Auto warte, kann ich leider grade nicht testen, wie es sich mit angeschlossenem Auto verhält. Ich warte hier noch auf einen Freund mit seinem Auto zum testen und gebe dann Feedback.