evcc-io/evcc #1558: Improve SoC-Estimator resilience

premultiply, 2021-09-14:
Before adapting the rate the soc estimator needs to check if charging state of vehicle matches the charging state of the loadpoint.
If they do NOT match the rate should NOT be touched because the vehicle API state is providing outdated data.
This is espacially the case on many cars if the charging starts long time after the vehicle was connected after a journey (e.g. connected at the evening but charging starts the next morning for the first time). In this time the cars soc may have been changed internally (changed battery temperature) but this is not updated to the API backend until the charging starts as the car is sleeping. After wake up for charge there is additional delay caused by the async nature of manufactures backend infrastructure and EVCCs caching. So the old state from the time the car has been parked will be presented even seconds to minutes after the charging has started.

In current implementation the soc estimator will adapt his rate on the second data update it receives from the car but this data is just the state after wakeup for charging. This can be avoided by a state compare between vehicle and loadpoint. Only if they match (both indicate charging or both indicate not charging) the vehicle data sample should be handeled as up-to-date otherwise the former rate should be keept.

@andig The soc estimator code needs access to the vehicles and the loadpoints state. Can you help me with this pls?

### Discussed in https://github.com/evcc-io/evcc/discussions/1538

<div type='discussions-op-text'>

<sup>Originally posted by **blenni** September 11, 2021</sup>
Hi,

erstmal nur als Frage und noch nicht als Issue, da ich (vermutlich) noch nicht die richtigen Logdateien für das Problem habe. Evtl. liegts auch an BMW und man kann gar nichts dagegen machen...

Mir ist  heute aufgefallen, dass die Ladezeitprognose falsch ist:
[image]

Dem Ladepunkt ist mein BMW i3 zugeordnet, in der Config ist der i3 wie folgt definiert:
```vehicles:
- name: i3
  type: bmw
  title: BMW i3s
  capacity: 39 # kWh
  user: xxx
  password: xxx
  vin: xxx
  #cache: 5m
```
Bei einem Ladezustand von ca. 50% und 3,6 kW Ladeleistung (einphasig) sollte die Restzeit bei etwas über 5 h liegen.

Die Log-Settings habe ich wie folgt definiert:
```log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug
```

Zwei Auffälligkeiten sehe ich im Logfile: Bei Ladestart ist der SOC vom Fahrzeug noch nicht aktuell, beim Anstecken wird scheinbar der Wert vom Server übermittelt, der aber vom Fahrzeug noch nicht aktualisiert wurde (der Wert entspricht dem Wert vor der Fahrt):

```Sep 11 13:51:28 raspberrypi evcc[16540]: [lp-2  ] DEBUG 2021/09/11 13:51:28 soc estimated: 57.00% (vehicle: 57.00%)
Sep 11 13:51:48 raspberrypi evcc[16540]: [lp-2  ] DEBUG 2021/09/11 13:51:48 soc estimated: 57.02% (vehicle: 57.00%)
Sep 11 13:52:31 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 13:52:31 soc estimated: 28.01% (vehicle: 28.00%)
Sep 11 13:52:40 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 13:52:40 soc estimated: 28.02% (vehicle: 28.00%)
Sep 11 13:53:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 13:53:00 soc estimated: 28.07% (vehicle: 28.00%)
Sep 11 13:53:20 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 13:53:20 soc estimated: 28.11% (vehicle: 28.00%)
```

Behoben habe ich das durch den Neustart von EVCC - dann wird der SOC nochmal neu abgefragt. Ich habe mir schon angewöhnt, das Fahrzeug sofort nach dem Anstecken abzuschließen, dann wird scheinbar vom Fahrzeug der Wert an den Server gesendet und aktualisiert - meistens funktioniert das und der korrekte SOC wird in EVCC angezeigt, aber leider nicht immer. Hier wäre es ggf. noch schön, wenn der Wert bei Ladestart nochmal neu geholt wird (ist aber jetzt nicht das ursprüngliche Problem, das ich hier zeigen wollte).

Die zweite Auffälligkeit ist folgende:
Anfangs läuft der Schätzer recht gut, vielleicht ein bisschen schneller als der tatsächliche Ladevorgang bzw. was vom Server gemeldet wird.

Dann passiert folgendes:
```Sep 11 15:08:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:08:00 soc estimated: 33.99% (vehicle: 32.00%)
Sep 11 15:08:20 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:08:20 soc estimated: 34.04% (vehicle: 32.00%)
Sep 11 15:08:41 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:08:41 soc estimated: 34.09% (vehicle: 32.00%)
Sep 11 15:09:20 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:09:20 soc estimated: 41.20% (vehicle: 41.00%)
Sep 11 15:09:40 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:09:40 soc estimated: 41.39% (vehicle: 41.00%)
Sep 11 15:10:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:10:00 soc estimated: 41.59% (vehicle: 41.00%)
Sep 11 15:10:21 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:10:21 soc estimated: 41.59% (vehicle: 41.00%)
Sep 11 15:10:40 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:10:40 soc estimated: 41.98% (vehicle: 41.00%)
Sep 11 15:11:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:11:00 soc estimated: 42.17% (vehicle: 41.00%)
...
Sep 11 15:22:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:22:00 soc estimated: 48.59% (vehicle: 41.00%)
Sep 11 15:22:20 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:22:20 soc estimated: 48.79% (vehicle: 41.00%)
Sep 11 15:22:40 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:22:40 soc estimated: 48.99% (vehicle: 41.00%)
Sep 11 15:23:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:23:00 soc estimated: 49.18% (vehicle: 41.00%)
Sep 11 15:23:21 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:23:21 soc estimated: 49.38% (vehicle: 41.00%)
Sep 11 15:23:40 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:23:40 soc estimated: 49.57% (vehicle: 41.00%)
Sep 11 15:24:01 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:24:01 soc estimated: 49.76% (vehicle: 41.00%)
Sep 11 15:24:40 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:24:40 soc estimated: 43.19% (vehicle: 43.00%)
Sep 11 15:25:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:25:00 soc estimated: 43.39% (vehicle: 43.00%)
Sep 11 15:25:20 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:25:20 soc estimated: 43.59% (vehicle: 43.00%)
Sep 11 15:25:41 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:25:41 soc estimated: 43.78% (vehicle: 43.00%)
Sep 11 15:26:00 raspberrypi evcc[30523]: [lp-2  ] DEBUG 2021/09/11 15:26:00 soc estimated: 43.97% (vehicle: 43.00%)
```

Hier kommt also vom Fahrzeug um 15:09 auf einmal ein Sprung, der dazu führt, dass der EVCC-Schätzer scheinbar viel zu schnell läuft.

Soweit ich weiß, ist es beim i3 so, dass das Fahrzeug nicht laufend die Daten auf den Server übermittelt, sondern der Server selber auch noch den SOC schätzt - kommt dann ein neuer Wert vom Fahrzeug, kann so ein Sprung schon mal vorkommen. Es wundert mich allerdings ein bisschen, da ich sowas eher im PV- als im Sofort-Modus erwarten würde.

Welche logs in welchem Level wären denn hilfreich, um dem Thema weiter auf die Spur zu kommen? Oder sagt ihr, das liegt an BMW und lässt sich aufgrund der Umstände nicht verbessern?

Wenn ich EVCC dann übrigens wieder neu starte, passt die Prognose wieder - mal sehen, ob sie nochmal wegläuft...
[image]
</div>

premultiply, 2021-09-14:
Oh ich seh gerade im Code dass der Charger-Zugriff im Estimator inzwischen nachgerüstet wurde.
@DerAndereAndi Warst du das?

Dann bekomme ich das vielleicht doch nun alleine hin.

andig, 2021-09-15:
> In current implementation the soc estimator will adapt his rate on the second data update it receives from the car but this data is just the state after wakeup for charging. This can be avoided by a state compare between vehicle and loadpoint.

Du sagst also der erste SoC während des Ladevorgangs ist derjenige der als Startpunkt dienen soll und nicht der erste empfangene?

premultiply, 2021-09-15:
Ja, weil er vorher mit einem veralteten Wert gefüttert wurde. Und genau diese veralteten Werte will ich nun rausfiltern indem ich den Charger-Status mit dem Vehicle-Status abgleiche.

Genau genommen geht es um dieses ToDo:
https://github.com/evcc-io/evcc/blob/ccf83bb31027d1dd197c9dbd631764f9a548be03/core/soc/estimator.go#L145-L152