evcc-io/evcc #1567: vehicle not available: login failed: unexpected status: 403

bicaluv, 2021-09-15:
**Describe the bug**
Beim Start von evcc liefert dieses für Skoda aktuell den Fehler "vehicle not available: login failed: unexpected status: 403"

**To Reproduce**
evcc -vehicle
 mit meine yaml (and hat sie)

**Expected behavior**
Kein Fehler und SoC Ermittlung funktionieren.

**EVCC details:**
siehe Trace im Anhang
[v_error.txt.zip](https://github.com/evcc-io/evcc/files/7170794/v_error.txt.zip)

andig, 2021-09-15:
Gerne Zugangsdaten an cpuidle@gmx.de

bicaluv, 2021-09-15:
vehicles:
- name: Citigo
  type: skoda
  title: CITIGOe iV
  capacity: 36 # kWh
  user: ***@***.***
  password: Sp//dy-G0nzales
  vin: TMBZZZAAZMD805684

bicaluv, 2021-09-15:
Blöderweise ist das Fahrzeug gerade unterwegs und nicht evcc verbunden.

andig, 2021-09-15:
Deine Zugangsdaten haben nichts bei Github zu suchen! Ich empfehle das Passwort zu ändern!

Sillium007, 2021-09-16:
Brauchst du dafür noch was?
Kann auch bestätigen dass für den Skoda 403 kommt

andig, 2021-09-16:
Hilfreich wäre zu wissen, obs bei irgendeinem Tool noch funktioniert. Scheint ja als wäre das hier eine echte echte Änderung zu sein statt ein Ausfall.

Sillium007, 2021-09-16:
https://www.skoda-connect.com geht für mich und die MySkoda App ging heute morgen als ich geantwortet habe. Gerade streikt sie. Ich versuche es später nochmal. Vielleicht ist grad wirklich nur was bei denen kaputt

bicaluv, 2021-09-16:
Seit heute morgen geht bei mir auch der Zugang via MySkoda App nicht mehr. Das war gestern schon mal so. Am Abend klappte die App dann mal wieder aber heute wieder nix. Laut Hotline von Skoda ist denen da bekannt

Sillium007, 2021-09-16:
https://xn--allestrungen-9ib.de/stoerung/skoda-connect/
Ja scheint bei denen nicht zu gehen. Also würde ich fast behaupten dass vermutlich evcc nix machen muss.

Sillium007, 2021-09-16:
App geht bei mir gerade wieder aber evcc hat noch 403

bicaluv, 2021-09-16:
Laut Skoda Hotline von gerade eben haben sie seit Montag Serverprobleme und es geht ab und an mal und dann wieder nicht. "Iss so" kam als Antwort :-(

Sillium007, 2021-09-16:
ok.. dann wohl erstmal abwarten

andig, 2021-09-16:
Skoda hat die App-ID geändert. Auch mit der neuen gehts allerdings nicht. Waiting for Skoda...

andig, 2021-09-17:
Leider sowohl mit altem als auch neuem Login. Muss weiter analysiert werden.

andig, 2021-09-17:
Ist behoben

closed 2021-09-17 by andig