evcc-io/evcc #1592: Shelly Plug only usable as charger without password protection

erikgieseler, 2021-09-19:
If you want to use a shelly plug as charger you have to disable the password protection in the shelly gui.
If you dont deactivate it, evcc cant read the current power from the plug and reports in the log "invalid channel, power meter missing". Without the password protection it is working without any problem.

premultiply, 2021-09-19:
Config & Log?

closed 2021-09-19 by premultiply

erikgieseler, 2021-09-19:
`
[site  ] INFO 2021/09/19 14:34:33 site config:

[site  ] INFO 2021/09/19 14:34:33   meters:    grid ✓ pv ✓ battery ✓

[site  ] INFO 2021/09/19 14:34:33     grid:    power ✓ energy ✗ currents ✗

[site  ] INFO 2021/09/19 14:34:33     pv:      power ✓ energy ✗ currents ✗

[cache ] DEBUG 2021/09/19 14:34:33 siteTitle: Gieseler

[cache ] DEBUG 2021/09/19 14:34:33 gridConfigured: true

[cache ] DEBUG 2021/09/19 14:34:33 pvConfigured: true

[site  ] INFO 2021/09/19 14:34:33     battery: power ✓ energy ✗ currents ✗ soc ✓

[cache ] DEBUG 2021/09/19 14:34:33 batteryConfigured: true

[cache ] DEBUG 2021/09/19 14:34:33 prioritySoC: 75

[lp-1  ] INFO 2021/09/19 14:34:33 loadpoint 1:

[lp-1  ] INFO 2021/09/19 14:34:33   mode:      pv

[lp-1  ] INFO 2021/09/19 14:34:33   charger:   power ✓ energy ✓ currents ✗ timer ✗

[lp-1  ] INFO 2021/09/19 14:34:33   meters:    charge ✓

[lp-1  ] INFO 2021/09/19 14:34:33     charge:  power ✓ energy ✓ currents ✗

[lp-1  ] INFO 2021/09/19 14:34:33   vehicles:  ✓

[lp-1  ] INFO 2021/09/19 14:34:33     car 0:   range ✓ finish ✓ status ✓ climate ✗

[lp-2  ] INFO 2021/09/19 14:34:33 loadpoint 2:

[lp-2  ] INFO 2021/09/19 14:34:33   mode:      pv

[lp-2  ] INFO 2021/09/19 14:34:33   charger:   power ✓ energy ✓ currents ✗ timer ✗

[lp-2  ] INFO 2021/09/19 14:34:33   meters:    charge ✓

[lp-2  ] INFO 2021/09/19 14:34:33     charge:  power ✓ energy ✓ currents ✗

[lp-2  ] INFO 2021/09/19 14:34:33   vehicles:  ✓

[lp-2  ] INFO 2021/09/19 14:34:33     car 0:   range ✓ finish ✓ status ✓ climate ✗

[lp-3  ] INFO 2021/09/19 14:34:33 loadpoint 3:

[lp-3  ] INFO 2021/09/19 14:34:33   mode:      pv

[lp-3  ] INFO 2021/09/19 14:34:33   charger:   power ✓ energy ✗ currents ✗ timer ✗

[lp-3  ] INFO 2021/09/19 14:34:33   meters:    charge ✓

[lp-3  ] INFO 2021/09/19 14:34:33     charge:  power ✓ energy ✗ currents ✗

[lp-3  ] INFO 2021/09/19 14:34:33   vehicles:  ✓

[lp-3  ] INFO 2021/09/19 14:34:33     car 0:   range ✓ finish ✓ status ✓ climate ✗

[cache ] DEBUG 2021/09/19 14:34:33 chargeConfigured: true

[cache ] DEBUG 2021/09/19 14:34:33 chargeConfigured: true

[cache ] DEBUG 2021/09/19 14:34:33 chargeConfigured: true

[site  ] DEBUG 2021/09/19 14:34:33 ----
[... 491 lines cut ...]
[cache ] DEBUG 2021/09/19 14:34:43 chargeRemainingDuration: -1ns

[cache ] DEBUG 2021/09/19 14:34:43 range: -1

[site  ] DEBUG 2021/09/19 14:34:44 ----

[site  ] DEBUG 2021/09/19 14:34:44 pv power: 4039W

[site  ] DEBUG 2021/09/19 14:34:44 grid power: -3457W

[cache ] DEBUG 2021/09/19 14:34:44 pvPower: 4039

[cache ] DEBUG 2021/09/19 14:34:44 gridPower: -3457

[site  ] DEBUG 2021/09/19 14:34:44 battery power: -30W

[site  ] DEBUG 2021/09/19 14:34:44 battery soc: 100%

[cache ] DEBUG 2021/09/19 14:34:44 batteryPower: -30

[cache ] DEBUG 2021/09/19 14:34:44 batterySoC: 100

[site  ] DEBUG 2021/09/19 14:34:44 site power: -3487W

[cache ] DEBUG 2021/09/19 14:34:44 mode: pv`

erikgieseler, 2021-09-19:
`uri: 0.0.0.0:7070
interval: 1s

log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug
  lp-3: debug

chargers:

- name: wallbox_links
  type: warp
  broker: 192.168.178.1:1883
  topic: wallbox/links
  useMeter: true
  timeout: 15s

- name: wallbox_rechts
  type: warp
  broker: 192.168.178.1:1883
  topic: wallbox/rechts
  useMeter: true
  timeout: 15s

- name: schukosteckdose
  type: shelly
  uri: http://192.168.178.187
  channel: 0  
  standbypower: 5

meters:

- name: gridmeter
  type: custom
  power: 
    source: mqtt
    topic: powerwall/grid 
    timeout: 15s

- name: pvmeter
  type: custom
  power: 
    source: mqtt
    topic: powerwall/solar 
    timeout: 15s

- name: batterymeter
  type: custom
  power: 
    source: mqtt
    topic: powerwall/battery
    timeout: 15s
  soc:
    source: mqtt
    topic: powerwall/soc
    timeout: 15s 

site:

  title: Gieseler
  meters:
    grid: gridmeter
    pv: pvmeter
    battery: batterymeter
  prioritySoC: 75

vehicles:

- name: tesla
  type: tesla
  title: Model 3
  capacity: 75
  tokens:
    access:
    refresh: 
    vin: 
  cache: 5m
 
loadpoints:

- title: Wallbox links
  mode: pv
  charger: wallbox_links
  vehicles:
  - tesla
  soc:
    poll:
      mode: connected
      interval: 30m
    min: 20
    target: 80 
    estimate: true 
  onDisconnect: 
    mode: pv 
    targetSoC: 80
  onIdentify:
  - tesla:
      mode: pv 
      targetSoC: 80
  phases: 3 
  enable: 
    delay: 1m 
    threshold: -3800
  disable: 
    delay: 5m
    threshold: 800 
  guardduration: 5m 
  mincurrent: 6
  maxcurrent: 16

- title: Wallbox rechts
  mode: pv
  charger: wallbox_rechts
  vehicles:
  - tesla
  soc:
    poll:
      mode: connected
      interval: 30m
    min: 20
    target: 80 
    estimate: true 
  onDisconnect: 
    mode: pv 
    targetSoC: 80 
  onIdentify:
  - tesla:
      mode: pv 
      targetSoC: 80
  phases: 3 
  enable: 
    delay: 1m 
    threshold: -3800
  disable: 
    delay: 5m
    threshold: 800
  guardduration: 5m 
  mincurrent: 6
  maxcurrent: 16

- title: Schukosteckdose
  mode: pv
  charger: schukosteckdose
  vehicles:
  - tesla
  soc:
    poll:
      mode: charging
      interval: 30m
    min: 20
    target: 80 
    estimate: true 
  onDisconnect: 
    mode: pv 
    targetSoC: 80 
  onIdentify:
  - tesla:
      mode: pv 
      targetSoC: 80
  phases: 1 
  enable: 
    delay: 1m 
    threshold: 0
  disable: 
    delay: 5m
    threshold: 200
  guardduration: 1m 
  mincurrent: 6
  maxcurrent: 6

mqtt:
  broker: 192.168.178.1:1883
  clientid: evcc
`

premultiply, 2021-09-19:
Ich sehe da kein Passwort