evcc-io/evcc #1595: Add optional passwort protected data retrieval & control to Shelly implementation

premultiply, 2021-09-19:
### Discussed in https://github.com/evcc-io/evcc/discussions/1593

<div type='discussions-op-text'>

<sup>Originally posted by **erikgieseler** September 19, 2021</sup>
If you want to use a shelly plug as charger you have to disable the password protection in the shelly gui.
If you dont deactivate it, evcc cant read the current power from the plug and reports in the log "invalid channel, power meter missing". Without the password protection it is working without any problem.
</div>

premultiply, 2021-09-19:
/cc @thierolm

premultiply, 2021-09-19:
Already implemented: https://github.com/evcc-io/evcc/discussions/1593#discussioncomment-1353695

andig, 2021-09-20:
Für den Anwender ist das ja nicht ganz offensichtlich bzw. müsste man dafür verstehen, dass das mit Basic Auth passiert und wie diese in der URL anzugeben ist. Alternativ könnten wir auch user/password extra anbieten und die URL dann zusammen bauen?

premultiply, 2021-09-20:
Wäre definitiv benutzerfreundlicher!
Ebenso das führende http://

andig, 2021-09-20:
Also:

```
uri:
user:
password:
```

falls die URI kein http hat wirds vorgehängt

closed 2021-09-20 by andig