evcc-io/evcc #1599: Detect vehicle without api when all other vehicles have api

andig, 2021-09-19:
> Gibts denn ein Log wo es nicht funktioniert? Ich würd lieber das Problem lösen als einen Workaround zu bauen.

```
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 charger status: B
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 vehicle api refresh
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 vehicle status: A (Stefans Tesla)
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] WARN 2021/09/19 16:23:22 target charging: not possible
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 max charge current: 2.75A = 2.01A + 0.738A (-509W @ 3p)
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 !!pvScalePhases available power 509 for target current 0.7 @ 3p/3p
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 !!pvScalePhases timer remaining: 5m50s
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 available power below 3p min threshold of 4140W
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 phase disable timer remaining: 4m9s
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 reset pv enable timer: 1m0s
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 pv enable timer: keep disabled
Sep 19 16:23:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:22 pv max charge current: 0A
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 charge power: 0W
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 charge currents: [0.009 0.008 0.008]A
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 charger status: B
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 vehicle status: A (Stefans Tesla)
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] WARN 2021/09/19 16:23:32 target charging: not possible
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 max charge current: 2.73A = 2.01A + 0.724A (-499W @ 3p)
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 !!pvScalePhases available power 499 for target current 0.7 @ 3p/3p
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 !!pvScalePhases timer remaining: 6m0s
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 available power below 3p min threshold of 4140W
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 phase disable timer remaining: 4m0s
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 reset pv enable timer: 1m0s
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 pv enable timer: keep disabled
Sep 19 16:23:32 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:32 pv max charge current: 0A
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 charge power: 0W
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 charge currents: [0.009 0.008 0.008]A
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 charger status: B
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 vehicle status: A (Stefans Tesla)
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] WARN 2021/09/19 16:23:42 target charging: not possible
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 max charge current: 2.79A = 2.01A + 0.777A (-536W @ 3p)
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 !!pvScalePhases available power 536 for target current 0.8 @ 3p/3p
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 !!pvScalePhases timer remaining: 6m10s
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 available power below 3p min threshold of 4140W
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 phase disable timer remaining: 3m50s
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 reset pv enable timer: 1m0s
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 pv enable timer: keep disabled
Sep 19 16:23:42 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:42 pv max charge current: 0A
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 charge power: 0W
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 charge currents: [0.009 0.008 0.008]A
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 charger status: B
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 vehicle status: A (Stefans Tesla)
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] WARN 2021/09/19 16:23:52 target charging: not possible
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 max charge current: 2.76A = 2.01A + 0.751A (-518W @ 3p)
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 !!pvScalePhases available power 518 for target current 0.8 @ 3p/3p
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 !!pvScalePhases timer remaining: 6m20s
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 available power below 3p min threshold of 4140W
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 phase disable timer remaining: 3m40s
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 reset pv enable timer: 1m0s
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 pv enable timer: keep disabled
Sep 19 16:23:52 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:23:52 pv max charge current: 0A
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 charge power: 0W
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 charge currents: [0.009 0.008 0.008]A
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 charger status: B
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 vehicle status: A (Stefans Tesla)
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] WARN 2021/09/19 16:24:02 target charging: not possible
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 max charge current: 2.75A = 2.01A + 0.744A (-514W @ 3p)
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 !!pvScalePhases available power 514 for target current 0.7 @ 3p/3p
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 !!pvScalePhases timer remaining: 6m30s
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 available power below 3p min threshold of 4140W
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 phase disable timer remaining: 3m30s
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 reset pv enable timer: 1m0s
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 pv enable timer: keep disabled
Sep 19 16:24:02 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:02 pv max charge current: 0A
Sep 19 16:24:12 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:12 charge power: 0W
Sep 19 16:24:12 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:12 charge currents: [0.009 0.008 0.008]A
Sep 19 16:24:12 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:12 charger status: B
[... 10 lines cut ...]
Sep 19 16:24:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:22 charge power: 0W
Sep 19 16:24:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:22 charge currents: [0.009 0.008 0.008]A
Sep 19 16:24:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:22 charger status: B
Sep 19 16:24:22 piEVCC evcc[32230]: [lp-1  ] DEBUG 2021/09/19 16:24:22 vehicle status: A (Stefans Tesla)
Sep 19 16:24:22 piEVCC evcc[32230]: [lp-1  ] WARN 2021/09/19 16:24:22 target charging: not possible
```

Zum Setup:

- evcc 0.64
- easee Home v289 (aktuelle FW)
- Kia e-Niro
- Tesla Model 3

Die Abfrage der Fahrzeug-API über `evcc vehicle` liefert für beide Fahrzeuge korrekte Ergebnisse, wobei der Fahrzeugstatus (`B`, `C`) im Ergebnis des angeschlossenen Kia nicht enthalten ist. Der Tesla steht entspannt daneben und hat nichts zu tun. Liefert daher korrekt den Status `A`.

Wie dem Logfile zu entnehmen, kommt evcc wohl nicht auf die Idee, dass bei Tesla im Status `A` wohl der Kia angeschlossen sein muss, wenn der Charger korrekt `B` liefert.

Im Frontend wird ein grüner Ladebalken angezeigt, ein Limit kann nicht eingestellt werden, eine Fahrzeugbezeichnung wird nicht angezeigt. Als Warnung wird mit gelben Dreieck `target charging: not possible` ausgegeben.

Ich sehe hier zudem die Herausforderung, dass wenn der Tesla von mir am Arbeitsplatz und der Kia Zuhause geladen wird, die Heuristik dann in evcc den Tesla ermittelt und nicht den Kia.

Daher die Idee nach Auswahl für [Automatich | Fahrzeug1 | Fahrzeug2 | ..]

_Originally posted by @stfn7 in https://github.com/evcc-io/evcc/discussions/929#discussioncomment-1354031_

andig, 2021-09-19:
@DerAndereAndi brauchen wir das?

DerAndereAndi, 2021-09-20:
Das generelle Problem sehe ich auch wenn man 2 EVs hat: ich lade zu Hause und meine Frau unterwegs, welcher ist es jetzt wenn der Charger keine Kennung übergibt. Die Reihenfolge aus der Konfig-Liste nehmen wird auch Fehler bringen.

D.h. generell brauchen wir hier eine Lösung, aber wir haben ja auch mit @naltatis da Überlegungen am Laufen wie man das in der UI machen kann. Einen Schalter: "Automatisch | EV 1 | EV2" sehe ich da momentan nicht wirklich als Lösung. Hast das dann an allen EVSEs? Und was ist mit einem Gast?

Also ja, das Problem wird im Allgemeinen schon immer öfter auftreten und da brauchen wir eine Lösung.

stfn7, 2021-09-20:
Wie sehen die Überlegungen fürs UI aus? Das Thema Gast hatte ich gestern in Ergänzung zum Post angedeutet - hier einfach einen Schalter "Gast" bei dem gar keine Abragen am EV stattfinden sondern Charger und EV das Ding unter sich ausmachen.

DerAndereAndi, 2021-09-20:
Es gibt noch nichts zum Zeigen oder Diskutieren. Der Ansatz ist EVs vom Charger zu trennen und dann frei zuordnen zu können, falls es keine automatische Erkennung gibt.

stfn7, 2021-09-20:
Das geht ja inhaltlich in die gleiche Richtung wie ein Schalter analog zum Modus. Ist vermutlich "nur" schöner :)

naltatis, 2021-09-20:
Ja, führend sollte weiterhin die Heuristik von evcc sein. Erkennt die das Fahrzeug nicht korrekt ist die Idee, dass man das Fahrzeug über die UI für diesen einen Ladevorgang manuell überschreiben kann.

stfn7, 2021-09-20:
Die Herausforderung besteht in diesem Fall in der Frage, wann erkennt die Heuristik, dass sie es nicht korrekt erkennt?

In dem von mir beschriebenen Szenario mit obigem Logfile war die Lage ja eindeutig. Tesla meldet A, EVCC erkennt am Charger B und glaubt dennoch an den Tesla. Würden jetzt beide Fahrzeuge parallel an verschiedenen Orten geladen werden, kommt das Verfahren womöglich an seine Grenzen.

Die Herausforderung Tesla mit State und Kia ohne wird auch eine Trennung vom Charger womöglich nicht zuverlässig lösen.

DerAndereAndi, 2021-09-20:
Es gibt Szenarien wo eine eindeutige Erkennung möglich ist, und es gibt welche (wie in deinem Beispiel) wo es nicht möglich ist. Z.B. wenn der Charger durch eine Kennung des EVs dieses erkennen kann, oder eine RFID Karte, oder wenn es nur 1 angemeldetes EV gibt, etc. pp

stfn7, 2021-09-20:
Wenn man in der Konfig des Chargers konfigurieren kann, ob man den Schalter haben möchte?

DerAndereAndi, 2021-09-20:
Wie bereits gesagt, es gibt noch nichts das man präsentieren oder diskutieren kann. Daher musst du dich erst mal gedulden.

andig, 2021-09-20:
> Das generelle Problem sehe ich auch wenn man 2 EVs hat: ich lade zu Hause und meine Frau unterwegs, welcher ist es jetzt wenn der Charger keine Kennung übergibt. Die Reihenfolge aus der Konfig-Liste nehmen wird auch Fehler bringen.

Das ist ein wenig OT- hier geht es um *einen* ganz konkreten Fall der sich lösen liesse, siehe Beschreibung.

DerAndereAndi, 2021-09-20:
Die Auswahl wie vorgeschlagen sehe ich nicht als gute Lösung.

Davon auszugehen wenn ein EV sagt es ist nicht angeschlossen, dass es das 2te sein könnte, kann man machen. Aber was ist wenn es in dem Fall ein Gast ist, dann ist es auch falsch. Diese ganze Heuristik ist imho ein Loch ohne Boden.

stfn7, 2021-09-20:
Idee: Default für jeden Charger ist die Heuristik, wie bisher. In der Config kann diese jedoch für einzelne Charger deaktiviert und ein Default-EV angegeben werden. Dieses EV wird bei deaktivierter Heuristik immer angenommen und im UI kann der User dann am Charger eines der anderen EVs auswählen.

Die Frage ob Gäste möglich sind, könnte auch in der Config am Charger konfiguriert und nur dann im UI eingeblendet werden.

premultiply, 2021-09-20:
Warum nicht einfach kurz ein zugehöriges RFID-Tag am Leser vorbeziehen wenn man das Kabel aus der Halterung nimmt? 🤷🏻‍♂️

stfn7, 2021-09-20:
@premultiply ich habe verstanden, dass Du die Dinger großartig findest. In dem Ausgangsszenario fährt meine Frau den Kia und lädt hauptsächlich zu Hause. Der Tesla wird von mir überwiegend am Arbeitsplatz geladen und ab und an daheim. Aus Erfahrung klug, muss das Laden Zuhause mit möglichst wenigen Handgriffen erledigt sein. Also Anschluss auf, Kabel rein, Laden. Das an einem anderen Charger erprobte RFID war, wenn nicht korrekt vorgehalten, eine Fehlerquelle, die das Laden verhinderte.

premultiply, 2021-09-20:
Nein, ich finde sie nicht großartig aber es wäre eine Möglichkeit ohne Heuristik-Automatismus ein Fahrzeug per Benutzerinteraktion dynamisch aber sicher einem LP zuzuordnen.
Und zumindest die Leser die ich kenne sind sehr tolerant und geben auch gleich einen Bestätigungston aus wenn die der Tag korrekt gelesen wurde.
Da die Teile so klein sind kann man sie auch gut an bewegliche Teile wie z. B. Keycards oder ggf. auch Ladekabelstecker (wenn nicht wie empfohlen fest angeschlagen) des jeweiligen Autos.

Eine weitere Variante wäre das per Telegram als Push-Rückfrage zu lösen.
Telegram kann über Bots auch so Dialogschaltflächen etc. anzeigen und Rückmelden.

martingraml, 2021-09-22:
Ich fände das großartig, wenn es in der UI bei der Ladestation ein Dropdown gibt, wo man manuell die Heuristik overrulen kann. Nur für den aktuellen Ladevorgang. Später wenn es mal die "Garage" gibt in der alle Fahrzeuge aufgelistet sind, könnte man es auch am Fahrzeug festmachen, wo es gerade lädt. 
Zusätzlich wäre das mit Telegram und der Push-Rückfrage natürlich am bequemsten.

andig, 2021-09-23:
Die Diskussion hier ist mittlerweile völlig off-topic. 

These: wir brauchen keine bessere Heuristik sondern einen Mechanismus für manuelle Eingriffe. 

Falls korrekt können wir das Thema auch retitlen.

closed 2021-09-24 by andig