evcc-io/evcc #1631: Allow change of target SoC when car is disconnected

Alestrix, 2021-09-24:
Since upgrade to the slider-based target-SoC the target level cannot be adjusted anymore when the car is disconnected. This is particularly a problem when the person connecting the car and the person accessing the GUI are not the same. But also in general it is very annoying, as it disallows target-SoC planning ahead of time and forces a certain order of steps which ~might~will not fit everyone.

Please allow adjusting of target SoC when car is disconnected.

_Originally posted by @Alestrix in https://github.com/evcc-io/evcc/discussions/998#discussioncomment-813116_

Also see https://github.com/evcc-io/evcc/discussions/998#discussioncomment-1376511

andig, 2021-09-24:
/cc @naltatis ist das eine Einschränkung in UI?

Alestrix, 2021-09-29:
Not fixed with #1650 
[image]

Alestrix, 2021-09-29:
...das sollte in docker image andig/evcc:latest nach meinem Verständnis drin sein, oder?

andig, 2021-09-30:
/cc @naltatis