evcc-io/evcc #1636: Vehicle Identify

Chris591, 2021-09-25:
**Describe the bug**
2 Cars configured with 2 LP's and both cars configured to both LP's then you can run into following issue:
If you connect 2 Cars to 2 Wallboxes nearly at the same time or if you restart EVCC during both Cars are connected.
The identification isn't working EVCC don't know which Car is connected to which LP.
But as I got RFID identification it should and it is identifying them correctly for a very short time.

**To Reproduce**
Steps to reproduce the behavior:
If you connect 2 Cars to 2 Wallboxes nearly at the same time or if you restart EVCC during both Cars are connected.

**Expected behavior**
if you use RFID tokens it should stop trying to identify the Car by which is connected.

**EVCC details:**
Show output of <code>evcc -v</code>:
```
<- paste here

```

<details><summary>Show output of <code>evcc dump -c configfile</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc configuration file <code>evcc.yaml</code>:</summary><br><pre>
<- paste here

</pre></details>

<details><summary>Show evcc log output with <code>--log debug</code>:</summary><br><pre>
<- paste here

</pre></details>

**If using Docker:**
Show output of <code>docker run andig/evcc -v</code>:
```
evcc version 0.64 (d2e3c47)

```

<details><summary>Show evcc log output with <code>docker logs <container id></code>:</summary><br><pre>

[lp-1  ] INFO 2021/09/25 10:14:53 loadpoint 1:
[lp-1  ] INFO 2021/09/25 10:14:53   mode:      off
[lp-1  ] INFO 2021/09/25 10:14:53   charger:   power ✓ energy ✓ currents ✓ timer ✓
[lp-1  ] INFO 2021/09/25 10:14:53   meters:    charge ✓
[lp-1  ] INFO 2021/09/25 10:14:53     charge:  power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2021/09/25 10:14:53   vehicles:  ✓
[lp-1  ] INFO 2021/09/25 10:14:53     car 0:   range ✓ finish ✗ status ✓ climate ✗
[lp-1  ] INFO 2021/09/25 10:14:53     car 1:   range ✓ finish ✓ status ✓ climate ✓
[lp-2  ] INFO 2021/09/25 10:14:53 loadpoint 2:
[lp-2  ] INFO 2021/09/25 10:14:53   mode:      off
[lp-2  ] INFO 2021/09/25 10:14:53   charger:   power ✓ energy ✓ currents ✓ timer ✓
[lp-2  ] INFO 2021/09/25 10:14:53   meters:    charge ✓
[lp-2  ] INFO 2021/09/25 10:14:53     charge:  power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2021/09/25 10:14:53   vehicles:  ✓
[lp-2  ] INFO 2021/09/25 10:14:53     car 0:   range ✓ finish ✓ status ✓ climate ✓
[lp-2  ] INFO 2021/09/25 10:14:53     car 1:   range ✓ finish ✗ status ✓ climate ✗
[lp-1  ] DEBUG 2021/09/25 10:14:53 charge power: 0W
[lp-1  ] DEBUG 2021/09/25 10:14:53 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/09/25 10:14:53 charger status: B
[lp-1  ] DEBUG 2021/09/25 10:14:53 charger vehicle id: xxxxxx
[lp-1  ] INFO 2021/09/25 10:14:53 vehicle updated: unknown -> Mustang Mach-E
[lp-1  ] DEBUG 2021/09/25 10:14:53 running vehicle action: {now 80}
[lp-1  ] INFO 2021/09/25 10:14:53 set charge mode: now
[lp-1  ] INFO 2021/09/25 10:14:53 set target soc: 80
[lp-1  ] DEBUG 2021/09/25 10:14:54 vehicle status: B (Mustang Mach-E)
[lp-1  ] DEBUG 2021/09/25 10:14:54 vehicle status: B (Cubra Leon)
[lp-1  ] DEBUG 2021/09/25 10:14:54 vehicle status: >1 matches, giving up
[lp-1  ] INFO 2021/09/25 10:14:54 vehicle updated: Mustang Mach-E -> unknown
[lp-1  ] DEBUG 2021/09/25 10:14:54 charger disable
[lp-1  ] DEBUG 2021/09/25 10:14:54 charge power: 0W
[lp-1  ] DEBUG 2021/09/25 10:14:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2021/09/25 10:14:54 charger status: B
[lp-1  ] DEBUG 2021/09/25 10:14:54 vehicle status: B (Mustang Mach-E)
[lp-1  ] DEBUG 2021/09/25 10:14:54 vehicle status: B (Cubra Leon)
[lp-1  ] DEBUG 2021/09/25 10:14:54 vehicle status: >1 matches, giving up
[lp-1  ] WARN 2021/09/25 10:14:54 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2021/09/25 10:14:54 max charge current: 16A
[lp-1  ] DEBUG 2021/09/25 10:14:54 charger enable
[lp-2  ] DEBUG 2021/09/25 10:15:03 charge power: 0W
[lp-2  ] DEBUG 2021/09/25 10:15:03 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2021/09/25 10:15:03 charger status: B
[lp-2  ] DEBUG 2021/09/25 10:15:03 charger vehicle id: yyyyyy
[lp-2  ] INFO 2021/09/25 10:15:03 vehicle updated: unknown -> Cubra Leon
[lp-2  ] DEBUG 2021/09/25 10:15:03 running vehicle action: {now 100}
[lp-2  ] INFO 2021/09/25 10:15:03 set charge mode: now
[lp-2  ] INFO 2021/09/25 10:15:03 set target soc: 100
[lp-2  ] DEBUG 2021/09/25 10:15:03 vehicle status: B (Cubra Leon)
[lp-2  ] DEBUG 2021/09/25 10:15:03 vehicle status: B (Mustang Mach-E)
[lp-2  ] DEBUG 2021/09/25 10:15:03 vehicle status: >1 matches, giving up
[lp-2  ] INFO 2021/09/25 10:15:03 vehicle updated: Cubra Leon -> unknown
[lp-2  ] DEBUG 2021/09/25 10:15:03 charger disable
[lp-2  ] DEBUG 2021/09/25 10:15:03 charge power: 0W
[lp-2  ] DEBUG 2021/09/25 10:15:03 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2021/09/25 10:15:03 charger status: B
[lp-2  ] DEBUG 2021/09/25 10:15:03 vehicle status: B (Cubra Leon)
[lp-2  ] DEBUG 2021/09/25 10:15:03 vehicle status: B (Mustang Mach-E)
[lp-2  ] DEBUG 2021/09/25 10:15:03 vehicle status: >1 matches, giving up
[lp-2  ] WARN 2021/09/25 10:15:03 charger out of sync: expected disabled, got enabled
[lp-2  ] DEBUG 2021/09/25 10:15:03 max charge current: 16A
[lp-2  ] DEBUG 2021/09/25 10:15:03 charger enable

</pre></details>

andig, 2021-09-25:
Ich kann in Deinem Log nicht erkennen, dass Du RFID verwendet hast. An welcher Stelle soll das passieren? Ansonsten hast Du natürlich recht- nach RFID Identifikation sollte es nicht mehr über den Status versucht werden.

andig, 2021-09-25:
Ansonsten mache ich hier mal zu bis auch der relevante Teil der Config sowie `evcc dump` zu sehen ist.

Chris591, 2021-09-25:
[lp-1  ] DEBUG 2021/09/25 10:14:53 charger vehicle id: xxxxxx

[lp-1  ] INFO 2021/09/25 10:14:53 vehicle updated: unknown -> Mustang Mach-E

Chris591, 2021-09-26:
Ich hoffe ich hab jetzt alles in den Bug Report hinzugefügt.
Wenn noch was benötigt wird oder getestet werden soll bitte bescheid geben.

PR #1889 (2021-11-18): Improve vehicle identification experience
- allow multiple identifiers per vehicle to support RFID and MAC ~~and VIN~~ across multiple chargers
- ~~add `VIN` as default identifier~~
- move `onIdentify` action config from loadpoint to vehicle
- add `minSoC` to action config
- replace `onDisconnect` with simpler `resetOnDisconnect`

Fix #1636, fix https://github.com/evcc-io/evcc/issues/1891, fix https://github.com/evcc-io/evcc/issues/1893

/cc @chris591 @xerion @bicaluv
closed 2021-11-18 by andig